merbanan/rtl_433 #1581: Exit on device errors (optional), eg LIBUSB_ERROR_INVALID_PARAM

netspoon, 2020-12-14:
Hi,

I occasionally get errors after rtl_433 has been running for a few hours:

LIBUSB_ERROR_INVALID_PARAM: Invalid parameter!
Check your RTL-SDR dongle, USB cables, and power supply.

Which are generally solved by stopping and restarting the rtl_433 process.

Could we have a run time option to make rtl_433 exit on such errors ?

I control the process via supervisord, so it will attempt to restart if it exited, but in this case it just keeps logging the above warning rather than the rtl_433 process stopping / exiting.

Thanks,

grb

zuckschwerdt, 2020-12-14:
Yes, that's a fatal error and we did exit. I guess I messed this up on refactoring the SDR loop.
I'll take a look now.

closed 2020-12-14 by zuckschwerdt

netspoon, 2020-12-14:
many thanks

commit 405df885 (2020-12-14): Fix rtlsdr_read_async() abort on read stall (closes #1581)