merbanan/rtl_433 #1585: Decoding Thingsend XZ TP10 and TH10

silverman10, 2020-12-20:
Hi,
Per the discussion over at https://groups.google.com/g/rtl_433/c/Ymvjn2YFW-s/m/6uxl1_riBQAJ

Trying to decode the signal from a Thingsend Temp/Humidity sensors TP10 and TH10. Analyzer shows FSK packages but "No Clue" with decoding.
Signal recordings attached as TH10.zip
[TH10.zip](https://github.com/merbanan/rtl_433/files/5721252/TH10.zip)
. I have tried my best to avoid any "noise" from other devices, but removing the antenna and placing the sensors closer to the receiver.

merbanan, 2021-03-21:
`/rtl_433 -r TH10/g016_433.92M_250k.cu8 -X "n=TH10,m=FSK_PCM,s=50,l=50,r=4000,y=5000,preamble={24}0x5555C8"`

```
Temp:
af6983f3e3feeefff3d3bea62eece7d407ff07aebac097c277b4e6070111155557
af6983f3e3feeefff3dca6a373ff33d15bef1abeabc06994f03483d54c15445557

Trigger mode:
af6983f3e3feeefff3dca6a373fff22ee3fceafbb3d0789bd4af9116f010415557c
af6983f3e3feeefff3dca6a373fff22ed3bf96afa6d0789b25d1a1c9d1554555578
```
Based on http://thingsend.com/uploadfile/2018/0727/20180727054320128.pdf
the transmitted signal is a uart signal.

rtl_433 -r TH10/g001_433.92M_250k.cu8 -W test.sr will open sigrok and you can play around with the signal Sigrok has a uart decoder and it looks like it matches quite well, but the question is what parameters are used.

merbanan, 2021-03-22:
With a baud of 19200 the timing looks right but the decoded uart frames have incorrect stop bits. The occurrences of uart frames are irregular also. Maybe we have a whitening pattern here. AA or 55 that needs to be removed before we decode the uart frames.

merbanan, 2021-03-22:
@silverman10 you can connect to the XZ-DT25 serial pins to see what the sensor is transmitting. Then we can match the date from the signal.

You can also remove the metal shield to look at the transceiver chip. To me it looks like an Arduino connected to an external chip. If we get the transceiver that might help.

silverman10, 2021-03-22:
thanks for picking this up @merbanan .

I had opened up the receiver, the module is listed as a XZ-DT25-UX v1.1. 

I am happy to send over receiver and a sensor over if that helps.

gdt, 2023-09-30:
@silverman10 What's the status and path forward?  Can you test with up-to-date rtl_433?   I don't have a reason to think this was added. but a lot happened in 2.5 years.

silverman10, 2023-09-30:
Yep, sure. I'll try to test with the latest version.