merbanan/rtl_433 #1586: No data received after reboot on rasperry pi, only after detaching and reattaching device

amisk, 2020-12-20:
Hi,
I use a Realtek rtl 2838 device (ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T) on a raspberry pi 3b running on raspbian 10 buster.
I noticed, that if I restart the pi, with the stick plugged in, and immediately start recording, no data is received. If I detach the stick and put it back in, it works as usual. It doesn't matter if I do it with or without sudo.

Is there a way so it works immediately after a reboot? I want to have this run 24/7, but it would be bad if there is power outage and I don't notice it.
Thanks.

zuckschwerdt, 2020-12-20:
Can you show the error in more detail? Won't rtl_433 abort (`Async read stalled, exiting`) if there is no data from the RTL-SDR? Or is I/Q data actually delivered but not usable?
What version are you running? The `20.11` release version has a bug where it notes the stall condition but doesn't exit.

RandomSporadicProjects, 2020-12-22:
I've had similar issues in the past (long time ago), but found a work around (cannot remember the original source)

Check out 
https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line

for instructions on compiling usbreset  (I store mine in /home/pi/scripts/)

I then have a script (resetsdr.sh) that resets my rtl2838
that contains 

#!/bin/bash
USBNAME=RTL2838 
LSUSB=$(lsusb | grep --ignore-case $USBNAME) 
DEVICE="/dev/bus/usb/"$(echo $LSUSB | cut --delimiter=' ' --fields='2')"/"$(echo $LSUSB | cut --delimiter=' ' --fields='4' | tr --delete ":")
/home/pi/scripts/usbreset $DEVICE 2>&1 >/dev/null

You could put this in a cronjob to run on a reboot or periodically as required

HTH

bkenobi, 2021-01-08:
I have been noticing the same issue.  My setup is identical: Buster, RPi3B, RTL2838  (EZCap USB dongle though it seems to be detected as Fitipower FC0013).  If I manually start rtl_433 everything seems to start up reliably.  If I run it in a systemctl service and reboot the RPi, I have to manually stop the service, run rtl_433 interactively, stop the interactive, and relaunch the service.  I do not have to disconnect the SDR, just restart the program.

Is it possible that the service is launching too early and the SDR isn't ready when the program launches?

zuckschwerdt, 2021-01-10:
Very likely an USB issue and nothing rtl_433 can fix.

closed 2021-01-10 by zuckschwerdt