merbanan/rtl_433 #1587: Support for Bresser 3 in 1 wind meter

ruabo, 2020-12-22:
Hello,

is this supported ?

[https://www.bresser.de/it/Ora-Tempo/Centro-meteo/Anemometro-professionale-BRESSER-3-in-1-con-terminale-collegato-tramite-segnale-radio.html?gclid=EAIaIQobChMIwNWFuszi7QIVSuh3Ch0IdwwjEAAYASAAEgLQ4vD_BwE](url)

I'd like to buy it and to interface it to hassio by means of rtl_433
[](url)

zuckschwerdt, 2020-12-23:
What product code is the outdoor sensor?
It's likely that it works with the Bresser Weather Center 6-in-1 protocol. See if it is listed compatible with 5-in-1,  6-in-1, or 7-in-1 -- we have support for those.

ruabo, 2020-12-23:
Sorry, I can't find the p/n in the manual; also, no mention of x-in-1 compatibility

peteakalad, 2020-12-23:
It *may* work with the existing implementations for the other Bresser stations as mentioned above, but not definite until you try it.  The receiver does look similar to the 5 in 1.  Dont see any negative nor positive reports of 3 in 1 users.

If you haven't bought it yet perhaps consider a 5 in 1 as it is known to work (I own one, product code 7002510). +10 euro, additional wind direction and rain:
https://www.amazon.it/BRESSER-Stazione-meteorologica-5-1/dp/B0170A47DI

ruabo, 2020-12-25:
Thanks. Unluckily i have no room for that sensor, I have a narrow space and the 3 in 1 would fit. Anyway, I'll think about trying the 3 in 1.

ruabo, 2021-01-24:
I typed:
`rtl_433 -f 868.3M`
and got this output:

`time      : 2021-01-24 17:03:10
model     : Bresser-6in1 id        : 20010691
channel   : 0            Battery OK: 1             Temperature: 8.0 C
Humidity  : 60           Sensor type: 1            Wind Gust : 0.0 m/s
Wind Speed: 0.0 m/s      Direction : 0             Integrity : CRC
`
Values of T, H and wind are correct !

zuckschwerdt, 2021-01-24:
So the Bresser 3in1 wind meter works and is shown as "Bresser-6in1", right? If confirmed I'll add that to the notes.

ruabo, 2021-01-24:
@zuckschwerdt yes, correct

zuckschwerdt, 2021-01-24:
Note added with f138c31, thanks!

closed 2021-01-24 by zuckschwerdt