merbanan/rtl_433 #1589: rtl_433 stopped working with Bresser 5in1 and can't get it to work again

pmknowles, 2020-12-26:
About 2 weeks ago rtl_433 stopped sending MQTT messages, I've reinstalled the os and rtl_433 but can't retrieve any data.
If I  use _rtl_433 -f 868.3M -s 1M -X 'n=Bresser-6in1,m=FSK_PCM,s=100,l=100,r=2000'_
I get
```
rtl_433 version 20.11-9-g8cdb56d9 branch master at 202012151640 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 146 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-12-26 20:31:01
model     : Bresser-6in1 count     : 1             num_rows  : 1
rows      : 
len       : 269          data      : aaaaaaaaaa8b753cd19ffadbf737ceffbc677fffc62e600a2408c4188021cc800000
codes     : {269}aaaaaaaaaa8b753cd19ffadbf737ceffbc677fffc62e600a2408c4188021cc800000
```
so I presume it's receiving data
but if I use my .conf file I get no output from MQTT (I've stripped it down to the bare minimum)
_device        0
gain          0
frequency     868.3M
sample_rate  1000k
output mqtt://192.168.0.211:1883,user=openHAB,pass=NadiaR26,retain=0,devices=st$
convert si
protocol 119_
Can anyone see what I'm doing wrong?

zuckschwerdt, 2020-12-26:
I guess the `devices=st$` is not the actual line but a copy from a pager/vi?

pmknowles, 2020-12-27:
The MQTT section is unchanged from when it worked. The sample rate was originally 250k. The conf file is the original one which worked but I have removed lines in attempts to get it to work.

pmknowles, 2020-12-27:
This is the output after running rtl_433
```
rtl_433 version 20.11-9-g8cdb56d9 branch master at 202012151640 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Reading conf from "rtl_433.conf".

New defaults active, use "-Y classic -s 250k" for the old defaults!

Publishing MQTT data to 192.168.0.211 port 1883
Publishing device info to MQTT topic "stat/weather[/id]".
Registered 1 out of 175 device decoding protocols [ 119 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
MQTT Connected...
MQTT Connection established.
```

pmknowles, 2020-12-27:
Now, all of a sudden, it's started to work again

pmknowles, 2020-12-27:
Now it's stopped again after 20 messages via MQTT and refuses to send anything again

merbanan, 2020-12-27:
When it is not working try running with "rtl_433 -f 868.3M -s 1M" most likely no decodes will be shown. If that is the case something is interfering with the signal. Try to adjust the frequency a little bit that might help. If not then try with a spectrum analyser to see what might be the issue. Running with -A might also give a hint what could be the issue.

pmknowles, 2020-12-27:
Finally solved it. I could not get an output at 250k (which I did before) but could only get an output at 1M. I also had to use -Y classic (which I didn't before).

closed 2020-12-27 by pmknowles