merbanan/rtl_433 #1591: Support Cotech Wireless Rain Gauge E0013TRF-A1V-YX (Clas Ohlson art.no. 36-7900)

robinsmidsrod, 2021-01-01:
I recently got this device, and I'd like to use it with rtl_433. Any chance of supporting it?

Here is the technical information I could gleam from the stickers and manual:

Cotech Wireless Rain Gauge
Clas Ohlson Art. no. 36-7900
Model no. E0330WST2 (Indoor unit)
Model no. E0013TRF-A1V-YX (Rain gauge)
433.92 MHz

https://www.manualslib.de/manual/559395/Cotech-E0330Wst2.html

What can I do to help this along?

zuckschwerdt, 2021-01-01:
Verify the frequency with CubicSDR, Gqrx, or SDRangel.
Capture packets with `-S unknown`.
Check for a good signal with https://triq.org/pdv/ 
Run the analyzer mode `-A`
Post findings and a zipped packet sample here for more help.

merbanan, 2021-03-07:
No update for a long time and no signal samples upload. Closing issue.

closed 2021-03-07 by merbanan

robinsmidsrod, 2021-03-07:
@merbanan Sorry about that. Life has gotten in the way. I'll reopen once I have time to generate the required files.