merbanan/rtl_433 #1593: Does this works for MX-05V Receiver?

GF-Huang, 2021-01-02:
[image]

https://www.geekbuying.com/item/MX-05V-433MHZ-RF-Wireless-Receiver-Module-MX-FS-03V-Transmitter-Module-Kit-For-Arduino-Raspberry-Pi-AVR-ARM-344591.html

zuckschwerdt, 2021-01-02:
No.

closed 2021-01-02 by zuckschwerdt

GF-Huang, 2021-01-02:
Any suggestions? Please.

zuckschwerdt, 2021-01-02:
Those ultra-cheap receiver are basically just 1 bit OOK.
SDR like the cheap RTL-SDR are  8 bit OOK and FSK.

rtl_433 works with [RTL-SDR](https://github.com/osmocom/rtl-sdr/) and/or [SoapySDR](https://github.com/pothosware/SoapySDR/).
Actively tested and supported are Realtek RTL2832 based DVB dongles (using RTL-SDR) and LimeSDR ([LimeSDR USB](https://www.crowdsupply.com/lime-micro/limesdr) and [LimeSDR mini](https://www.crowdsupply.com/lime-micro/limesdr-mini) engineering samples kindly provided by [MyriadRf](https://myriadrf.org/)), PlutoSDR, HackRF One (using SoapySDR drivers).

sp372, 2022-11-30:
Can work this: https://es.aliexpress.com/item/32706433141.html?spm=a2g0o.order_list.order_list_main.132.1ee5194dfUnwFo&gatewayAdapt=glo2esp

sp372, 2022-11-30:
Or this: https://es.aliexpress.com/item/1005004200544468.html?spm=a2g0o.order_list.order_list_main.77.1ee5194dfUnwFo&gatewayAdapt=glo2esp

zuckschwerdt, 2022-12-01:
Maybe. Lots of work. See #1927

sp372, 2023-01-26:
This guy use MX-05V Receiver, how? https://www.youtube.com/watch?v=13vPuve_be0&t=533s