merbanan/rtl_433 #1594: klimalog tfa receive problems

devZer0, 2021-01-02:
i try replacing tfrec with rtl_433 for klimalog, but apparently it does not work as good as tfrec. 

i tried to tune various params, but i can only receive 2 sensors. tfrec receives all 5 of them

# ./rtl_433 -s 2.5e6 -f 868.5e6 -H 30 -M hires -M level  -R 150 -v
rtl_433 version 20.11-12-ga20cd1a6 branch master at 202012232126 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 176 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 2500000.107620 Hz
[R82XX] PLL not locked!
Sample rate set to 2500000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 868.500MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-01-02 20:37:32.055088
model     : Klimalogg Pro Id        : 6f44
Battery   : 1            Temperature: 22.0 C       Humidity  : 35            Sequence Number: 6        Integrity : CRC
Modulation: ASK          Freq      : 868.5 MHz
RSSI      : -12.1 dB     SNR       : 13.4 dB       Noise     : -25.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-01-02 20:37:42.195611
model     : Klimalogg Pro Id        : 6f44
Battery   : 1            Temperature: 22.0 C       Humidity  : 35            Sequence Number: 7        Integrity : CRC
Modulation: ASK          Freq      : 868.5 MHz
RSSI      : -12.1 dB     SNR       : 13.1 dB       Noise     : -25.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-01-02 20:37:53.503918
model     : Klimalogg Pro Id        : 4356
Battery   : 1            Temperature: 22.9 C       Humidity  : 29            Sequence Number: 3        Integrity : CRC
Modulation: ASK          Freq      : 869.1 MHz
RSSI      : -0.1 dB      SNR       : 29.0 dB       Noise     : -29.1 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-01-02 20:38:02.476438
model     : Klimalogg Pro Id        : 6f44
Battery   : 1            Temperature: 22.1 C       Humidity  : 35            Sequence Number: 9        Integrity : CRC
Modulation: ASK          Freq      : 868.5 MHz
RSSI      : -12.1 dB     SNR       : 13.6 dB       Noise     : -25.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-01-02 20:38:03.925687
model     : Klimalogg Pro Id        : 4356
Battery   : 1            Temperature: 23.0 C       Humidity  : 29            Sequence Number: 4        Integrity : CRC
Modulation: ASK          Freq      : 869.1 MHz
RSSI      : -0.1 dB      SNR       : 30.9 dB       Noise     : -31.0 dB

# /root/tfrec
Registering demod for TFA_1 KlimaLoggPro
Registering demod for TFA_2 sensors, 17240 bit/s
type 0x1: Samples per bit: 22.3
Registering demod for TFA_3 sensors, 9600 bit/s
type 0x2: Samples per bit: 40.0
Found Rafael Micro R820T tuner
Allocating 15 zero-copy buffers
TFA1 ID 6f44 +22.3 35% seq 3 lowbat 0 RSSI 68
TFA1 ID 4356 +23.7 29% seq e lowbat 0 RSSI 81
TFA1 ID 3f14 +22.4 35% seq d lowbat 0 RSSI 63
TFA1 ID 6341 +22.2 38% seq 9 lowbat 0 RSSI 57
TFA1 ID 3996 +24.4 29% seq c lowbat 0 RSSI 81

i'm also having trouble receiving those sensors with Fitipower FC0012 based usb stick - rtl_443 does not recevive anything from tfa klimalogg sensors with this stick, but as also tfrec has problems with this different tuner. maybe someone has a clue, why... see  https://github.com/baycom/tfrec/issues/20

zuckschwerdt, 2021-01-02:
> ./rtl_433 -s 2.5e6 -f 868.5e6 -H 30 -M hires -M level -R 150 -v

- `e6` is simpler written as `M`
- no need for `-H` if there is only one frequency
- 2.5M sample rate is a lot, are your sensors that far apart? rtl_433 works best at 250k or 1024k.

merbanan, 2021-01-03:
tfrec has a better demodulation code and rtl_433 will most likely not change for the better. The signal from these sensors are really weird. FC0012 can be made better, a change to rtl-sdr is needed.

merbanan, 2021-01-03:
The high sample rate is needed for the demod to work. klimalog is really strange signal wise.

merbanan, 2021-01-03:
@devZer0 the fc0012 rtlsdr code doesn't change the UHF filter (if there is one). So no patch needed. But you need to set the gain near max. (-g 20)

devZer0, 2021-01-03:
mhh, even with R820T tuner, i only receive 2 out of 5 sensors with rtl_433 - even with gain 20

any clue, why it's missing 3 out of 5 sensors ?

merbanan, 2021-01-03:
The gain comment was for FC0012. The klimalog sensor is really strange. I think there is some trick to demodulate the signal that I am missing.

riegelbrau, 2021-02-14:
Hi @devZer0, 

your settings look strange for me. I only get the KlimaLogg Pro sensors running with "sample_rate   1536k".
See output:
```
root@raspi4:~# rtl_433 -c /etc/rtl_433/klimalogg.conf
rtl_433 version 20.11-72-g914552b9 branch master at 202102121205 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 2 out of 178 device decoding protocols [ 112 150 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1536000 S/s.
Tuner gain set to Auto.
Tuned to 868.250MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-14 17:55:04.534015             Protocol  : 150
model     : Klimalogg Pro Id        : 45f1
Battery   : 1            Temperature: 10.1 C       Humidity  : 106           Sequence Number: 7        Integrity : CRC
Modulation: ASK          Freq      : 868.4 MHz
RSSI      : -4.5 dB      SNR       : 18.7 dB       Noise     : -23.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-14 17:55:05.254575             Protocol  : 150
model     : Klimalogg Pro Id        : 7922
Battery   : 0            Temperature: -5.1 C       Humidity  : 28            Sequence Number: 6        Integrity : CRC
Modulation: ASK          Freq      : 868.6 MHz
RSSI      : -0.4 dB      SNR       : 22.3 dB       Noise     : -22.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-14 17:55:08.897906             Protocol  : 150
model     : Klimalogg Pro Id        : 772b
Battery   : 1            Temperature: 10.0 C       Humidity  : 44            Sequence Number: 9        Integrity : CRC
Modulation: ASK          Freq      : 868.3 MHz
RSSI      : -6.7 dB      SNR       : 16.2 dB       Noise     : -23.0 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-14 17:55:15.377314             Protocol  : 150
model     : Klimalogg Pro Id        : 45f1
Battery   : 1            Temperature: 10.1 C       Humidity  : 106           Sequence Number: 8        Integrity : CRC
Modulation: ASK          Freq      : 868.4 MHz
RSSI      : -4.6 dB      SNR       : 19.1 dB       Noise     : -23.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-14 17:55:15.864147             Protocol  : 150
model     : Klimalogg Pro Id        : 7922
Battery   : 0            Temperature: -5.1 C       Humidity  : 28            Sequence Number: 7        Integrity : CRC
Modulation: ASK          Freq      : 868.6 MHz
RSSI      : -0.5 dB      SNR       : 22.2 dB       Noise     : -22.7 dB
^CSignal caught, exiting!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-14 17:55:16.663945             enabled   : 2             since     : 2021-02-14T17:55:00
frames    :
count     : 7            fsk       : 0             events    : 0             stats     :
Reattached kernel driver
```
I also have issues with some sensors to be read successfully. I assume this might be caused by the antenna position of the SDR stick.
Regards Christoph

merbanan, 2021-03-07:
The current demodulation code is not suitable for this signal. This will probably not be fixed in the near future. If we get to overhaul the dsp demodulation flow then a specific demod for this signal can be added. Closing for now,

closed 2021-03-07 by merbanan