merbanan/rtl_433 #1599: Support WVC-Inverters

tiego-007, 2021-01-06:
Hello, thanks for your great work, I love rtl_433.

For a short time I have had a WVC Solar Inverter which has a 433 MHz interface installed.
Would it be possible to read this out via rtl_433?

Here there would already be some information about it
https://github.com/krywenko/WVC-inverters

Thanks

zuckschwerdt, 2021-01-06:
"Wireless serial modem connection" sounds very much like a bi-directional transmission, rtl_433 can't do that. You also need other hardware, something like a CC1101 might work.

tiego-007, 2021-01-06:
ok, thanks.
i have found more information in
https://community.openenergymonitor.org/t/wvc-inverter-mqtt-data-logging-for-all-versions-of-inverters/12352/240

can please take a look

zuckschwerdt, 2021-01-06:
The advice "connect a serial adapter to the serial tx & rx by passing all the the intermediary" sounds good. If you can reach the Inverter with a cable: use a cable. A cable is good for serial data.

merbanan, 2021-01-06:
I think you have to send commands to get any output. If the WVC is sending out data randomly you could write a protocol decoder for that.

gdt, 2023-09-30:
This was basically a question and does not appear headed for an improvement to the codebase.   Please feel free to open a discussion or ask on the mailinglist  if you have further questions.

closed 2023-09-30 by gdt