merbanan/rtl_433 #1603: Support for TFA Marbella Pool thermometer

JensBohlin, 2021-01-10:
Hi 

Great project!!

Please add support for TFA Dostmann Marbella (30.3238.06).
https://www.tfa-dostmann.de/produkt/funk-poolthermometer-marbella-30-3066/

I've made a PR to rtl_433_tests with .cu8 and json-files and some analysis on how to interpret the temperature values.

/Jens

merbanan, 2021-03-14:
I wrote a decoder: https://github.com/merbanan/rtl_433/pull/1675 for this protocol. But the temperature is not decoded properly.

I created this bitbench also:

https://triq.net/bitbench?c=5555555555555555695516ea341f8b0513a555358%20%5Bg001%3A%2022.8%5D&c=%EF%BF%BC5555555555555555695516ea341f8b0613a555310%20%5Bg002%3A%2022.8%5D&c=%EF%BF%BC5555555555555555695516ea341f8b0713a555328%20%5Bg003%3A%2022.8%5D&c=%EF%BF%BC5555555555555555695516ea341f8b0013a555380%20%5Bg004%3A%2022.8%5D&c=%EF%BF%BC5555555555555555695516ea341f8b071c6d551e8%20%5Bg005%3A%2059.0%5D&c=%EF%BF%BC5555555555555555695516ea341f8b001c0555760%20%5Bg006%3A%2049.6%5D&c=%EF%BF%BC5555555555555555695516ea341f8b011b4d55288%20%5Bg007%3A%2047.3%5D&c=%EF%BF%BC5555555555555555695516ea341f8b021a9555018%20%5Bg008%3A%2045.0%5D&c=%EF%BF%BC5555555555555555695516ea341f8b0319ed555c8%20%5Bg009%3A%2042.9%5D&c=%EF%BF%BC5555555555555555695516ea341f8b04194d55318%20%5Bg010%3A%2040.9%5D&c=%EF%BF%BC5555555555555555695516ea341f8b0518cd55570%20%5Bg011%3A%2039.3%5D&c=%EF%BF%BC5555555555555555695516ea341f8b01142555630%20%5Bg017%3A%2024.6%5D&c=%EF%BF%BC5555555555555555695516ea341f8b000f6d557a0%20%5Bg022%3A%209.3%5D&c=%EF%BF%BC5555555555555555695516ea341f8b010f3d55060%20%5Bg023%3A%208.7%5D&f=SYNC%3Ahhhh%20SERIAL%3Ahhhhhh%20ZERO%3Ah%20COUNTER%3Ah%20TEMP%3Ahhh%20POST%3Ahhh%20LFSR%3Ahh%20ZERO%3Ah&a=Sync&m=2dd4&cw=4

merbanan, 2021-03-14:
This flex code seems to handle things well. I think we need the classic demod to get the last byte with the lsfr.

rc/rtl_433 -s 1000k -Y classic -r ../rtl_433_tests/tests/TFA-Marbella-30.3238.06/*.cu8 -X "n=marbella,m=FSK_PCM,s=105,l=105,r=2000,preamble=aa2dd4"

merbanan, 2021-03-14:
[values.txt](https://github.com/merbanan/rtl_433/files/6135854/values.txt)

The attached file gives this revdgst output:
./revdgst values.txt 
Code len is 8 bytes (16 nibbles), 14 codes read.
Processing...
Time elapsed in s: 0.04 for: Galois 

Time elapsed in s: 0.04 for: Galois BYTE_REFLECT

Done with g 98 k bc final XOR 27 using xor xor (100 %)
Time elapsed in s: 0.04 for: Galois BIT_REFLECT

Time elapsed in s: 0.04 for: Galois BIT_REFLECT BYTE_REFLECT

Time elapsed in s: 0.11 for: Rev-Galois 

Done with g 31 k 31 final XOR 00 using xor xor (100 %)
Done with g 31 k 31 final XOR 00 using xor sub (100 %)
Time elapsed in s: 0.13 for: Rev-Galois BYTE_REFLECT

So it seems there is an integrity check that matches the posted codes.

merbanan, 2021-03-14:
The minmax demod drops a few bits at the end for some reason.

merbanan, 2021-03-14:
@JensBohlin anyway, can you run with:

rtl_433 -s 1000k -f 868M -Y classic "n=marbella,m=FSK_PCM,s=105,l=105,r=2000,preamble=aa2dd4" and post a few more sensor messages together with temperature readings?

And can you elaborate some more in detail on how you converted the temperature?

merbanan, 2021-03-14:
Ok, I figured out the correct temperature formula also. No need for more samples. I'll sort everything out after some sleep.

merbanan, 2021-03-14:
Implemented support with https://github.com/merbanan/rtl_433/pull/1675. Needs rtl_433 -f 868M -Y classic to work properly.

closed 2021-03-14 by merbanan