merbanan/rtl_433 #1604: Generating Checksum for GT-WT-03 sensor data

axel1202, 2021-01-10:
Hello,

first and foremost: Thanks for supplying the declined world with that very helpful tool! Hopefully, I could get some hints her on the following:

It is not really an issue regarding the RTL_433 tool, but rather me having diffuclties to understand how to generate a proper checksum as I want to emulate a GT-WT-03 sensor with a ESP8266 based weather station, that besides publishing data via MQTT also should send information on 433MHz for a stand alone display. 

The following shows an original message from an actual GT-WT-03 sensor and a generated message, both analyzed with URH. The transmitted information is sensor ID 184, battery low, button not pressed, channel 1, 23.5°C, and 48% humidity.
```
10111000.00101010.0.0.00.000011101011.**10111111**.0 <-- original message
10111000.00101010.0.0.00.000011101011.**10010010**.0 <-- generated message
```
As can be seen, the only difference is the checksum - 0xBF from the actual sensor, 0x92, what I calculated, virtually using the code from _chk_rollbyte_ in the decoder source file for this sensor. This is the relevant code snippet in my implementation:

`	payload = (uint8_t *)&bits;

	// calculate the checksum - code copied from 
	// https://github.com/merbanan/rtl_433/blob/master/src/devices/gt_wt_03.c
	// thanks so far :-)

	for (k = 0; k < 4; k++)
	{
		data = payload[k];

		sprintf (gtwt03dbg, "\n0x%02X\n", data);
		wsdebug (gtwt03dbg);

		key = GTWT03_CHKSUM_KEY;

		for (i = 7; i >= 0; --i)
		{
			// XOR key into sum if data bit is set
			if ((data >> i) & 1)
			{
				checksum ^= (key & 0xff);
			}

			sprintf (gtwt03dbg,"0x%02X ", key & 0xFF);
			wsdebug (gtwt03dbg);

			// roll the key right
			key = (key >> 1);
		}
	}
`
The variable payload contains the relevant 4 data bytes in the proper order (payload[0] -> ID, payload[1] -> humidity, etc...), I checked the conternt of the data bytes and also the values of the rotating key - used 0x3100 - as can bee seen in the debug output:

`
0xB8
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
0x2A
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
0x00
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
0xEB
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
chksum = 0x92
`
the only difference is, that in the original source code the bit buffer gets inverted before calling the checksum funktion:
`
 bitbuffer_invert(bitbuffer);
`
As far as I understood this function, it flips all used data bits of a row - assuming I got that right, this bitbuffer row would be equivalent to my 32 bit payload. Inverting this, does change the content of the payload, but doesn't affect the checksum - it remains 0x92 - no matter what I do. Reversing the byte order wouldnt change anything anyway...

Guess I am lost :-) I shall be very happy, if anyone could help my on that subject, eventho it is no issue with the RTL_433 tool, which recognizes the sensor perfectly - but not the data I am sending :-(

Best regards

Axel

merbanan, 2021-01-10:
Hi, you should be able to feed byte strings to rtl_433 with the help of -y parameter. Then you can pass the same payload to both code bases. Then you can instrument the code to locate the issue.

zuckschwerdt, 2021-01-11:
Nothing obvious stands out, but did you account for the `init` value of `0x2d`?
About the `bitbuffer_invert`, this changes the interpretation of the PWM pulses (usual: long=0, short=1, i.e. higher frequency: higher value). The bits are as you see them in the code without any inversion.

axel1202, 2021-01-11:
> Nothing obvious stands out, but did you account for the `init` value of `0x2d`?

that leads to a checksum of 0x1A, which is still wrong :-)

> About the `bitbuffer_invert`, this changes the interpretation of the PWM pulses (usual: long=0, short=1, i.e. higher frequency: higher value). The bits are as you see them in the code without any inversion.

Ah, OK - that doesn't have any effect on the actual encoding, as it is "distance based" (1 being a long on followed be a short off and vice vesa)

axel1202, 2021-01-11:
> Hi, you should be able to feed byte strings to rtl_433 with the help of -y parameter. Then you can pass the same payload to > both code bases. Then you can instrument the code to locate the issue.

To be honest, I don't understand your advise. As shown above, the paylod is generated properly and matches the payload sent by the GT-WT-03 sensor I already have here. The problem I am having is to calculate the proper checksum - as you can see, I copied your function and the calculated checksum differs. Un the other hand side, rtl_433 accepts the original message and checksum from the sensor itself. So my guess is, that I didn't understand the checksum calculation :-)

But certainly I now tried deeper analysis of what rtl_433 receives.

So currently my sensor transmitted 3.1°C, 91%, CH 1, Battery OK (0), no button pressed (0), ID 184 and rtl_433 reads:

```
[00] {41} 47 a4 ff e0 aa 80 : 01000111 10100100 11111111 11100000 10101010 1

```
then a modelled this signal in my software and sent it. rtl_433 reads:
```
[00] {41} 47 a4 ff e0 87 80 : 01000111 10100100 11111111 11100000 10000111 1
                     ^^ :-(
```
Obviously, rtl_433 inverted everything, but it does so in both cases and receives exactly, what my debug output said:

```
0xB8 (0x47)
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
0x5B (0xA4)
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
0x00 (0xFF)
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
0x1F (0xE0)
0x00 0x80 0x40 0x20 0x10 0x88 0xC4 0x62 
chksum = 0x78 (0x87)

```

Feeding these strings back to rtl_433 using the -y option, unsurprisingly didn't show any other behaviour - the demodulated signal string of the original worked just fine, the generated one didn't. In the end the result is the same: The checksum doesn't match. 

So obviously I still don't know, how to properly calculate this checksum. :-(

Anyone? :-) ( I am desperated... if 2021 starts like this...) :-)

merbanan, 2021-01-11:
My point is that you can add printfs to both code bases and trace all the parameters (all inputs and every internal state change). Send the output of all parameters to a log file and then diff with meld or kdiff. Then you will find what you are missing.

zuckschwerdt, 2021-01-11:
I see now, you printed the code before the invert. The actual buffer you want to process is
`B8 5B 00 1F` and the checksum there is `55` (this is your example `[00] {41} 47 a4 ff e0 aa 80`)

axel1202, 2021-01-11:
> I see now, you printed the code before the invert. The actual buffer you want to process is
> `B8 5B 00 1F` and the checksum there is `55` (this is your example `[00] {41} 47 a4 ff e0 aa 80`)

Nope, that's not it, because either way, the checksum is not properly calculated ;-( The original checksum is 0x55 or 0xAA if you flip the bits. But my program generates 0x78 resp. 0x87 - the lines you were referring to, both were output by rtl_433 - first receiving the singal of the actual GT-WT-03 and then the signal of my transmitter... As yu see - all data are the same, except the checksum... :-)

merbanan, 2021-01-11:
So dump the internal states of the checksum calculation and compare against rtl_433.

zuckschwerdt, 2021-01-11:
Yes, ` 0x78` is correct, but do use the init of `0x2d`!

axel1202, 2021-01-11:
> My point is that you can add printfs to both code bases and trace all the parameters (all inputs and every internal state 
> change). Send the output of all parameters to a log file and then diff with meld or kdiff. Then you will find what you are missing.

OK - got that now - nice idea. I did so and unfortunately both implementations produce the wrong checksum. Which I expected, because as I said I copied your function. So I thought, it must have to do with byte order as I am casting a 32bit unsigned into an array of 8 bit unsigned. So I swapped the bayte order - no change, I flipped the bits within the data bytes - no change, I did both - always the same (wrong) checksum. And btw - also the original function from your code produces the wrong checksum now... 

So - I took anopther closer look and found that after your function returns:

```
    // accept only correct checksum
    int chk = chk_rollbyte(b, 4, 0x3100) ^ b[4] ^ 0x2d;

```
Then I recaped, what zuckschwdrdt suggested and I didn't understand and saw that 0x2D - so when I calculated my checksum starting with a gen of 0x3100, I xor'ed the result with 0x2D and voilá: Works like a charme.

The only remaining problem is: I don't understand that last bit :-) It's not mentioned in the otherwise well understandably source code comments and having no clue of ths, I didn't get the difference between gen, key and init :-)

Anyways - når du är från sverige: takk så mykket! If not: Thanks or Danke for pointing me to the right direction :-)

Best regards

Axel

axel1202, 2021-01-11:
> Yes, ` 0x78` is correct, but do use the init of `0x2d`!

I finally found that line of code in the original code base, that had this - at least for me mysteriuous - 0x2D and as you repeatedly pointed at that 0x2D I did what I just described - it works and I dont completely understand why - see comment above :-)

Danke auf jeden Fall - looft nu jut :-)

merbanan, 2021-01-11:
Varsågod. If the code might be of use to someone else feel free to post a link to it in this thread.

closed 2021-01-11 by merbanan

axel1202, 2021-01-11:
> Varsågod. If the code might be of use to someone else feel free to post a link to it in this thread.

Which code? The code of the generator? Or my checksum calculation? Unfortunately I am not at github, as fpor political reasons ;-) I moved everything to gitlab and even worse can't remember, what of my many email addresses I registered with my actual github account (HerrRausB). But of course I could link in the respective gitlab repo, if that's OK...

zuckschwerdt, 2021-01-11:
Re gen/init/key:
- it's good practice to seed a checksum with some random initial value so all 0's data+check won't be valid. If the sum doesn't feed back into the key then we can just XOR it later (or compare to that fixed value).
- The generator is an intial state to generate a stream of keys (one key per bit of data), for CRC and LFSR the concepts between key and gen are not really distinct.

axel1202, 2021-01-11:
> * it's good practice to seed a checksum with some random initial value so all 0's data+check won't be valid. If the sum 
> doesn't feed back into the key then we can just XOR it later (or compare to that fixed value).

So why 0x2D?

> * The generator is an intial state to generate a stream of keys (one key per bit of data), for CRC and LFSR the concepts between key and gen are not really distinct.

I better don't state, I studied CS ;-) Well - it was at the TU Berlin and thus was less theoretical plus I avoided math as often  as possible :-) Pays back now, it seems ;-) And at almost 57 I am way too old to dig into that now... ;-)

Thanks for helping anyway :-)

zuckschwerdt, 2021-01-11:
> So why 0x2D?

No idea, it's just what the designers used here. They clearly botched the checksum anyway, a LFSR should not reset for each byte and the gen could have been 8 bits if the lower bits were choosen carefully, esp. not 0x00...

> I avoided math as often as possible :-) way too old to dig into that now... ;-)

[Finite fields](https://en.wikipedia.org/wiki/Finite_field) and [Galois theory](https://en.wikipedia.org/wiki/Galois_theory) are a rather elegant way to look at binary numbers :) Maybe look at [The almost impossible chessboard puzzle](https://www.youtube.com/watch?v=as7Gkm7Y7h4) (Matt **and** Grant!) for an entertaining take on this.

axel1202, 2021-01-11:
> [Finite fields](https://en.wikipedia.org/wiki/Finite_field) and [Galois theory](https://en.wikipedia.org/wiki/Galois_theory) are a rather elegant way to look at binary numbers :) Maybe look at [The [...]

I guess, I stay de-math'ed for the rest of my humble life :-)