merbanan/rtl_433 #1608: Simplisafe phantom calls despite closed sensor?

lucifudge, 2021-01-12:
I'm pushing to my MQTT server for alerts (alert on "Sensor Open" and "Sensor Closed" messages) and constantly getting phantom pings despite no activity happening at the sensors.

Do these sensors typically phone home to the base station periodically for various debugging/refresh/battery/pinging reasons when there is no activity?

lucifudge, 2021-01-15:
I'm pushing to my MQTT server and constantly getting phantom messages when in a closed state, despite no open/close activity happening at the sensors. My Simplisafe base station is not chiming when rtl_433 sees the phantom messages. If the sensor is left in the state "Closed" (IE; window/door closed), changes can be detected by waiting for it to change to an Open state, but this only works for the door/window sensors.

We run into problems when some Simplisafe sensors, specifically the motion sensor, only report "Closed" as it's beacon when sensing motion. This means tons of false positives presumably just by being idle. 

Since my Simplisafe homebase is not chiming, I do not believe the problem is faulty placement of sensors or sensors breaking position, but a fault of rtl_433 readout.

Example of a message from an already closed sensor that had been closed for some time;
rtl_433 -q -R 102 -F "mqtt://192.168.1.30:1883,user="user",pass="pass",events=Simplisafe"

Message 2 received on Simplisafe at 9:42 PM:

{
    "time": "2021-01-11 21:42:23",
    "model": "SimpliSafe Sensor",
    "device": "1P7U5",
    "seq": 21,
    "state": 2,
    "extradata": "Contact Closed"
}

QoS: 0 - Retain: false

merbanan, 2021-01-15:
Hi, observe the output without pushing to the MQTT broker. The describe the user scenario that you consider is an error. I don't fully understand your problem. One thing I am quite sure of is that a signal i transmitted but rtl_433 might not interpret it correctly. Please elaborate so we can figure out the issue.

zuckschwerdt, 2021-01-15:
It's my understanding that the SimpliSafe sensors are better implemented than most. I would expect them to send periodic messages -- that's part of a robust system. It might be a good idea to track the periodic messages in your system and alert when sensors go quiet. Generally you want some state logic, just `rtl_433 -> MQTT -> Alert` isn't a complete setup. Perhaps look into Node-RED.

lucifudge, 2021-01-15:
> It's my understanding that the SimpliSafe sensors are better implemented than most. I would expect them to send periodic messages -- that's part of a robust system. It might be a good idea to track [...]

There doesn't seem to be any rhyme or reason as to when the transmission occur - could this be some sort of inactivity ping check I need to track via time lapsed since the last legitimate open/close transmission? These transmissions appear three at a time in rtl_433's output, just as they do in my MQTT broker.

With the door/window sensors, they have an "Open" and "Closed" state, which I can apply to/from logic. The motion sensor only reports "Closed" when motion is sensed. Because of this, I don't know what sort of state logic I could implement to differentiate a legitimate open/close call with these random pings.

merbanan, 2021-01-15:
Ok, the PIR sensor maybe should not set the Closed state. But instead report Motion detected. Please post the command line output when talking about this. It is hard to discuss this without it.

merbanan, 2021-01-15:
Ok the actual code does not know what type a sensor is. That is why it works like it does. To fix this we need to figure out how to identify the sensor type. We need sensor readings from all the sensors you have. Can you supply that?

lucifudge, 2021-01-15:
> 
> 
> Ok the actual code does not know what type a sensor is. That is why it works like it does. To fix this we need to figure out how to identify the sensor type. We need sensor readings from all the sensors you have. Can you supply that?

https://pastebin.com/DpGWs572 are my console outputs from the motion sensor.

I would be happy to provide those sensor readings. What command should I pass to rtl_433 in order to supply what is needed?

merbanan, 2021-01-15:
Just post the rtl_433 output when triggering the sensor. It looks like the only way to know what type of sensor is transmitting is by looking at the device id.

lucifudge, 2021-01-16:
Are full logs necessary? Just want to make sure before I go running around my house opening every door/window to debug.

Here are my serials;

Door/entry sensors:
"1P7U5" "1M6WC" "1M79X" "1M7EQ" "1M6YR" "1P6QQ" "1P9ZU" "1PD1D" "1PAW4" "1PE8L" "1PBWF" "1P8HB" "1P8ES" "1P8JN" "1P7T3" 

Motion sensors:
"1AN9M"

merbanan, 2021-01-17:
Do you register the type of a sensor in the Simplisafe base station? The idea was that from the id infer the type of sensor but from this list I don't see any pattern. In the end it looks like you need a middle layer for logic and the decoder is as complete as it is possible to do it.

The aim for rtl_433 is to as raw as possible translate the rf-signal to a data object to be used for further processing if needed. Adding states and event logic has to happen in another layer. I myself uses Node-red which is really nice to work with.

closed 2021-04-14 by lucifudge