merbanan/rtl_433 #1609: Can't recive Data from Bresser 5 in 1 Sensor

nanotux1, 2021-01-12:
I bought this weather station: https://www.aldi-nord.de/aldi-liefert/p/5-in-1-wetterstation-bresser-1008343/
 and thought it would be supported from rtl_433. But I cannot recive Data from the sensor.
The rtl_433 installation works, because I recive Data from a GT-WT03 sensor.
The 5 in 1 Bresser sensor works correct, because I can see the Data at the Display of the weather station.

Please help.

zuckschwerdt, 2021-01-12:
Check if the transmission is in the right frequency range, use CubicSDR, Gqrx or SDRangel.
As poormans substitude record a sample (`-w file_433.92M_250k.cu -T 120`) and drop on https://triq.org/iqs/

peteakalad, 2021-01-12:
Also try running with -Y classic
e.g.
`rtl_433 -f 868300000 -Y classic -s 250k`

nanotux1, 2021-01-13:
@peteakalad rtl_433 -f 868300000 -Y classic -s 250k don't works

the in the rtl_433 implemented Bresser 5 in 1 sensor suggested, that it transmits on 868.3 MHz

If I start the rtl_433 application only with ./rtl_433 scan it also on 868.3 MHz or only on 433 MHz?

@zuckschwerdt: if I save a sample with (-w file_433.92M_250k.cu -T 120) is the correct application call 
./rtl_433 -w file_433.92M_250k.cu -T 120 
or
 ./rtl_433 -f 868300000 -s 250k -w file_433.92M_250k.cu -T 120
or
 ./rtl_433 -f 868300000 -Y classic -s 250k -w file_433.92M_250k.cu -T 120

?

zuckschwerdt, 2021-01-13:
> If I start the rtl_433 application only with ./rtl_433 scan it also on 868.3 MHz or only on 433 MHz?

The default is only 433.92M.

> ./rtl_433 -w file_433.92M_250k.cu8 -T 120

That will save a 433.92M sample.

> ./rtl_433 -f 868.3M -s 250k -w file_868.3M_250k.cu8 -T 120

That will save a 868.3M sample.

> ./rtl_433 -f 868.3M -Y classic -s 250k

That (mode = classic) might help to decode the sensor.

BeanBag112233, 2021-01-13:
Hi, I too am having problems with a Bresser 5 in 1 purchased from Amazon just before Christmas. If I dont specify the data format, rtl_433 identifies it as the 6 in 1. If I specify the format with 119, it fails to report any data.
`
    pi@PiWeewx:~/ $ **sudo /usr/local/bin/rtl_433 -Y classic -F json -f 868.2M -R 119 -s 1M**
    rtl_433 version 20.11-18-gb9de8421 branch master at 202101130815 inputs file rtl_tcp RTL-SDR
   ...
    Registered 1 out of 176 device decoding protocols [ 119 ]
    Found Rafael Micro R820T tuner
    Exact sample rate is: 1000000.026491 Hz
    [R82XX] PLL not locked!
    Sample rate set to 1000000 S/s.
   Tuner gain set to Auto.
    Tuned to 868.200MHz.
    ^CSignal caught, exiting!

`

.
    pi@PiWeewx:~/rtl_433/build $ **sudo /usr/local/bin/rtl_433 -Y classic -F json -f 868.2M -s 1M**
    rtl_433 version 20.11-18-gb9de8421 branch master at 202101130815 inputs file rtl_tcp RTL-SDR
...
   Sample rate set to 1000000 S/s.
   Tuner gain set to Auto.
   Tuned to 868.200MHz.
{"time" : "2021-01-13 14:02:01", "model" : "**Bresser-6in1**", "id" : 541070586, "channel" : 0, "battery_ok" : 1, "temperature_C" : 11.700, "humidity" : 82, "sensor_type" : 1, "wind_max_m_s" : 2.200, "wind_avg_m_s" : 2.000, "wind_dir_deg" : 248, "mic" : "CRC"}
{"time" : "2021-01-13 14:02:13", "model" : "**Bresser-6in1**", "id" : 541070586, "channel" : 0, "battery_ok" : 1, "sensor_type" : 1, "wind_max_m_s" : 2.600, "wind_avg_m_s" : 2.600, "wind_dir_deg" : 270, "rain_mm" : 4.800, "mic" : "CRC"}
{"time" : "2021-01-13 14:02:25", "model" : "**Bresser-6in1**", "id" : 541070586, "channel" : 0, "battery_ok" : 1, "temperature_C" : 11.700, "humidity" : 83, "sensor_type" : 1, "wind_max_m_s" : 2.600, "wind_avg_m_s" : 2.300, "wind_dir_deg" : 248, "mic" : "CRC"}

Hope this helps. let me know if you require any data collection.

Thanks for all your hard work.

zuckschwerdt, 2021-01-13:
More recent 5in1 branded devices use the 6in1 protocol, also the 7in1 indoor sensors use 6in1. It's a (labeling) mess.

rwpjones, 2021-01-14:
Dear Readers

I had teh same issue - I am recieving output from the recently purchased Bresser 5in1 on Pi 3 console labelled as Bresser 6in1. 

My question is how do I post this to a text file (for now) that I can read in a text editor? At the moment posted output appears as gobbledygook (?binary?)

 sudo rtl_433 -f 868.272 -F csv -W ~/shared/rtl_433_test.csv

Screen:
Found Fitipower FC0012 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
{"time" : "2021-01-14 09:42:25.537785", "protocol" : 172, "model" : "Bresser-6in1", "id" : 432017282, "channel" : 0, "battery_ok" : 1, "temperature_C" : 2.700, "humidity" : 94, "sensor_type" : 1, "nd_max_m_s" : 0.000, "wind_avg_m_s" : 0.000, "wind_dir_deg" : 135, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.193, "freq2" : 868.327, "rssi" : -4.167, "snr" : 10.927, "noise" : -15.094}
2021-01-14 09:42:25.537785,,,172,FSK,,868.193,868.327,-4.167,10.927,-15.094,Bresser-6in1,,432017282,0,1,2.700,CRC,,,94,,,,,,,,,,,0.000,0.000,135,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,1,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

rtl_433_test.csv
"‘{V>Ÿf¾w©¹›‘S• ˜tlaˆ\ŒuŸZ <`U`X™Iq‹Ä—z˜ž~„\£c_bfÈ:y¡¢°a§aDHfjq+ŠÇn˜q „Wx  ............................................"

It's a mystery.

Thanks for advice

Richard

merbanan, 2021-01-14:
Use rtl_433 -f 868.272 -F csv >~/shared/rtl_433_test.csv

closed 2021-01-14 by merbanan

nanotux1, 2021-01-15:
It works. The distance between sensor and reciver was to long.

Is there a possibility with only one tuner with rtl_433 to receive in parallel on 433 and 868 Mhz?

zuckschwerdt, 2021-01-15:
Use `rtl_433 -f 433.92M -f 868.272 -F csv:$HOME/shared/rtl_433_test.csv`
That will change the frequency every 10 minutes (adjust with `-H <seconds>`).