merbanan/rtl_433 #1611: Ubuntu package reports version unknown

FosterCL1, 2021-01-14:
After installing RTL-433 on Ubuntu (apt install) I noticed that the version string reports as "Unknown"

Ideally it would coincide with a git tag.  Specifically the example.conf file doesn't work, since there are more protocols in the example conf than in the specific Ubuntu version.

$ rtl_433 -V                                       
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR          
Use -h for usage help and see https://triq.org/ for documentation.

zuckschwerdt, 2021-01-14:
I don't know about the example.conf. In source it always matches. What went wrong in packaging?

The version is extracted from the Git tag, we don't check a version tag into source. Ideas welcome -- but don't expect us to manually carry a version number in source ;)

FosterCL1, 2021-01-14:
Fair enough. I definitely understand the aversion to dragging the version number along. Especially since I see how you've grabbed the version from the git tag directly. I like it!

I don't know what the process of submitting rtl-433 to Ubuntu / Debian looks like, so I don't have much in the way of suggestions. My guess is a source tar is submitted instead of a git tag, so there'd have to be some manual intervention there.  Plus different versions of Ubuntu, updates, etc.

zuckschwerdt, 2021-01-14:
Porting the Travis builds to Actions is on my to do list, maybe I can add source tar gz creation with an added version-tag file.

FosterCL1, 2021-01-19:
That makes sense. As I said, it is a nice-to-have. Thanks for all your work on maintaining a great Open Source project!

zuckschwerdt, 2021-02-04:
With f3917ef the release version number (for tarball builds) is now parsed from the Changelog.
This reuses the info we already have with no need for additional version tags. I feel rather clever now, also I expect this to break anyday ;)

closed 2021-02-04 by zuckschwerdt