merbanan/rtl_433 #1615: Amazon Basics Meat Thermometer

twieland, 2021-01-17:
Hi!

I recently purchased this meat thermometer from Amazon because it was only €10: https://www.amazon.de/gp/product/B07VYKYMG3/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

I'm new to this topic, but I've been playing around with it a bit and managed to receive the signal using a Sonoff RF Bridge with Tasmota/Portisch firmware. Trying to decode the data I stumbled over this project and looked at the results from e.g. here https://triq.org/pdv/#AAB1030230041012AC0011001110010000000000000000000010010011000000000011000010010000001000000000000000000255

The example above is for 22°C, and that's for 24°C: https://triq.org/pdv/#AAB1030226041A12B60011100110010000000000000000000010010011000000000011001100000000001000000000000000000255

Now I'm not sure where to go from here. I don't know what to look for in the data. All the sensor is sending should be a single temperature value every 12 seconds, but I'm not sure what the protocol is and whether it transmit °C, °F or the raw analog value (these thermometers are based on changing resistance at different temperatures) that's then translated by the receiver.

Does anyone have experience with thermometers like this or an idea how to proceed?

zuckschwerdt, 2021-01-28:
The data looks like valid Manchester code. Open the links, switch the slicer to "MC", copy the codes
```
69 FF CD FB 9E 00 00
49 FF CD FB 7E 00 00
```
to [BitBench](https://triq.net/bitbench?c=69FFCDFB9E0000&c=49FFCDFB7E0000&f=8h%20%0A8d%20%0A.&i=true) and play around with the shift and invert to see where the data fields are.
Likely you need to let the sensor heat up or cool down slowly and record all values, that way you can see a field in the data ramping up or down easily.

twieland, 2021-01-30:
Thanks, that already helps a lot! However, I didn't have any luck so far. Here's the data I used with [BitBench ](https://triq.net/bitbench?c=63%20FF%20EB%20F7%207C%2000%20%20%20%20&c=23%20FF%20EB%20F7%203C%2000&c=31%20FF%20F5%20FB%203E%2000%20&c=5F%20FF%20EB%20F3%207C%2000&c=0F%20FF%20F5%20F9%201E%2000&c=6F%20FF%20AF%20BF%20F0%2000&f=8h%20%0A8%20d%0A.&cw=4):
`
22°C: 63 FF EB F7 7C 00    
23°C: 23 FF EB F7 3C 00
26°C: 31 FF F5 FB 3E 00 
32°C: 5F FF EB F3 7C 00
37°C: 0F FF F5 F9 1E 00
40°C: 6F FF AF BF F0 00
`

I played around with Shift and Invert but couldn't find a pattern yet. I'm still not sure if the sender sends already the converted temperature or if it's some kind of raw measurement. Anyway, maybe this helps someone in the same situation.

zuckschwerdt, 2021-01-30:
Just invert and align were needed :) See [this BitBench](https://triq.net/bitbench?c=63%20FF%20EB%20F7%207C%2000%20%5B22%20C%5D&c=23%20FF%20EB%20F7%203C%2000%20%5B23%20C%5D&c=31%20FF%20F5%20FB%203E%2000%20%5B26%20C%5D&c=5F%20FF%20EB%20F3%207C%2000%20%5B32%20C%5D&c=0F%20FF%20F5%20F9%201E%2000%20%5B37%20C%5D&c=6F%20FF%20AF%20BF%20F0%2000%20%5B40%20C%5D&f=12h%20&a=Preamble&m=ffa&i=true).

twieland, 2021-01-31:
This is awesome, thanks so much!
One more question: I managed to read the data from MQTT using halfbakery's branch, but I'm struggling with setting the other CLI parameters to reproduce the decoding above. Can you help me with that?

twieland, 2021-02-07:
I've been playing around quite a bit with the MQTT branch but can't get the settings to work: `rtl_433 -F "mqtt://localhost:1883,events=rtl_433" -F kv  -r mqtt/rfraw:tele/sonoff-bridge/RESULT -X name=meat,modulation=OOK_MC_ZEROBIT,short=560,reset=1260,long=1050,sync=1260`

zuckschwerdt, 2021-02-07:
The old rfraw format (all number in the later part of the string below are 0-7) wasn't read properly. It's now fixed. This could work:
`rtl_433 -R 0 -X 'name=meat,modulation=OOK_MC_ZEROBIT,short=550,long=550,reset=1500,invert,preamble=ff'`

twieland, 2021-02-08:
Thanks so much for pointing me in the right direction! However, for whatever reason I didn't require invert anymore and therefore the inverted preamble:
`rtl_433 -R 0 -X 'name=meat,modulation=OOK_MC_ZEROBIT,short=550,long=1550,reset=1500,preamble=50' -y "AA B1 03 042E 021C 12D4 000110011110111111111111110011000000111111111100111100111111110111111111111111111
2 55"`

I'm able to use node-red now to call rtl_433, extract the temperature and publish it to mqtt.

netspoon, 2021-03-10:
Hi,

I also have one of these devices, but whenever the batteries are removed it must reset some value present in the data.

All of these examples are for 19C, captured after the batteries were put back each time:

```
[00] { 8} ff                : 11111111 
[01] {44} 6f fd df ed df f0 : 01101111 11111101 11011111 11101101 11011111 1111
[02] {44} 6f fd df ed df f0 : 01101111 11111101 11011111 11101101 11011111 1111
[03] {44} 6f fd df ed df f0 : 01101111 11111101 11011111 11101101 11011111 1111
[04] {44} 6f fd df ed df f0 : 01101111 11111101 11011111 11101101 11011111 1111
[05] {35} 6f fd df ed c0    : 01101111 11111101 11011111 11101101 110

[00] { 8} ff                : 11111111 
[01] {42} a7 fa df b7 7f c0 : 10100111 11111010 11011111 10110111 01111111 11
[02] {42} a7 fa df b7 7f c0 : 10100111 11111010 11011111 10110111 01111111 11
[03] {42} a7 fa df b7 7f c0 : 10100111 11111010 11011111 10110111 01111111 11
[04] {42} a7 fa df b7 7f c0 : 10100111 11111010 11011111 10110111 01111111 11
[05] {33} a7 fa df b7 00    : 10100111 11111010 11011111 10110111 0

[00] { 8} ff                : 11111111 
[01] {42} 27 fd 7f b7 7f c0 : 00100111 11111101 01111111 10110111 01111111 11
[02] {42} 27 fd 7f b7 7f c0 : 00100111 11111101 01111111 10110111 01111111 11
[03] {42} 27 fd 7f b7 7f c0 : 00100111 11111101 01111111 10110111 01111111 11
[04] {42} 27 fd 7f b7 7f c0 : 00100111 11111101 01111111 10110111 01111111 11
[05] {33} 27 fd 7f b7 00    : 00100111 11111101 01111111 10110111 0

[00] { 8} ff                : 11111111 
[01] {44} 6b fe ff ed df f0 : 01101011 11111110 11111111 11101101 11011111 1111
[02] {44} 6b fe ff ed df f0 : 01101011 11111110 11111111 11101101 11011111 1111
[03] {44} 6b fe ff ed df f0 : 01101011 11111110 11111111 11101101 11011111 1111
[04] {44} 6b fe ff ed df f0 : 01101011 11111110 11111111 11101101 11011111 1111
[05] {35} 6b fe ff ed c0    : 01101011 11111110 11111111 11101101 110

[00] { 8} ff                : 11111111 
[01] {44} bd fd bf ed df f0 : 10111101 11111101 10111111 11101101 11011111 1111
[02] {44} bd fd bf ed df f0 : 10111101 11111101 10111111 11101101 11011111 1111
[03] {44} bd fd bf ed df f0 : 10111101 11111101 10111111 11101101 11011111 1111
[04] {44} bd fd bf ed df f0 : 10111101 11111101 10111111 11101101 11011111 1111
[05] {35} bd fd bf ed c0    : 10111101 11111101 10111111 11101101 110

[00] { 8} ff                : 11111111 
[01] {44} 4f ff df ed df f0 : 01001111 11111111 11011111 11101101 11011111 1111
[02] {44} 4f ff df ed df f0 : 01001111 11111111 11011111 11101101 11011111 1111
[03] {44} 4f ff df ed df f0 : 01001111 11111111 11011111 11101101 11011111 1111
[04] {44} 4f ff df ed df f0 : 01001111 11111111 11011111 11101101 11011111 1111
[05] {35} 4f ff df ed c0    : 01001111 11111111 11011111 11101101 110
```

Any wisdom here would be appreciated...

grb.

netspoon, 2021-03-11:
Hi,

Hope this is the correct file:

[g003_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/6121293/g003_433.92M_250k.cu8.zip)

Temperature is 14C on the display unit

```
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB0120401020C041C12802A34808080808080808255+AAB0350404020C041C12802A349080818091808080808080808080919180809081808080808080919180808080809080808080808080808255+AAB02C0401020C041C12802A349080818091808080808080808080919180809081808080808080919180808080809355
Attempting demodulation... short_width: 548, long_width: 1064, reset_limit: 4744, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=548,l=1064,r=4744,g=1112,t=206,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 6 
[00] { 8} ff                : 11111111 
[01] {43} 77 fc df cf bf e0 : 01110111 11111100 11011111 11001111 10111111 111
[02] {43} 77 fc df cf bf e0 : 01110111 11111100 11011111 11001111 10111111 111
[03] {43} 77 fc df cf bf e0 : 01110111 11111100 11011111 11001111 10111111 111
[04] {43} 77 fc df cf bf e0 : 01110111 11111100 11011111 11001111 10111111 111
[05] {34} 77 fc df cf 80    : 01110111 11111100 11011111 11001111 10
```

netspoon, 2021-03-12:
looks like shifting by 25 bits reveals the temperature, but the preamble changes when the batteries are replaced

```
3b007a80c07fe   [18C]
33007a80b87fe   [17C]
52802200b07fe   [16C]
7f804f00b07fe   [16C]
67003680b07fe   [16C]
```
shifted 13 bits left (batteries replaced for lines 3,4 & 5):

[image]

merbanan, 2021-03-12:
Grab a couple more signals with varying conditions, then we should be able to take into account the changes in the preamble.

merbanan, 2021-03-13:
Decoder is available here https://github.com/merbanan/rtl_433/pull/1671. Some more samples would be nice. One byte is still unknown.

merbanan, 2021-03-13:
Support merged in b101a4b. A few more signal samples would be nice.

closed 2021-03-13 by merbanan

netspoon, 2021-03-15:
Thanks for this...  I'll see if I can grab some more samples this week.

twieland, 2021-03-21:
Thanks for working on this!

However, I can't seem to get it working:
`rtl_433 -R 181 -y "AA B1 03 0230 0410 12CA 0011000000110000000000000000000011100000010000000000100100001100000010000000000000000002 55"
rtl_433 version 20.11-99-g090f1169 branch master at 202103202118 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/openhabian/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 183 device decoding protocols [ 181 ]`

What am I missing?