merbanan/rtl_433 #1619: Doorbell - flexible decoder on first row

netspoon, 2021-01-20:
HI,

I've a Byron doorbell, which seems to alternate between two values on consecutive presses, otherwise working well:

```
rtl_433 -F json  -R 0 -X 'n=doorbell,m=OOK_PWM,s=532,l=1596,r=1596,match={33}0x7fa82a6c8'  -X 'n=doorbell2,m=OOK_PWM,s=532,l=1596,r=1596,match={33}0x02c47a2c8'
```

Output from 2 button pushes (same button):

```
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:51", "model" : "doorbell2", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "02c47a2c8"}], "codes" : ["{33}02c47a2c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
{"time" : "2021-01-20 15:26:54", "model" : "doorbell", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 33, "data" : "7fa82a6c8"}], "codes" : ["{33}7fa82a6c8"]}
```

But is it possible to get it to 'detect' on the first row, otherwise there is a couple of second delay before the push is recognised ?

Many thanks,

grb.

zuckschwerdt, 2021-01-20:
rtl_433 will always wait for a complete transmission (or maximum time) before processing decoders. If the packets (rows) are too close together then that will stall detection.
Try the `-A` analyzer to see the timing (gap) between packets.

zuckschwerdt, 2021-01-20:
Yes, the longest gap with the fewest counts is the packet distance, 7ms:

>  [ 2] count:   19,  width: 7188 us [7184;7200]  (1797 S)

Use a flex decoder with `-X 'n=doorbell,m=OOK_PWM,s=500,l=1600,r=2000'`

netspoon, 2021-01-20:
Does this suggest the packets are 'too close' - so it will only detect once all 19 rows are received ?

zuckschwerdt, 2021-01-20:
`r=7204` grabs all packets in one go, with `r=2000` each packet should be on it's own.

netspoon, 2021-01-25:
Thanks, already using m=OOK_PWM,s=532,l=1596,r=1596 which works well...  with a slightly delay, as #1

Not sure I understand the significance of the 'r' value, but shorter should take less packets before decoding ?

zuckschwerdt, 2021-01-25:
`r`= `reset`, maximum gap length before stopping to read a packet.

closed 2021-01-25 by zuckschwerdt