merbanan/rtl_433 #1625: Question about average wind speed for Fine Offset WH24 device

epatey, 2021-01-24:
I'm just getting into SDR, so let me start off with a thank you for rtl_433.

I've got an Ambient Weather WS-2902C which appears to have Fine Offset  WH24 guts.

I'm confused by two things about `wind_avg_m_s`. First, even though the code documents this as meters per second, it sure seems like it's mph. When I compare the `windd_avg_m_s` from `rtl_433` with the mph wind speed display on my 2902C console, the values pretty much match each time a new packet is processed each 16s. It sure seems that my sensor unit is sending mph data.

The _pretty much_ part is the second question. The data on the console isn't an exact match with the `rtl_433` output. It's often off by between about 0.1 and 0.5 mph. Obviously, that's not a big deal, but it's a bit confounding. I can minimize the difference by bumping the scale factor from 1.12 up to something like 1.142, but I can't do anything to get it to match what the console shows.

FWIW, here's an example of the `rtl_433` output.
```
time      : 2021-01-24 09:11:04
model     : Fineoffset-WH24                        ID        : 46
Battery   : 1            Temperature: -6.3 C       Humidity  : 47 %          Wind direction: 321       Wind speed: 17.5 m/s
Gust speed: 21.3 m/s     Rainfall  : 46.2 mm       UV        : 350           UVI       : 0             Light     : 20165.0 lux
Integrity : CRC

```

zuckschwerdt, 2021-01-24:
Wind speed for that protocol isn't any particular unit and also varies by specific device.
https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c#L268
Iirc, someone did get specs from FineOffset (they seem rather friendly) but those don't match well either.
If there is any way to distinguish your WS-2902C from the WH24 then we could add another factor. Otherwise you'd need to hack the source. Preferably send a PR with a commented line to track this and hint other user.

zuckschwerdt, 2021-01-28:
Reopen if there is some info on a better calibration or conversion.

closed 2021-01-28 by zuckschwerdt

epatey, 2021-01-28:
Thanks for the answer @zuckschwerdt. I have to deal with real life during the week, but I'll look into distinguishing my Ambient Weather unit from other FineOffset devices over the upcoming weekends.