merbanan/rtl_433 #1626: Siemens Heat Cost Allocator data decoding

ruabo, 2021-01-24:
Using protocol 105 for Siemens WHE542-D291S heat cost allocator, output is something like this (launching with params -R 105 -f 868.3M):

[image]

Is there a way to understand data ? I'd like to read heat count reading.

merbanan, 2021-01-25:
Can you record a sample signal and upload to the issue? rtl_433 -R 105 -f 868.3M -S known

merbanan, 2021-01-25:
@jedi7 might be interested in this also.

jedi7, 2021-01-25:
When I see the data, then I think, there is no measurement. You will need active communication (transmit ask for data). Periodic measurement transmitting is imho only in mode T1.

merbanan, 2021-01-25:
@jedi7 I can configure some of my devices to transmit in mode S and they then send the same telegram as over mode T. Anyway I can't decode the raw data in my head. It looks like more then 3 extra parameters to me though.

merbanan, 2021-01-25:
https://github.com/weetmuts/wmbusmeters/blob/master/src/meter_whe46x.cc is similar.

ruabo, 2021-01-25:
> Can you record a sample signal and upload to the issue? rtl_433 -R 105 -f 868.3M -S known

I think so; please find the attached file

[image]

[g079_868.3M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/5869799/g079_868.3M_1000k.cu8.zip)

jedi7, 2021-01-26:
As you can see, there is really only time/date, no measurement. So active measure will be needed (or reconfigure the meter)

**decoding dif_coding=11, vif=0x00, vif_uam=0x6e, dif_ff=0, dif_sn=0, dif_su=0, b[3]=0x4B, b[2]=0x00,b[1]=0x03, b[0]=0x04, val=0**

**decoding dif_coding=11, vif=0x00, vif_uam=0x6e, dif_ff=0, dif_sn=1, dif_su=0, b[3]=0x42, b[2]=0x00,b[1]=0x03, b[0]=0x66, val=0**

**decoding dif_coding=2, vif=0x00, vif_uam=0x6c, dif_ff=0, dif_sn=1, dif_su=0, b[3]=0x6C, b[2]=0x32,b[1]=0x29, b[0]=0x9E, val=10654**

**decoding dif_coding=2, vif=0x00, vif_uam=0x6c, dif_ff=3, dif_sn=0, dif_su=0, b[3]=0x6D, b[2]=0x04,b[1]=0xFF, b[0]=0xFF, val=-1**

**decoding dif_coding=4, vif=0x00, vif_uam=0x6d, dif_ff=0, dif_sn=0, dif_su=0, b[3]=0x21, b[2]=0xB9,b[1]=0x16, b[0]=0x14, val=565777940**

merbanan, 2021-01-26:
Well isn't decoding of vif_uam=0x6e missing?

merbanan, 2021-01-26:
// E110 1110	Units for H.C.A.	 	dimensionless

merbanan, 2021-01-26:
0b6e040300 4b6e660300 seems to be missing from the output at least. I agree that things are missing but this should be represented in the output.

jedi7, 2021-01-26:
yes, them implementation is missing, but it is only units if I understand. Anyway it is val=0.

merbanan, 2021-01-26:
Can you add that and then we can close this issue? There is nothing more we can do.

merbanan, 2021-01-26:
And the error date is 2127-15-31 wmbusmeters and 127-15-31 in rtl_433. Is that an issue also?

jedi7, 2021-01-26:
Hmm, I can add test for wrong month and then hide the record if invalid.
Regarding 0x6e I'm not sure what to do here. I'm not able to find documentation what should be here.

merbanan, 2021-01-26:
Regarding 0x6E name it HCA_# and expose the raw decimal value. And regarding the date just output "invalid/error" don't hide anything.

merbanan, 2021-01-26:
Output of attached signal.

```
time      : @0.074968s
model     : Wireless-MBus Mode      : S            Manufacturer: LSE         ID        : 91835132
Version   : 52           Device Type: 0x08         Device Type String: Heat Cost Allocator             Control   : 0x44
Data Length: 39          Data      : 244465323251839134087a4f0000000b6e0403004b6e660300426c9e29326cffff046d1416b921dd2f
Integrity : CRC
Control Info: 0x7A       Access number: 0x4F       Device Type: 0x00         Configuration Word: 0x0000     Date     : 20-09-30      Err Date  : 127-15-31      TimeDate : 21-01-25T22:20:00

```

ruabo, 2021-01-26:
> As you can see, there is really only time/date, no measurement. So active communication will be needed (or reconfigure the meter)
@jedi7 
Sorry, I'm I totally lost here; where can I find docs about all those strings ?

merbanan, 2021-01-26:
@ruabo I think this device needs a trigger signal to send more data or the device send 2 or more types of telegrams. Keep logging this for a while and see if something changes in the output..

ruabo, 2021-01-26:
@merbanan is there a way to capture/log only this specific device ?

merbanan, 2021-01-26:
Try with -F csv and then just grep for the specific id and you should get lines for only that device. Or open the file as a spreadsheet and sort it there.

merbanan, 2021-01-26:
rtl_433 -R 105 -f 868.3M -F csv >logs.csv

weetmuts, 2021-01-26:
The qcaloric meter can decode the parts in the telegram merbanan provided above (time      : @0.074968s...etc)
Unfortunately the telegram is truncated? The difvif decoder complains about the last field.

This is the content, and it looks like ot extracts the HCA value properly...

(qcaloric) 0f: 0B dif (6 digit BCD Instantaneous value)
(qcaloric) 10: 6E vif (Units for H.C.A.)
(qcaloric) 11: * 040300 current consumption (304.000000 hca)
(qcaloric) 14: 4B dif (6 digit BCD Instantaneous value storagenr=1)
(qcaloric) 15: 6E vif (Units for H.C.A.)
(qcaloric) 16: * 660300 consumption at set date (366.000000 hca)
(qcaloric) 19: 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(qcaloric) 1a: 6C vif (Date type G)
(qcaloric) 1b: * 9E29 set date (2020-09-30)
(qcaloric) 1d: 32 dif (16 Bit Integer/Binary Value during error state)
(qcaloric) 1e: 6C vif (Date type G)
(qcaloric) 1f: * FFFF error date (2127-15-31)
(qcaloric) 21: 04 dif (32 Bit Integer/Binary Instantaneous value)
(qcaloric) 22: 6D vif (Date and time type)
(qcaloric) 23: * 1416B921 device datetime (2021-01-25 22:20)
(qcaloric) 27: DD dif (variable length Maximum value storagenr=1)
(qcaloric) 28: 2F dife (subunit=0 tariff=2 storagenr=31)

weetmuts, 2021-01-26:
Please collect more telegrams. Also note that the qcaloric meters sends two completely different telegrams, one with hca data and one without, only meta-data about the meter.

But in the case of the whe4x HCA, it looked like there was no HCA data at all. In merbanans example, there was HCA data. From the comment @jedi7  it seems like at least one telegram from whe5x also lacks hca data.

So is it one of two types of telegrams, or do we need to transmit something to the meter to have it speak up. Waking it up does not seem that likely, since it sends so much data anyway....so apparently it is quite awake.

Or is the meter not configured yet?

ruabo, 2021-01-26:
@

> Please collect more telegrams. Also note that the qcaloric meters sends two completely different telegrams, one with hca data and one without, only meta-data about the meter.
>
> But in the case of the [...]

That device is in an apartment inside a bulding, data are collected by boxes located outside (I can check the box content if necessary), in the common staircases. Devices are running since 3 years ago as billings. I'll collect more data when back home (how many hours ?). If I 'm not wrong, "660300 consumption at set date (366.000000 hca)" is correct and it is last year comsumption memory.

weetmuts, 2021-01-26:
Nice @ruabo , since the telegram decoded ok and you confirmed the value, I have pushed support for the new driver whe5x to wmbuseters. If you build the latest wmbusmeter then you can do:
rtl_433 -F csv -s 1200k -r g079_868.3M_1000k.cu8 2>/dev/null | wmbusmeters stdin:rtl433

And it will print:
Started config rtl433 on stdin listening on any
No meters configured. Printing id:s of all telegrams heard!
(rtl433) dll_len adjusted to 40 from 36. Fix rtl_433? This warning will not be printed again.
Received telegram from: 91835132
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x34
                driver: whe5x

So, it seems that the rtl_433 capture is missing 4 bytes. But it decodes partially anyway, since its not block encrypted.
Specifying the meter id/driver/key:
rtl_433 -F csv -s 1200k -r g079_868.3M_1000k.cu8 2>/dev/null | wmbusmeters --format=json stdin:rtl433 Room whe5x 91835132 NOKEY

gives the output:

{"media":"heat cost allocation","meter":"whe5x","name":"Room","id":"91835132","current_consumption_hca":304,"set_date":"2020-09-30","consumption_at_set_date_hca":366,"set_date_1":"2020-09-30","consumption_at_set_date_1_hca":366,"error_date":"2127-15-31","device_date_time":"2021-01-25 22:20","timestamp":"2021-01-26T16:18:00Z"}

jedi7, 2021-01-26:
ah I see "issue", actually are supported value codings 0..4 I will try add also this code 11

merbanan, 2021-01-26:
I am surprised the crc check is not rejecting the block. Anyway maybe increasing the sample rate will fix the telegram. @ruabo Try recording again and add -s 1600k to the command. rtl_433 -R 105 -s 1600k -f 868.3M -S known

ruabo, 2021-01-27:
@merbanan I had only time to launch a 7 hours scan with that options, I found 5 devices, only one has two records, the others only one, including another scan for the device submitted yesterday.
@weetmuts sorry, I had no time to build the latest wmbusmeter, I'll try as soon as I can.
[records.zip](https://github.com/merbanan/rtl_433/files/5878322/records.zip)

jedi7, 2021-01-27:
should be fixed by PR: https://github.com/merbanan/rtl_433/pull/1630

ruabo, 2021-01-28:
@weetmuts ok, just built wmbusmeters, updated also rtl_433, still using capture from rtl_433 previous build (i'm not home), launched

`rtl_433 -F csv -s 1200k -r g629_868.3M_1600k.cu8 2>/dev/null | wmbusmeters/build/wmbusmeters --format=json stdin:rtl433 Room whe5x 91835132 NOKEY`

received

`Started config rtl433 on stdin listening on any
(rtl433) dll_len adjusted to 40 from 36. Fix rtl_433? This warning will not be printed again.
{"media":"heat cost allocation","meter":"whe5x","name":"Room","id":"91835132","current_consumption_hca":306,"set_date":"2020-09-30","consumption_at_set_date_hca":366,"set_date_1":"2020-09-30","consumption_at_set_date_1_hca":366,"error_date":"2127-15-31","device_date_time":"2021-01-27 05:58","timestamp":"2021-01-28T13:48:35Z"}`

Still complaining of missing bytes despite the increased sample rate. Heat readings are correct.
@merbanan is there any other test needed  ?

ruabo, 2021-01-31:
Please find the attached telegram acquired with the new build (rtl_433 and wmbusmeters). Now the 433_output shows heat readings:

[image]

and

[image]

anyway, the out still complaining about missing bytes, ie:

`
rtl_433 -F csv -s 1200k -r g357_868.3M_1000k.cu8 2>/dev/null | wmbusmeters/build/wmbusmeters --format=json stdin:rtl433 Room whe5x 91835132 NOKEY
Started config rtl433 on stdin listening on any
(rtl433) dll_len adjusted to 40 from 36. Fix rtl_433? This warning will not be printed again.
{"media":"heat cost allocation","meter":"whe5x","name":"Room","id":"91835132","current_consumption_hca":319,"set_date":"2020-09-30","consumption_at_set_date_hca":366,"set_date_1":"2020-09-30","consumption_at_set_date_1_hca":366,"error_date":"2127-15-31","device_date_time":"2021-01-30 22:10","timestamp":"2021-01-31T07:12:44Z"}
`

Any clues on this ?
[Archive.zip](https://github.com/merbanan/rtl_433/files/5899347/Archive.zip)

ruabo, 2021-01-31:
here a new sample of the telegram with the recommended higher sampling rate. @merbanan @weetmuts Please let me know if further tests are required. But no changes, wmbusmeters always complains about missing bytes

[image]

`
rtl_433 -F csv -s 1600k -r g414_868.3M_1600k.cu8 2>/dev/null | wmbusmeters/build/wmbusmeters --format=json stdin:rtl433 Room whe5x 91835132 NOKEY
Started config rtl433 on stdin listening on any
(rtl433) dll_len adjusted to 40 from 36. Fix rtl_433? This warning will not be printed again.
{"media":"heat cost allocation","meter":"whe5x","name":"Room","id":"91835132","current_consumption_hca":322,"set_date":"2020-09-30","consumption_at_set_date_hca":366,"set_date_1":"2020-09-30","consumption_at_set_date_1_hca":366,"error_date":"2127-15-31","device_date_time":"2021-01-31 17:35","timestamp":"2021-01-31T16:44:44Z"}

` 

[132-3.png.zip](https://github.com/merbanan/rtl_433/files/5900418/132-3.png.zip)

merbanan, 2021-01-31:
@ruabo Most likely a missing calculation in the decoder. I might look into it later. It is in a usable state for you right now. I'll keep the ticket open for a week or two.

closed 2021-03-03 by merbanan

algorni, 2022-12-01:
Finally, do we decode the "data" portion of the telegram for those Heat Cost Allocation devices?
It sounds like the data field should be AES128 encrypted? isn't?

merbanan, 2022-12-01:
I you post a datagram we will see if it is encrypted.