merbanan/rtl_433 #1632: wmbus: Kamstrup Multical 21 water meter

xenos1984, 2021-01-27:
I recorded a few samples from two [Kamstrup Multical 21](http://optimatic.ee/en/products/ultrasonic-water-meters/kamstrup-multical-21/) water meters (cold and warm water), which are only partially decoded. The device sends data every 16 seconds, and I took samples while using water, so that the meter reading - if it is in the data - should change. There is quite a bit of information in [the datasheet](http://optimatic.ee/wp-content/uploads/2018/02/MULTICAL%C2%AE-21-Data-Sheet-English.pdf).

Command line: `rtl_433 -f 868950000 -s 1200000 -g 15 -R 104 -M hires -M level -S known`

Sample output:
```
time      : 2021-01-28 00:18:19.663447
model     : Wireless-MBus Mode      : C            Manufacturer: KAM         ID        : 76522329
Version   : 27           Device Type: 0x16         Device Type String: Cold Water                      Control   : 0x44          Data Length: 32           Data      : 23442d2c292352761b168d208152bf9b20470162fe90342234a8feef1abe9da8
Integrity : CRC
Modulation: ASK          Freq      : 0.0 MHz
RSSI      : 0.0 dB       SNR       : 0.0 dB        Noise     : 0.0 dB
```
Recorded data: [wmbus.zip](https://github.com/merbanan/rtl_433/files/5883546/wmbus.zip)

Related issue: #1626

merbanan, 2021-03-07:
Compare the output with the decoded parameters from the https://github.com/weetmuts/wmbusmeters/ project. You can use rtl_433 decoding as input to wmbusmeters. I sort of aim to cover the OMS spec for rtl_433. Covering all the different proprietary wmbus devices would mean I would have to re-implement all the decoding that wmbusmeters has which would be redundant.

closed 2021-03-07 by merbanan

xenos1984, 2021-03-17:
Thanks, that worked out!

Also as a comment to future readers of this issue, it turned out that I had forgotten to use the AES keys, since the meter data is encrypted.

merbanan, 2021-03-17:
Yeah, aes support might come in the future. But I think my time is better served in other areas. Right now wmbusmeters is much better for wmbus support.