merbanan/rtl_433 #1633: Request to support TLS on MQTT

gbsmith, 2021-01-28:
Encrypted MQTT works with OpenHAB, NodeRed, etc. It would be great if RTL_433 could too!

zuckschwerdt, 2021-01-28:
S.a. #1569

zuckschwerdt, 2021-01-28:
Use `-F mqtts` for TLS.
Options are:
- tls_cert: Client certificate to present to the server.
- tls_key: Private key corresponding to the certificate. If tls_cert is set but tls_key is not, tls_cert is used.
- tls_ca_cert: Verify server certificate using this CA bundle. If set to "*" (this is the default for "mqtts"), then TLS is enabled but no cert verification is performed.
- tls_cipher_suites: Colon-delimited list of acceptable cipher suites. Names depend on the library used, for example: ECDH-ECDSA-AES128-GCM-SHA256:DHE-RSA-AES128-SHA256 (OpenSSL) For OpenSSL the list can be obtained by running "openssl ciphers". If unset, a reasonable default is used.
- tls_server_name: Server name verification. If tls_ca_cert is set and the certificate has passed verification, its subject will be verified against this string. By default (if tls_server_name is unset) hostname part of the address will be used. Wildcard matching is supported. A special value of "*" disables name verification.
- tls_psk_identity: PSK identity string. Note: Default list of cipher suites does not include PSK suites, if you want to use PSK you will need to set tls_cipher_suites as well.
- tls_psk_key: PSK key hex string, must be either 32 or 64 hex digits (16 or 32 bytes) for AES-128 or AES-256 respectively.

closed 2021-01-28 by zuckschwerdt

radek-turecki, 2022-02-01:
I have the same problem (Windows + Mosquitto) and I'm trying to put command:

rtl_433-rtlsdr.exe -F mqtts://myinstance.hivemq.cloud:8883,user=myLOGIN,pass=myPASS,retain=0,devices=sensors/rtl_433/

but unfortunately I got error:
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "D:\_Home\Installs\rtl_433-win-msvc-x64-21.12\rtl_433.conf"...
Trying conf file at "C:\Users\Dom\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Publishing MQTT data to 380bc81230e44ccda560a6c0a02034ee.s2.eu.hivemq.cloud port 8883 (TLS)
Publishing device info to MQTT topic "sensors/rtl_433/".
**mqtts (TLS) not available**

How to resolve this problem?

radek-turecki, 2022-02-01:
add:
without "s" - for simple "mqtt://...." the message is:
like above + 
MQTT Connected...
MQTT Connection failed...
MQTT Connected...
MQTT Connection failed...
MQTT Connected...
:-(

zuckschwerdt, 2022-02-01:
The `rtl_433.exe` you are using is build *without TLS*, you want `rtl_433-rtlsdr-tls.exe`.
The version line *with TLS* would look like:
```
rtl_433 version ... inputs file rtl_tcp RTL-SDR SoapySDR with TLS
```

radek-turecki, 2022-02-01:
Great! It works! Thank you very much for your help! :-)

commit 61d24ba9 (2021-01-28): Add optional TLS support to MQTT (closes #1633)