merbanan/rtl_433 #1641: RTL_433 cannot capture the signal of FineOffset WH31 thermometer.

dodamn, 2021-02-01:
I have Fineoffset WH31 thermometers, which uses 868Mhz. 
And I have a RTL SDR device.
```
$ lsusb
Bus 001 Device 007: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
```

The WH31 sends the data about every 1 min. GQRX receives the signal and shows something like the following pictures. The result shows that the frequency is around 868.3Mhz.
[https://imgur.com/iKpMiO9.png](https://imgur.com/iKpMiO9.png)

So I run command like the following.
```
$ rtl_433 -f 868.3M -R 113
rtl_433 version -128-NOTFOUND branch master at 202011290914 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 1 out of 175 device decoding protocols [ 113 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
```

But it doesn't show anything from WH31.
What is the problem?

zuckschwerdt, 2021-02-01:
With 868.3M try `-Y classic -s 250k`
But I wouldn't put that DC spike right into the signal, i.e. use 868.1M or 868.2M, and try `-Y classic`

dodamn, 2021-02-01:
868M, 868.1M, 868.2M, 868.3M with `-Y classic -s 250k`, the all are failed. No result is shown.

But there is something weird. When `rtl_433 -f 868.2M -Y classic` is running, I restarted WH31 by removing the battery and replacing. After then, right result are shown. I retried this command and situation several times. But no result is shown.

It is so weird. Why is this unreliable?

zuckschwerdt, 2021-02-01:
When you grab a signal (try 868.1M, 868.2M, 868.3M)  with `-S all` and drop the .cu8 on https://triq.org/pdv/ does it look ok or is there a DC spike like in your Gqrx waterfall?

dodamn, 2021-02-01:
I ran the command `rtl_433 -f 868.2M -Y classic -S all`
But nothing is saved. So when the above command is running, I restarted WH31 by the battery reinserting. And then the following is happened.
```
pulse_FSK_detect(): Maximum number of pulses reached!
Signal bigger than buffer, signal = 6422528 > buffer 3145728 !!
*** Saving signal to file g001_868.2M_1000k.cu8 (3192734 samples, 3145728 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-01 21:37:42
model     : AmbientWeather-WH31E                   id        : 84
Channel   : 2            Battery   : 1             Temperature: 24.2 C
Humidity  : 46 %         Extra Data: 9200000000    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-02-01 21:37:42
model     : AmbientWeather-WH31E                   id        : 84
Channel   : 2            Battery   : 1             Temperature: 24.2 C
Humidity  : 46 %         Extra Data: 9200000000    Integrity : CRC
*** Saving signal to file g002_868.2M_1000k.cu8 (86286 samples, 262144 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-02-01 21:40:48
model     : AmbientWeather-WH31E                   id        : 84
Channel   : 2            Battery   : 1             Temperature: 24.2 C
Humidity  : 46 %         Extra Data: 9200000000    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-02-01 21:40:48
model     : AmbientWeather-WH31E                   id        : 84
Channel   : 2            Battery   : 1             Temperature: 24.2 C
Humidity  : 46 %         Extra Data: 9200000000    Integrity : CRC
*** Saving signal to file g003_868.2M_1000k.cu8 (86285 samples, 262144 bytes)
```

But after then, no result is shown.

The following picture is about `g002_868.2M_1000k.cu8`.
[https://imgur.com/0yw2tpI.png](https://imgur.com/0yw2tpI.png)

I think it is like Gqrx waterfall.

dodamn, 2021-02-07:
Is anybody there? Please help me.

zuckschwerdt, 2021-02-07:
You need to find out what the difference between "new batteries" and the regular transmission is.
Perhaps try `rtl_433 -f 868.2M -w filename_868.2M_1024k.cu8 -T 60` to save a 60 second file (or use 30 etc seconds for a shorter file). Repeat until you get a file with a transmission, then compare with the "new battery" sample.

merbanan, 2021-03-03:
`Found Fitipower FC0012 tuner`

This tuner needs a gain setting for proper operation. Try -g 20.

closed 2021-03-03 by merbanan