merbanan/rtl_433 #1644: Remove Autotools build

zuckschwerdt, 2021-02-03:
The autoconf/automake build was broken since 14c5b9a (7. October 2019) and nobody complained.
(If RTL-SDR input was enabled, libusb-1.0 was needed but not configured.)
I've fixed it with 9514e16, but I guess there isn't anybody still using this build option.
(also using libusb-1.0 for error messages is now optional with 9514e16)

The Autotools build isn't well maintained and features are missing.
The CMake build also supports traditional Unix Makefiles, as well as e.g. Ninja (which I use, very fast) and more generators like Visual Studio.

Aside: try Ninja with
```
mkdir build ; cd build ; cmake -DFORCE_COLORED_BUILD:BOOL=ON -GNinja .. && cmake --build . -j 10
```

Let's keep this issue open for a while to discuss.

zuckschwerdt, 2021-02-14:
Also note that the `librtlsdr.pc` pkg-config file from `librtlsdr-dev` on Debian Buster is broken. Automake trips on that. CMake can deal with it. This patch would be needed
```
--- /usr/lib/x86_64-linux-gnu/pkgconfig/librtlsdr.pc	2021-02-14 13:41:07.857851107 +0100
+++ /usr/lib/x86_64-linux-gnu/pkgconfig/librtlsdr.pc	2021-02-14 13:41:32.101618910 +0100
@@ -1,4 +1,4 @@
-prefix=
-exec_prefix=
-libdir=
-includedir=
+prefix=/usr
+exec_prefix=${prefix}
+libdir=${prefix}/lib/x86_64-linux-gnu
+includedir=${prefix}/include
```
Maybe someone wants to report that and get it fixed?

merbanan, 2021-03-07:
Kill it when you see fit.

Mindavi, 2021-08-29:
Closed by #1714

closed 2021-08-29 by Mindavi