merbanan/rtl_433 #1645: GPLv3 licensing

Mindavi, 2021-02-04:
Maybe we should ask all authors to relicense (and hope they don't mind) to GPLv2+. Because some parts are licensed GPLv3-only, which makes the whole tool GPLv3 because the rest is GPLv2+. Or we should make clear that the result of linking is GPLv3 in the README / LICENSE.

We should also check if no GPLv2-only code is in the repo, as that would be incompatible with GPLv3-only.

For initial discussion see https://github.com/merbanan/rtl_433/pull/1642

- [ ] List of authors to ping
- [ ] Add Github action for checking licenses

merbanan, 2021-02-05:
I have now asked Paul Ortyl.

Mindavi, 2021-02-19:
I think it's acceptable to assume that any files without license information are fine with the the license that was in the project root folder at the time of creation.

merbanan, 2021-02-19:
I emailed the fs20 author but did not get any response yet.

merbanan, 2023-02-09:
Reimplement the Cardin stuff as a config file and then close the issue.

zuckschwerdt, 2023-02-09:
I've touched every line on cardin and it was originally PPM which I rewrote to PWM, taking authorship. Also Authorship clarified on others I took over with 48157f6 now.

closed 2023-02-09 by zuckschwerdt

Mindavi, 2023-02-09:
Great to see this resolved ✨