merbanan/rtl_433 #1647: Enhancement request for "get" parameter in .conf files

mcsquared88, 2021-02-06:
The current behavior for the "get" parameter is that if the specified bits don't match any of the listed values, rtl_433 just decodes the specified bits into a decimal number and outputs that. When the output from rtl_433 is being sent via MQTT into some other application like Home Assistant, and if I don't know every possible message that a device might send, the "get" parameter may end up outputting an unexpected value.

The simplest fix is to change rtl_433 to not output a value for the custom variable when the bits specified by the get parameter don't match any of the listed values. Even better would be to allow people to specify an "else" value.

Here's an example where "else" values would be really useful. I've mostly reverse-engineered the protocol for the Govee H5054 battery-powered water leak sensor. The message format consists of a device ID, then an event code, and then what I think is a message integrity check that I haven't figured out. The event code can indicate that a leak was detected, that the button was pressed or batteries were inserted, or the event code can just be the device's battery level. I want to define one custom output variable to just indicate if a leak was detected and another custom output variable to just contain the battery level.

I've got the following get parameters defined in my .conf file:
  get = leak:@20:{12}:[0x404:ON 0x202:OFF 0x505:OFF 0x39b:OFF 0x3bc:OFF 0x3c4:OFF 0x3b6:OFF 0x3e5:OFF 0x3e7:OFF],
  get = battery_level:@20:{12}:[0x404:UNDEFINED 0x202:UNDEFINED 0x505:UNDEFINED 0x39b:100 0x3bc:80 0x3c4:80 0x3b6:80 0x3e5:60 0x3e7:40 ],

This mostly works, but when the device sends a leak message, the battery_level variable will be invalid. And when the device sends a battery level message, the leak variable is invalid. And if the device transmits some message I haven't seen before, both of these custom output variables will contain unexpected, invalid values.

If it were possible to specify an "else" value, these "get" parameters would be easier to write and wouldn't ever output an unexpected value:
  get = leak:@20:{12}:[0x404:ON ELSE:OFF],
  get = battery_level:@20:{12}:[0x39b:100 0x3bc:80 0x3c4:80 0x3b6:80 0x3e5:60 0x3e7:40 ELSE:UNDEFINED],

zuckschwerdt, 2021-02-06:
Yes, I'd like to improve the whole flex decoder "language", especially the getters should be friendlier and provide more options. Also simple sanity checks should be possible.

gdt, 2023-09-30:
ticket gardenin: This still seems valid.  

regular hat: In addition to specifying a value, the ELSE: branch should be able to say that the key should be omitted, perhaps as "ELSE:" with no value.