merbanan/rtl_433 #1650: MQTT Unix time is sent as a string

borpin, 2021-02-09:
The JSON output when using MQTT and Unix time results in Unix time as a string instead of a number.

Actual payload
```
{"time":"1612881528","model":"Prologue-TH","subtype":9,"id":185,"channel":3,"battery_ok":1,"temperature_C":21.5,"humidity":10,"button":0}
```

Expected payload
```
{"time":1612881528,"model":"Prologue-TH","subtype":9,"id":185,"channel":3,"battery_ok":1,"temperature_C":21.5,"humidity":10,"button":0}
```

command
```
 rtl_433 -F "mqtt://localhost:1883, events=rtl2[/model]" -M "time:unix"
```

zuckschwerdt, 2021-02-09:
That would be a rather involved change. All our code handles timestamps as strings (and they usually are).

Note that you should put options before the outputs `-M ... -F ...`, some options are positional.

borpin, 2021-02-11:
a Unix timestamp is an integer - JSON spec is that numbers should not be quoted so they can be handled as numbers, so strictly speaking, it should be unquoted.

I've handled in in NR. Cheers.

closed 2021-02-11 by borpin

zuckschwerdt, 2021-02-11:
Sure, I agree. It's just that Unix timestamps were added later as an option an there are no provisions to mark that field as a specific data type -- there is a formatter for the time that just prints a string (in lots of different formats).
And dealing with JSON the preferred datetime format is ISO8601 anyway.
(Historically rtl_433 is not JSON-first but targetting on CSV. If we could phase out CSV then things would look somewhat better...)

All options are read left to right and stack, accumulate or override. Prominent example is verbosity (-v), that applies only to decoders (-R) further on.