merbanan/rtl_433 #1651: [Enhancement] Service file for systemd

borpin, 2021-02-12:
I think a useful addition would be a service file for rtl_433

```
# Do not edit this file - make changes required via a dropin.
#
# First create a symbolic link from the install folder to the systemd folder
#
# sudo ln -sf /home/pi/rtl_433/rtl_433.service /lib/systemd/system
#
# register the Service file
# sudo systemctl daemon-reload
#
#
# Second, create you own set of runtime commands using the systemctl edit mechanism to create a dropin file.
# You can replace any of the commands in the standard service file by adding them to the dropin.
# as a minimum you will want to edit the RTL_433_OPTIONs environment varialbe with the command line options you want
#
# sudo systemctl edit rtl_433.service
#
# Example contents of the dropin (remember it needs the sections in [])
#
# [Service]
# Environment='RTL_433_OPTIONS= -vv -M time:unix -M level -R 03 -R 32 -F "mqtt://192.168.7.50:1883,user=xyz,pass=xyz,retain=0, events=rtl2[/model]"'
#
# Once the dropin in done, do a daemon-reload again
# sudo systemctl daemon-reload
#
# Start the service - check the syslog or the service status to see if OK.
# sudo systemctl start rtl_433.service
#
# To start the service at boot, enable it
# sudo systemctl enable rtl_433.service

[Unit]
Description=RTL_433 service script
StartLimitIntervalSec=5
Documentation=https://github.com/merbanan/rtl_433/README.md

[Service]
Type=exec
# Include a user via the Dropin if required
#User=pi

Environment='RTL_433_OPTIONS= -vv'

ExecStart=/usr/local/bin/rtl_433 $RTL_433_OPTIONS

# Restart script if stopped
Restart=always
# Wait 30s before restart
RestartSec=30s

# Tag things in the log
# View with: sudo journalctl -f -u rtl_433 -o cat
SyslogIdentifier=rtl_433

# modify via a dropin to use a dedicated log file for StdOut and StdErr.
# NOTE: only works in systemd v236+
#       Debain "stretch" includes v232, "buster" includes v239
# Note also the user must be able to create the file and logrotate must be amended to rote the log.
#StandardOutput=file:/var/log/rtl_433.log

StandardOutput=syslog
StandardError=syslog

[Install]
WantedBy=multi-user.target
```

YMMV

zuckschwerdt, 2021-02-12:
Thanks. Yes that might be a good guide or addition. Usually we suggest a supervisor like Supervisord.
It might be better to use the config file for options, default is `/etc/rtl_433/rtl_433.conf`, instead of environment vars.

And running as root is really not needed or safe. We should advocate using `User=pi` -- but there isn't really an universal account to use, unless we create out own :/

borpin, 2021-02-12:
Systemd is the supervisor that comes with Debian/Ubuntu/RaspberryOS so is the only one I know or have experience of.

I couldn't quite work out if the conf files were just for different decoders. That was why I didn't use it.

If it finds a conf file, will that override command line directives?

Agree about running as root is not ideal, but as you say there is no universal user unless you create one at install. However, I would not so far as to say it isn't safe unless the server is exposed to the internet and if you are doing that, you should know what other actions to take.

zuckschwerdt, 2021-02-12:
Supervisord is also available on Debian, kind of a user-managed Systemd.

Command line options will override the default config file in the order they are given, so you could use `-c file` later in the command line options too.

The default config file should give a good idea how to use it. https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

The point about root is that we never intended rtl_433 to run as root and have not checked anything for e.g. privilege escalation (i.e. rtl_433 will read and write files unchecked and in the worst case allow remote commands from the network).

borpin, 2021-02-12:
> The point about root is that we never intended rtl_433 to run as root

Fair enough.

I suggest then that the option `user=pi` could be enabled in the default file but if it needs to be changed for other systems, that should be done by dropin (editing the service file may cause an update via git to fail).

The environment variable could stay but by default be empty with an appropriate comment on either using a dropin to specify the command line options or a configuration file.

Mindavi, 2021-05-18:
We could ship a systemd config example, but I think such things are more a responsibility of someone packaging the tool for a distro instead of the upstream repo, IMO.

Having an example to work from might be beneficial though :).

As for the user, for best privilege separation it would be best to make a rtl433 user or something like that.

borpin, 2023-02-14:
Close as obviously not going to be adopted.

closed 2023-02-14 by borpin

zuckschwerdt, 2023-02-14:
It would be great if you could put this example in the wiki. But we can't support this script for all users and varied use-cases in the repo.