merbanan/rtl_433 #1652: Misidentified temperature sensor: TFA not Lacrosse

migurski, 2021-02-14:
I am running rtl_433 built from source and I have a LaCrosse TX141TH-Bv3 temperature sensor. I expect it to show up in the output with the name "LaCrosse TX141TH-Bv3". Instead, I see its output misidentified as a TFA-303221:

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-02-14 19:08:25
model     : TFA-303221   Sensor ID : 53
Channel   : 1            Battery   : 1             Temperature: 8.30 C       Humidity  : 71 %          Test mode : 0             Integrity : CRC
```

When running `rtl_433 -R 73` to pick out just the LaCrosse device, no output is generated.

We don’t have a TFA and I’ve checked that the temperature readings are consistent with the device we do own. Is it possible that this device ID is being mislabeled?

migurski, 2021-02-14:
Here are two sample pulses from the LaCrosse device:

* https://triq.org/pdv/#AAB00B0404034400F001E827148055+AAB0340401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A1928055+AAB00B0403034400F001E827148055+AAB0340401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A1928055+AAB00B0403034400F001E827148055+AAB0340401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A1928055+AAB00B0403034400F001E827148055+AAB0330401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A1A355
* https://triq.org/pdv/#AAB00B0404034400F001E827148055+AAB0340401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A1A18055+AAB00B0403034400F001E827148055+AAB0340401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A1928055+AAB00B0403034400F001E827148055+AAB0340401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A1A18055+AAB00B0403034400F001E827148055+AAB0330401034400F001E827149292A1A192A192A1929292929292A19292A19292A192929292A19292A192929292A192A1929292A19355

zuckschwerdt, 2021-02-14:
Interesting find. They appear to be the same protocol indeed. Strange that we didn't pick this up before.
https://github.com/merbanan/rtl_433/blob/master/src/devices/tfa_30_3221.c#L12
https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrosse_tx141x.c#L48
(the digest is uncommon enough (init f4) that this isn't likely coincidence.

I'd say the TFA 30.3221.02 is a rebranded LaCrosse TX141TH. I'll investigate and maybe delete one.

migurski, 2021-02-14:
Thanks for looking into this!

merbanan, 2021-03-14:
Any update on this ?

on4cg, 2021-03-26:
I bought a TFA 35.1140 with a sensor 30.3222.02. I read LaCrosse-TX141W.

zuckschwerdt, 2021-03-26:
Does the TFA  30.3222.02 show the correct values as LaCrosse-TX141W?

on4cg, 2021-03-27:
On first view, yes ...
But I have to view this for a longer period...

Domifry, 2021-10-08:
I have the same behavior. I own two TFA 30.3221.02 and it shows: LaCrosse-TX141THBv2 - works with the code: sudo rtl_433 -R 73

The data are correct so I will just use it.

zuckschwerdt, 2021-10-08:
Thanks for the report. TFA is just a reseller, your sensor is very likely an original LaCrosse ;)

Still todo: remove the TFA-303221 protocol someday.

Domifry, 2021-10-09:
Just a short Update - if you have the TFA it currently identify both. You can see from my 2 Sensors sometimes it tells me: I am a TFA and sometimes I am a Lacrosse.
[image]

zuckschwerdt, 2022-06-19:
Decoder priority stages have been introduced with #1895 and the TFA 30.3221 is now only a last resort fallback, you should see LaCrosse-TX141THBv2 usually, and never both.

closed 2022-06-19 by zuckschwerdt