merbanan/rtl_433 #1656: Changes with sensing inFactory [91]

mikethebee, 2021-02-24:
FYI: I'm not sure what changed, but I was logging protocol 91 on my DVB-T dongle fc00012 with just "rtl_433 -R 91" okay. 

This week it failed to detect the sensor, while the TG647 receiver displays the data correctly. I changed the sensor batteries without success. Looking in the code I see "rtl_433 -f 434052000 -R 91 -F json" suggested and this works fine. Perhaps this should be the default for inFactory.

mikethebee, 2021-02-26:
Closing as comments have moved on.

closed 2021-02-26 by mikethebee

zuckschwerdt, 2021-02-26:
We can't really help with picking a suitable frequency. Look at the actual spectrum with CubicSDR, Gqrx, or SDRangel if you have the opportunity. Maybe try `-s 1024k`