merbanan/rtl_433 #1657: LoRa support

eadmaster, 2021-02-24:
LoRa seems to use the same frequencies supported by this tool.

An open source decoder is [here](https://github.com/rpp0/gr-lora).

zuckschwerdt, 2021-02-24:
LoRa is a specific application. Our general case filters and demodulation won't work well.

closed 2021-02-24 by zuckschwerdt

merbanan, 2021-02-24:
@zuckschwerdt am I not allowed to work on this you mean ? ;)

merbanan, 2021-02-24:
[image]

Teaser image attached. We'll see where it goes. Demodulation is tricky, but it should be possible to do with the code we have.

zuckschwerdt, 2021-02-24:
We'd need new filters and demods. And that would be great to have! At least to solidify the idea of different "pipelines" for specific decoders. I'm also lazily working on some filter updates -- it's a little fragile because current setups implicitly assume stuff. E.g. wide bandwidths == wide filters.

One idea I'm toying with is to (automatically) shift the instant phase (pseudo-FFT) results afterwards to apply the low-pass where it's really wanted. (currently the FSK low-pass is adverse for signals that are not centered).

merbanan, 2021-02-24:
Code we have + rework of the process pipeline.

heeplr, 2023-11-18:
Are there any news on LoRa support? It'd be awesome!

heeplr, 2024-10-07:
Adding this for future reference, here's another project that can decode LoRa with RTL dongles: 
https://github.com/hexameron/rtl-sdrangelove/tree/master/plugins/channel/lora