merbanan/rtl_433 #1658: HTTP server usage?

jcwren, 2021-02-24:
I'm trying to start the HTTP server on port 8433 with the following command:
```
rtl_433 -M level -F http -F "mqtt://mqtt.private.com:1883,retain=1"
```
The HTTP server is starting, but after indicating "INITIALIZING..." at the top for about 2 seconds, it always switches to `DISCONNECTED`.  The log file is below.

What am I doing wrong, or why might it not be working correctly? It's reporting MQTT traffic correctly (as it has been for the last year) so all that part still works as expected.

```
rtl_433 version 20.11-80-g320a98a branch master at 202102230810 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/jcw/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
HTTP server at 0.0.0.0 port 8433
Publishing MQTT data to mqtt.private.com port 1883
Publishing device info to MQTT topic "rtl_433/rtl433/devices[/type][/model][/subtype][/channel][/id]".
Publishing events info to MQTT topic "rtl_433/rtl433/events".
Publishing states info to MQTT topic "rtl_433/rtl433/states".
Registered 149 out of 179 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-179 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
MQTT Connected...
MQTT Connection established.
```
[image]

zuckschwerdt, 2021-02-24:
Which browser is that? I test on Firefox, but also checked Chrome and Safari.
Can you open the console (DevTools/Inspector/F12/... then switch to Console tab) and see why the websocket doesn't connect?

jcwren, 2021-02-24:
Using Chrome on Windows, connecting to rtl-433 running on the RPi.
```
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
DevTools failed to load SourceMap: Could not load content for chrome-extension://bfogiafebfohielmmehodmfbbebbbpei/javascript/sandblaster.min.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
DevTools failed to load SourceMap: Could not load content for https://triq.org/rxui/js/chunk-vendors.49b747f0.js.map: Load canceled due to load timeout
DevTools failed to load SourceMap: Could not load content for https://triq.org/rxui/js/app.8fbd1950.js.map: Load canceled due to load timeout
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
chunk-vendors.js:118 WebSocket connection to 'ws://127.0.0.1:8433/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
value @ chunk-vendors.js:118
(anonymous) @ chunk-vendors.js:118
```

zuckschwerdt, 2021-02-24:
Ok, "ws://127.0.0.1:8433/" would be the wrong path then.
Try to open `http://192.168.1.99:8433/#192.168.1.99:8433` (where 192.168.1.99 is the address of your Raspi) -- seems I recently broken the automatic choice for the WS, sorry!

jcwren, 2021-02-24:
Excellent. That's working now. Thanks!

zuckschwerdt, 2021-02-24:
I fixed this in the web ui now :) Should work by default again.

closed 2021-02-24 by zuckschwerdt

zuckschwerdt, 2021-02-24:
Thanks for testing again! It's live, maybe a forced browser reload is needed? If you look in the Console the app resource is now `https://triq.org/rxui/js/app.1811d3c7.js` it was `8fbd1950` before.

jcwren, 2021-02-24:
Yeah, my bad. I was expecting the rtl_433 repository would need to be updated. Testing it with just the base URL works as expected.

Thanks for the super-fast fix!

zuckschwerdt, 2021-02-24:
The app is web hosted so users always get the most recent version even with older rtl_433 builds. If someday the app is feature-complete and stable we could perhaps pull it in while building and relasing rtl_433.