merbanan/rtl_433 #1660: Error in decoding Rain TFA 30.3161

ingotero, 2021-02-26:
Error data layout:
    IIIIIIII RRRRtttt TTTTTTTT rrrrrrrr CCCC
- I : 8-bit ID
- T : 12-bit Temp in C, signed, scaled by 10
- R : 12-bit Rain
- C : 4-bit Checksum (nibble sum)

Correct data layout:
    IIIIRRRR RRRPtttt TTTTTTTT rrrrrrrr CCCC
- I : 4-bit ID
- T : 12-bit Temp in C, signed, scaled by 10
- R : 15-bit Rain
- P : 1-bit Parity 
- C : 4-bit Checksum (nibble sum)

Order of Rain bits:
[image]

Rain click counter reaches 4095 and then restarts from 0.

zuckschwerdt, 2021-02-26:
4095 is 12 bit, but you marked 15 bit?

zuckschwerdt, 2021-02-26:
Decoder link: https://github.com/merbanan/rtl_433/blob/master/src/devices/inovalley-kw9015b.c

ingotero, 2021-02-27:
Yes, I tested this but it fails after 255 clicks

time : 2021-02-22 21:05:13
model : Inovalley-kw9015b id : 14
Temperature: 27.5 C Rain Count: 254 Rain Total: 114.300
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time : 2021-02-22 21:05:48
model : Inovalley-kw9015b id : 14
Temperature: 27.5 C Rain Count: 255 Rain Total: 114.750
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time : 2021-02-22 21:06:23
model : Inovalley-kw9015b id : 14
Temperature: 27.5 C Rain Count: 512 Rain Total: 230.400
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time : 2021-02-22 21:06:58
model : Inovalley-kw9015b id : 14
Temperature: 27.6 C Rain Count: 513 Rain Total: 230.850

That is using rtl_433 code with my rtl_sdr.

joaquinotero, 2021-03-01:
[image]

Hi Christian, I've checked again and that is the correct data layout.

Correct data layout:
IIIIII RR P RR P tttt TTTTTTTT rrrrrrrr CCCC

I : 6-bit ID
T : 12-bit Temp in C, signed, scaled by 10
R : 12-bit Rain
P : 2-bit Parity
C : 4-bit Checksum (nibble sum)

joaquinotero, 2021-03-01:
I'm now checking ID, becouse those parity bits may be part of the ID.

joaquinotero, 2021-03-01:
Correct data layout:
IIII ?? RR B RR P ttttTTTTTTTT rrrrrrrr CCCC
I : 4-bit ID
?? : 2-bit Dont know yet, they are always 00
R : 12-bit Rain
B : 1-bit Battery (0 means battery ok, 1 means low battery)
P : -bit Parity (when batterys are inserted that bit is 1, then always 0)
T : 12-bit Temp in C, signed, scaled by 10
C : 4-bit Checksum (nibble sum)

zuckschwerdt, 2021-03-02:
Is "Parity" really a parity check? Does it change when R changes? Or is it "New Batteries, Power Up"?

joaquinotero, 2021-03-02:
IIII ?? RR B RR P ttttTTTTTTTT rrrrrrrr CCCC
I : 4-bit ID
?? : 2-bit Dont know yet, they are always 00
R : 12-bit Rain
B : 1-bit Battery (0 means battery ok, 1 means low battery)
P : 1-bit Power Up (when batterys are inserted that bit is 1, then always 0)
T : 12-bit Temp in C, signed, scaled by 10
C : 4-bit Checksum (nibble sum)

P is a Power Up bit, is not parity, you are right I'was using the incorrect terminology.

joaquinotero, 2021-03-02:
Do you have any idea of what those two bits ?? could mean? 
I made all the possible changes, but they are still 00.

Thanks.

zuckschwerdt, 2021-03-02:
@rtega has some info in #1501

joaquinotero, 2021-03-02:
Thanks Christian but I've started working from there, and already made some improvements.

Thanks anyway.

zuckschwerdt, 2021-03-02:
Perhaps @rtega has some way to help -- I don't have this sensor ;)

merbanan, 2021-03-16:
@joaquinotero will you send a patch based on the findings you got here?

ingotero, 2021-03-16:
Just coded for arduino, I don't know how to do it in this case, sorry.

Joaquín.

merbanan, 2021-03-16:
Ok, can you share the code?

merbanan, 2021-03-16:
```
IIII ?? RR B RR P ttttTTTTTTTT rrrrrrrr CCCC
I : 4-bit ID
?? : 2-bit Dont know yet, they are always 00
R : 12-bit Rain
B : 1-bit Battery (0 means battery ok, 1 means low battery)
P : 1-bit Power Up (when batterys are inserted that bit is 1, then always 0)
T : 12-bit Temp in C, signed, scaled by 10
C : 4-bit Checksum (nibble sum)

P is a Power Up bit, is not parity, you are right I'was using the incorrect terminology.
```

Is this the final representation ?

ingotero, 2021-03-16:
Yes, that is the final representation.

Here goes the code, I run it with an ESP32, but should work with other
boards as well:

/* Decode TFA 30.3161 (Inovalley kw9015b rain and Temperature weather station)

 * Written by : Joaquin Otero

 * ***@***.***

*/

// ring buffer size has to be large enough to fit
// data between two successive sync signals
#define RING_BUFFER_SIZE  256

#define SYNC_LENGTH  9000
#define SEP_LENGTH   550
#define BIT1_LENGTH  4000
#define BIT0_LENGTH  2000

#define DATAPIN  15

unsigned long timings[RING_BUFFER_SIZE];
unsigned int syncIndex1 = 0;  // index of the first sync signal
unsigned int syncIndex2 = 0;  // index of the second sync signal
float last_temp1 = 0;         //crea variable para mantener valor anterior
float last_rain1 = 0;         //crea variable para mantener valor anterior
float temp1 = 0;
float rain1 = 0;
bool received = false;

// detect if a sync signal is present
  bool isSync(unsigned int idx) {
  unsigned long t0 = timings[(idx+RING_BUFFER_SIZE-1) % RING_BUFFER_SIZE];
  unsigned long t1 = timings[idx];

  // Accounting for error it should be within 8.0ms and 10.0ms
  if (t0>(SEP_LENGTH-100) && t0<(SEP_LENGTH+100) &&
    t1>(SYNC_LENGTH-1000) && t1<(SYNC_LENGTH+1000) &&
    digitalRead(DATAPIN) == HIGH) {
    return true;
  }
  return false;
}

/* Interrupt 1 handler */
void handler() {
  static unsigned long duration = 0;
  static unsigned long lastTime = 0;
  static unsigned int ringIndex = 0;
  static unsigned int syncCount = 0;

  // ignore if we haven't processed the previous received signal
  if (received == true) {
    return;
  }

  // calculating timing since last change
  long time = micros();
  duration = time - lastTime;
  lastTime = time;

  // store data in ring buffer
  ringIndex = (ringIndex + 1) % RING_BUFFER_SIZE;
  timings[ringIndex] = duration;

  // detect sync signal
  if (isSync(ringIndex)) {
[... 328 lines cut ...]
      Serial.print(bitRead(num,i));
    }
    //FIN DEBUG
 */
    for (i = 0; i < (digitos); i++)
    {
      reverse_num |= (bitRead(num,i)<<(digitos-i-1));
    }
/*
   //Solo DEBUG
    Serial.println();
    Serial.print("Reverso: ");
    for ( i=(digitos-1);i >= 0; i--)
    {
      Serial.print(bitRead(reverse_num,i));
    }
    Serial.println();
    //FIN DEBUG
*/
    return reverse_num;
}

Best,

Joaquín Otero.

merbanan, 2021-03-16:
@zuckschwerdt can you use this info to update the decoder?

rtega, 2021-10-20:
The current code in the repository is indeed wrong. My counter just jumped from 767 to 1024. Did anybody put time into correcting the code? Else I'll have a look.

zuckschwerdt, 2021-10-21:
@rtega please do!

rtega, 2021-10-21:
And in case you ask, no, I won't go to the trouble of doing a git commit and stuff. The code resides on a very simple raspberry pi zero.