merbanan/rtl_433 #1662: Async read stall error does not result in program exit

hschimke, 2021-02-28:
Found the syslog entry:  rtl_service.sh[5490]: Async read stalled, exiting!

But the program does not actually exit, rather it sits in a hung state until it is forcibly restarted. This prevents the process manager from detecting the issue and restarting the program.

Using nightly build from: 320a98a6ff515dec1748921dcbeba86e5d4fa192

zuckschwerdt, 2021-02-28:
Yes, there might be cases where the new sdr loop notices trouble but doesn't exit. I need to look into that.

hschimke, 2021-03-01:
I rolled back to 20.11 (0bb4672b1afb716d107b3060afc2d80a2159dbef) and cannot reproduce the issue with that build.

Additionally: with the build I mentioned in the original issue I saw the hang occur each night it was running. With the 20.11 build I do not see any reset or hangs at all. Previously I was using a nightly build from early February (I do not have the commit, unfortunately) which did not have the hang or reset.

zuckschwerdt, 2021-03-01:
This regression is likely from 619ca1f. The SDR input has a dedicated loop that has now no mechanism to exit early. The loop *should* crash on a hang. But that was originally also a problem and solved with the stall-timeout-alarm.
I'll try to propagate the stall condition to the loop.

zuckschwerdt, 2021-03-02:
I had a closer look. We do call sdr_stop() on stall. https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1201
That will then trigger rtlsdr_cancel_async() https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L1508
which breaks the input loop. https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L493

Can you add printf()s at those points to watch if that really happens, and where the loop gets stuck?
Not sure why the program wouldn't exit after stalling here.

hschimke, 2021-03-02:
Built and running, I'll monitor it and let you know any output, thanks!

hschimke, 2021-03-02:
It crashed after about two hours of running. 

I added prints to the three places you mentioned.

Here is my syslog output:
```
rtl_433_mqtt_433[6160]: Async read stalled, exiting!
rtl_433_mqtt_433[6160]: Reached sdr_stop
rtl_433_mqtt_433[6160]: Reached 'return rtlsdr_cancel_async(dev->rtlsdr_dev);'
```

Those correspond to the changes I made to your first two links (rtl_433.c 1201 and sdr.c 1508).

None of my changes at sdr.c 493 triggered.

The program hung and did not exit without a forced restart.

My changes for sdr.c 493:
```
        dev->polling = 1;

        r = rtlsdr_read_async(dev->rtlsdr_dev, rtlsdr_read_cb, dev, buf_num, buf_len);
        // rtlsdr_read_async() returns possible error codes from:
        //     if (!dev) return -1;
        //     if (RTLSDR_INACTIVE != dev->async_status) return -2;
        //     r = libusb_submit_transfer(dev->xfer[i]);
        //     r = libusb_handle_events_timeout_completed(dev->ctx, &tv,
        //     r = libusb_cancel_transfer(dev->xfer[i]);
        // We can safely assume it's an libusb error.
        if (r < 0) {
fprintf(stderr,"read return less than 0\n");
#ifdef LIBUSB1
            fprintf(stderr, "\n%s: %s!\n"
                            "Check your RTL-SDR dongle, USB cables, and power supply.\n\n",
                    libusb_error_name(r), libusb_strerror(r));
#else
            fprintf(stderr, "\nLIBUSB_ERROR: %d\n"
                            "Check your RTL-SDR dongle, USB cables, and power supply.\n\n",
                    r);
#endif
fprintf(stderr,"dev running now 0\n");
            dev->running = 0;

```

zuckschwerdt, 2021-03-02:
Thanks, that narrows it down. So `rtlsdr_cancel_async()` is run but `rtlsdr_read_async()` won't return. This suggests that libusb is in a bad state. We can't do anything about that :(
You could recompile librtlsdr.c and see what blocks in those function https://github.com/osmocom/rtl-sdr/blob/master/src/librtlsdr.c
But I guess it's libusb trouble.

hschimke, 2021-03-02:
I had a version working that I built earlier in the month. I'm going to do some different builds and see if I can zero in on when it started to frequently encounter Async errors. I looked back over my logs and I didn't have any appear until recently, and going back to a version built with the 20.11 tag didn't encounter any either.

I'm probably using rather old versions of some of these libraries, since they're what's included with raspbian deb packages.

zuckschwerdt, 2021-03-03:
Another hunch: it might be that the callback is blocking. Wrap https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L488 with printf()s
and perhaps a few printf()s in https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1205

mikethebee, 2021-03-03:
FWIW: I have been having similar occasional hangs. Restarting rtl_433 doesn't work, though. I am implementing some python code to reset the dvb-t usb device to see if that works. Previously I have had to unplug/re-plug the dongle or reboot. This is on a Pi2b running Buster.

hschimke, 2021-03-03:
I've been running 404d8ead8b6e8e94719a7e69588c2236f2cdd3f2 for about 20 hours now without any Async issues. I'm going to let it run for a full 24 and then I'll build a current nightly with the printfs you suggest and let you know.

hschimke, 2021-03-03:
I ran 404d8ead8b6e8e94719a7e69588c2236f2cdd3f2 for 23 hours and didn't get any Async issues. I found several `pulse_FSK_detect(): Maximum number of pulses reached!` in the logs, but the process doesn't hang or exit. I'm going to run the current nightly with the printfs you requested for a bit and I'll update with that data. After that I'll build and try the commit from Feb 15 and see if it will run for 24 hours without hanging.

RandomSporadicProjects, 2021-03-03:
> FWIW: I have been having similar occasional hangs. Restarting rtl_433 doesn't work, though. I am implementing some python code to reset the dvb-t usb device to see if that works. Previously I have had to unplug/re-plug the dongle or reboot. This is on a Pi2b running Buster.

I know that this isn't a solution, but rather a fix, but I thought I'd point you to a reply of mine that may save you some time, it's not python, but it does the same thing see https://github.com/merbanan/rtl_433/issues/1586#issuecomment-749621622 for details of how I do a usbreset

What I didn't mention in that post is that I monitor my SDRstick with a node red watchdog flow (previously I used a cron job).  The flow listens for mqtt output from the SDR, and uses a message counter node to count them every minute.  If no messages have been received from the stick, then the scripts mentioned in my previous post are run - this means that you only reset the stick if it needs it.   In my environment I've lot's of sensors, and my electricity consumption one fires off every 6s, so a minute without a messages means something is wrong, but of course you could change the period without a message to anything you want.

Flow is below in case it's of use.

 [{"id":"6e2077b0.f4f04","type":"tab","label":"Flow 3","disabled":false,"info":""},{"id":"7309ebce.77426c","type":"debug","z":"6e2077b0.f4f04","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":710,"y":120,"wires":[]},{"id":"f42b92f5.eb6b68","type":"exec","z":"6e2077b0.f4f04","command":"","addpay":true,"append":"","useSpawn":"false","timer":"","oldrc":false,"name":"","x":770,"y":340,"wires":[[],[],[]]},{"id":"7a335b4a.8f0394","type":"change","z":"6e2077b0.f4f04","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"/home/pi/scripts/resetsdr.sh &","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":360,"y":340,"wires":[["1ef206a.30960f9"]]},{"id":"982441d.442c04","type":"delay","z":"6e2077b0.f4f04","name":"","pauseType":"delay","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":520,"y":480,"wires":[["f42b92f5.eb6b68","fbbed19b.05b5f8"]]},{"id":"fbbed19b.05b5f8","type":"delay","z":"6e2077b0.f4f04","name":"","pauseType":"delay","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":520,"y":540,"wires":[["f42b92f5.eb6b68","39cf1258.a8c91e"]]},{"id":"1ef206a.30960f9","type":"delay","z":"6e2077b0.f4f04","name":"","pauseType":"delay","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":520,"y":420,"wires":[["f42b92f5.eb6b68","982441d.442c04"]]},{"id":"39cf1258.a8c91e","type":"change","z":"6e2077b0.f4f04","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"/home/pi/scripts/runsdrstick.sh &","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":520,"y":620,"wires":[["94d7d8e6.f96188"]]},{"id":"94d7d8e6.f96188","type":"delay","z":"6e2077b0.f4f04","name":"","pauseType":"delay","timeout":"5","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":520,"y":700,"wires":[["f42b92f5.eb6b68","4141def2.3d687"]]},{"id":"8c9da5e.8e43658","type":"Message Counter","z":"6e2077b0.f4f04","name":"","units":"minutes","interval":"1","alignToClock":true,"generator":"internal","x":530,"y":180,"wires":[["ef07f640.8f1438","d71615a6.3514b8"],[]]},{"id":"ef07f640.8f1438","type":"debug","z":"6e2077b0.f4f04","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":750,"y":180,"wires":[]},{"id":"d71615a6.3514b8","type":"function","z":"6e2077b0.f4f04","name":"","func":"if (msg.payload == 0){\nreturn msg;\n}","outputs":1,"noerr":0,"x":520,"y":280,"wires":[["7a335b4a.8f0394"]]},{"id":"9045e08.c3a0ba","type":"debug","z":"6e2077b0.f4f04","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":770,"y":600,"wires":[]},{"id":"4141def2.3d687","type":"change","z":"6e2077b0.f4f04","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"Restarted","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":751,"y":713,"wires":[["9045e08.c3a0ba"]]},{"id":"3ae4a7b1.3fdd18","type":"mqtt in","z":"6e2077b0.f4f04","name":"","topic":"RTL_433/RAW","qos":"2","datatype":"auto","broker":"1c5b520b.9bf09e","x":280,"y":120,"wires":[["7309ebce.77426c","8c9da5e.8e43658"]]},{"id":"1c5b520b.9bf09e","type":"mqtt-broker","z":"","name":"mqqt server","broker":"192.168.1.1","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]

zuckschwerdt, 2021-03-04:
We are really are interested in making rtl_433 robust for all use cases. If you find any error condition where rtl_433 doesn't work any more but also doesn't exit: please investigate and share.
Running rtl_433 in a restart loop should be all that's needed. (and possibly reset the USB between starts if your hardware is flaky in some way -- but that also could benefit from root cause analysis.)

@RandomSporadicProjects you are watching MQTT because the reception goes "deaf" without any noticeable errors otherwise? If so, can you put a printf() in rtlsdr_read_cb https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L488
and/or sdr_handler https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1205 to check wether there is still I/Q data comming in?

RandomSporadicProjects, 2021-03-04:
I've been using rtl_433 for about 3-4 years now.  In the first couple of years I would say it crashed every couple of weeks, and that's when I set up the usbreset scripts I've mentioned.  Over time it has become much more reliable.  I last compiled my running version in September 2020, and I doubt it's needed reset more than a couple of times since then (I don't have a count but know it's not a frequent occurrence).  I'd help this project in any way I could as I think it's a really good project, but sadly most of the technical discussions are beyond me here, and I don't program in C.  I was just trying to help provide a sticking plaster while you were sorting out a fix, and a watchdog's sometimes handy.

zuckschwerdt, 2021-03-04:
Thanks, if there is more need or interest in debugging (non-)crashes / stuck SDRs we might add some special "show data flow"-debug option.

hschimke, 2021-03-04:
I've been running the version with printfs as requested for 19 hours without any Async issues. My current working theory is that something went wonky with my power in those previous time frames and caused libusb to have a bad time.

A sample output from syslog, I get about three of these every second and so far nothing out of the ordinary:
```
rtlsdr_read_cb(unsigned char *iq_buf, uint32_t len, void *ctx) / len: 262144
rtlsdr_read_cb(unsigned char *iq_buf, uint32_t len, void *ctx) / len: 262144 / through if (len > 0)
```

For now I'm going to write a small service monitor script to check for the 'Async' message and restart the service whenever it's detected. If I can come up with a consistent set of reproducible steps to duplicate my issue I'll check back in.

Thanks for your help!

Simple monitor below:

Checkup script:
```
#!/bin/bash
#readonly SERVICES=("rtl_433_mqtt_43392.service" "rtl_433_mqtt_915.service")
readonly SERVICES=("rtl_433_mqtt_43392.service")
readonly SCAN_LINE="Async read stalled, exiting!"
readonly TIME_FRAME="-15minutes"

for SERVICE in ${SERVICES[@]}; do
  ACTION="none"
  FOUND="no issues"
  LINE_COUNT=`journalctl -u ${SERVICE} --since "${TIME_FRAME}" | grep --ignore-case "${SCAN_LINE}" | wc -l`
  if [ $LINE_COUNT -ne 0 ]
    then
      systemctl restart ${SERVICE} 
      ACTION="restarted"
      FOUND="issues"
  fi
  echo "Checkup of ${SERVICE} found ${FOUND} in the period ${TIME_FRAME} action taken: ${ACTION}"
done
```

SystemD service unit:
```
[Unit]
Description=rtl_433 Async issue checkup service

[Service]
Type=oneshot
WorkingDirectory=/home/pi/sandbox/weewx-sdr-mqtt/rtl_scripts
ExecStart=/home/pi/sandbox/weewx-sdr-mqtt/rtl_scripts/rtl_433_checkup.sh
User=root
SyslogIdentifier=rtl_433_checkup

[Install]
WantedBy=multi-user.target
```

SystemD timer unit
```
[Unit]
Description=rtl_433 Async checkup timer

[Timer]
OnCalendar=*:0/15
AccuracySec=1m
Persistent=true

[Install]
WantedBy=timers.target
```

rknobbe, 2021-03-12:
I've been seeing the same problem here. Let me know what I can do to collect data for you. 

`rtl_433 -F mqtt://localhost:1883,retain=0,devices=sensors/rtl_433/P[protocol:255]/C[channel:0] -M newmodel -M protocol -M time:iso -R -30
rtl_433 version 20.11-87-gad55a535 branch master at 202103101351 inputs file rtl_tcp RTL-SDR with TLS`

rknobbe, 2021-03-12:
I added the following:

'
static void rtlsdr_read_cb(unsigned char *iq_buf, uint32_t len, void *ctx)
{
    sdr_dev_t *dev = ctx;
    sdr_event_t ev = {
            .ev  = SDR_EV_DATA,
            .buf = iq_buf,
            .len = len,
    };
    if (len > 0) // prevent a crash in callback
    {
            fprintf(stderr, "rtlsdr_read_cb: len %d\n", len);
            dev->rtlsdr_cb(&ev, dev->rtlsdr_cb_ctx);
    }
    // NOTE: we actually need to copy the buffer to prevent it going away on cancel_async
}
'
and within a few minutes get this with no further output
'
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
rtlsdr_read_cb: len 262144
Async read stalled, exiting!
'

zuckschwerdt, 2021-03-12:
Thanks! So sdr_handler is called, read stalled occurs and the handler never returns.
Can you debug if it gets stuck in mg_mgr_poll, sdr_callback, or sdr_stop? https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1243
Perhaps we either need the poll conditional on !cfg->exit_async or remove the sdr_stop there.

rknobbe, 2021-03-12:
It's not getting to sdr_stop(), as I had a trace statement right before it.  I'll add some more traces around the mg_mgr_poll and sdr_callback and let you know
```

    if (ev->ev == SDR_EV_DATA) {
        fprintf(stderr, "sdr_handler: SDR_EV_DATA\n");
        if (cfg->mgr) {
            int max_polls = 16;
            while (max_polls-- && mg_mgr_poll(cfg->mgr, 0));
        }

        if (!cfg->exit_async)
            sdr_callback((unsigned char *)ev->buf, ev->len, ctx);
    }

    fprintf(stderr, "sdr_handler: exit_async=%d\n",cfg->exit_async);

    if (cfg->exit_async)
        sdr_stop(cfg->dev);
```

rknobbe, 2021-03-12:
```
sdr_handler: SDR_EV_DATA
sdr_handler: mg_mgr_poll 15
sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback
sdr_handler: back from sdr_callback
sdr_handler: exit_async=0
rtlsdr_read_cb: back from rtlsdr_cb()
rtlsdr_read_cb: len 262144
sdr_handler: SDR_EV_DATA
sdr_handler: mg_mgr_poll 15
sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback
Async read stalled, exiting!
```
```

    if (ev->ev == SDR_EV_DATA) {
        fprintf(stderr, "sdr_handler: SDR_EV_DATA\n");
        if (cfg->mgr) {
            int max_polls = 16;

            while (max_polls-- && fprintf(stderr, "sdr_handler: mg_mgr_poll %d\n", max_polls) && mg_mgr_poll(cfg->mgr, 0));
            fprintf(stderr, "sdr_handler: done mg_mgr_poll-ing\n");
        }

        if (!cfg->exit_async)
        {
                fprintf(stderr, "sdr_handler: calling sdr_callback\n");
            sdr_callback((unsigned char *)ev->buf, ev->len, ctx);
                fprintf(stderr, "sdr_handler: back from sdr_callback\n");
        }
    }

    fprintf(stderr, "sdr_handler: exit_async=%d\n",cfg->exit_async);

    if (cfg->exit_async)
        sdr_stop(cfg->dev);
}

```

zuckschwerdt, 2021-03-13:
That's rather unexpected, getting stuck in `sdr_callback()`? Thats the core signal processing, it's lengthy but simple sequential steps. Not sure, perhaps the MQTT output handler is stuck?

To get there faster, can you run https://github.com/merbanan/rtl_433/tree/wip-trace or apply this patch https://github.com/merbanan/rtl_433/commit/08092750295e491f2a5d771cf223f88cda397959.patch
each Ctrl-C should then print a stacktrace.

rknobbe, 2021-03-14:
Ran overnight with wip-trace and no hangs.  Going to try applying the patch to my previous version to see if I can get it to hang

rknobbe, 2021-03-14:
Got one. Not much help when compiled without symbols though... I'll try again this afternoon

```
rtlsdr_read_cb: back from rtlsdr_cb()
rtlsdr_read_cb: len 262144
sdr_handler: SDR_EV_DATA
sdr_handler: mg_mgr_poll 15
sdr_handler: mg_mgr_poll 14
sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback
sdr_handler: back from sdr_callback
sdr_handler: exit_async=0
rtlsdr_read_cb: back from rtlsdr_cb()
rtlsdr_read_cb: len 262144
sdr_handler: SDR_EV_DATA
sdr_handler: mg_mgr_poll 15
sdr_handler: mg_mgr_poll 14
sdr_handler: mg_mgr_poll 13
sdr_handler: mg_mgr_poll 12
sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback
rtl_433(+0x1bed9) [0x563e7c6d1ed9]
/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f20e49d4210]
rtl_433(+0x7d1a9) [0x563e7c7331a9]
rtl_433(+0x46d6e) [0x563e7c6fcd6e]
rtl_433(+0x472c1) [0x563e7c6fd2c1]
rtl_433(+0x4f2d0) [0x563e7c7052d0]
rtl_433(+0x1d956) [0x563e7c6d3956]
rtl_433(+0x1feeb) [0x563e7c6d5eeb]
rtl_433(+0x51d87) [0x563e7c707d87]
/usr/local/lib/librtlsdr.so.0(+0x6da1) [0x7f20e4cf1da1]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xab66) [0x7f20e4cd9b66]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xf49c) [0x7f20e4cde49c]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x10d98) [0x7f20e4cdfd98]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xa5a9) [0x7f20e4cd95a9]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_timeout_completed+0x19c) [0x7f20e4cda65c]                                                        /usr/local/lib/librtlsdr.so.0(rtlsdr_read_async+0x1a3) [0x7f20e4cf1ab3]
rtl_433(+0x53724) [0x563e7c709724]
rtl_433(+0x1ad93) [0x563e7c6d0d93]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f20e49b50b3]
rtl_433(_start+0x2e) [0x563e7c6d1dae]
Async read stalled, exiting!
```

rknobbe, 2021-03-14:
a bit more info. I could use some pointers in getting better symbols in the traceback

```
rtlsdr_read_cb: len 262144
sdr_handler: SDR_EV_DATA
sdr_handler: mg_mgr_poll 15
sdr_handler: mg_mgr_poll 14
sdr_handler: mg_mgr_poll 13
sdr_handler: mg_mgr_poll 12
sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback
src/rtl_433(+0x1bed9) [0x55cdc6569ed9]
/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f15638fb210]
src/rtl_433(+0x7d12c) [0x55cdc65cb12c]
src/rtl_433(+0x46d6e) [0x55cdc6594d6e]
src/rtl_433(+0x472c1) [0x55cdc65952c1]
src/rtl_433(+0x4f2d0) [0x55cdc659d2d0]
src/rtl_433(+0x1d956) [0x55cdc656b956]
src/rtl_433(+0x1feeb) [0x55cdc656deeb]
src/rtl_433(+0x51d87) [0x55cdc659fd87]
/usr/local/lib/librtlsdr.so.0(+0x6da1) [0x7f1563c18da1]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xab66) [0x7f1563c00b66]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xf49c) [0x7f1563c0549c]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x10d98) [0x7f1563c06d98]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xa5a9) [0x7f1563c005a9]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_timeout_completed+0x19c) [0x7f1563c0165c]
/usr/local/lib/librtlsdr.so.0(rtlsdr_read_async+0x1a3) [0x7f1563c18ab3]
src/rtl_433(+0x53724) [0x55cdc65a1724]
src/rtl_433(+0x1ad93) [0x55cdc6568d93]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f15638dc0b3]
src/rtl_433(_start+0x2e) [0x55cdc6569dae]
Async read stalled, exiting!

```

rknobbe, 2021-03-15:
Got another one after compiling with CC="gcc -rdynamic" and still no symbols in the stacktrace. Any pointers?

```
rtlsdr_read_cb: len 262144
sdr_handler: SDR_EV_DATA
sdr_handler: mg_mgr_poll 15
sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback
src/rtl_433(+0x1bed9) [0x564b4b172ed9]
/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f91e85b6210]
src/rtl_433(+0x7cf8c) [0x564b4b1d3f8c]
src/rtl_433(+0x46d6e) [0x564b4b19dd6e]
src/rtl_433(+0x472c1) [0x564b4b19e2c1]
src/rtl_433(+0x4f2d0) [0x564b4b1a62d0]
src/rtl_433(+0x1d956) [0x564b4b174956]
src/rtl_433(+0x1feeb) [0x564b4b176eeb]
src/rtl_433(+0x51d87) [0x564b4b1a8d87]
/usr/local/lib/librtlsdr.so.0(+0x6da1) [0x7f91e88d3da1]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xab66) [0x7f91e88bbb66]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xf49c) [0x7f91e88c049c]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x10d98) [0x7f91e88c1d98]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xa5a9) [0x7f91e88bb5a9]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_timeout_completed+0x19c) [0x7f91e88bc65c]
/usr/local/lib/librtlsdr.so.0(rtlsdr_read_async+0x1a3) [0x7f91e88d3ab3]
src/rtl_433(+0x53724) [0x564b4b1aa724]
src/rtl_433(+0x1ad93) [0x564b4b171d93]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f91e85970b3]
src/rtl_433(_start+0x2e) [0x564b4b172dae]
Async read stalled, exiting!

```

zuckschwerdt, 2021-03-15:
To get symbols try configuring like: `cmake -DBUILD_TESTING_ANALYZER:BOOL=ON -DCMAKE_BUILD_TYPE=Debug ..` and maybe just use to produced binary without installing (install might strip the symbols).

This whole "stuck" thing seems to be only with MQTT output and in certain states? I wonder if it's blocking TCP output?

hschimke, 2021-03-15:
To lend some info to: https://github.com/merbanan/rtl_433/issues/1662#issuecomment-797999658

I haven't had the issue occur since the last update from me, however something that I noticed when it was occurring: When I would manually kill the rtl_433 process my mqtt clients would receive a burst of messages from rtl_433. These messages would be time-stamped with around the same time that the process would hang. As the process usually hung overnight these messages would be several hours old and so were fairly obvious at the time.

zuckschwerdt, 2021-03-15:
Random idea: the stall-exit is triggered but there is still backlog in the MQTT connection, but that isn't processed as the event loop isn't running anymore -- which in turn blocks ending the event loop. I don't see an obvious way to determine when the event loop is idle ans safe to end though.

rknobbe, 2021-03-15:
Could not get it to catch with full debugging turned on, or get a usable stack trace with less than that. However, just ran for about 30 min feeding influxdb (not MQTT) inside gdb, and got this interesting trap. I'll leave the debugger up in case somebody wants me to do any poking around

```
rtlsdr_read_cb: len 262144  
sdr_handler: SDR_EV_DATA
sdr_handler: mg_mgr_poll 15
sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback                                                                                                                              =================================================================                                                                                              ==44833==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffffffeceb0 at pc 0x555555a3e635 bp 0x7ffffffecc40 sp 0x7ffffffecc30                     WRITE of size 1 at 0x7ffffffeceb0 thread T0
#0 0x555555a3e634 in _decode_v1_half /home/roger/Tools/rtl_433/src/devices/secplus_v1.c:69                                                                     
#1 0x555555a3f68a in secplus_v1_callback /home/roger/Tools/rtl_433/src/devices/secplus_v1.c:173                                                                
#2 0x555555946d18 in account_event /home/roger/Tools/rtl_433/src/pulse_demod.c:28                                                                              
#3 0x55555594a976 in pulse_demod_pcm /home/roger/Tools/rtl_433/src/pulse_demod.c:180                                                          
#4 0x55555596f244 in run_ook_demods /home/roger/Tools/rtl_433/src/r_api.c:444                                                                                  
#5 0x555555873b05 in sdr_callback /home/roger/Tools/rtl_433/src/rtl_433.c:420                                                                                  
#6 0x555555882f10 in sdr_handler /home/roger/Tools/rtl_433/src/rtl_433.c:1266                                                                                  
#7 0x555555984a1b in rtlsdr_read_cb /home/roger/Tools/rtl_433/src/sdr.c:491                                                                                    
#8 0x7ffff7211da0 in _libusb_callback (/usr/local/lib/librtlsdr.so.0+0x6da0)                                                                                   
#9 0x7ffff71f9b65  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xab65)                                                                                              
#10 0x7ffff71fe49b  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xf49b)                                                                                             
#11 0x7ffff71ffd97  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0x10d97)                                                                                            
#12 0x7ffff71f95a8  (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xa5a8)                                                                                             
#13 0x7ffff71fa65b in libusb_handle_events_timeout_completed (/lib/x86_64-linux-gnu/libusb-1.0.so.0+0xb65b)                                                    
#14 0x7ffff7211ab2 in rtlsdr_read_async (/usr/local/lib/librtlsdr.so.0+0x6ab2)                                                                                 
#15 0x555555984d9e in rtlsdr_read_loop /home/roger/Tools/rtl_433/src/sdr.c:508                                                                                 
#16 0x555555987947 in sdr_start /home/roger/Tools/rtl_433/src/sdr.c:1486                                                                                       
#17 0x5555558a14c2 in main /home/roger/Tools/rtl_433/src/rtl_433.c:1734                                                                                        
#18 0x7ffff65660b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x270b2)                                                                              
#19 0x555555870c6d in _start (/home/roger/Tools/rtl_433/build/src/rtl_433+0x31cc6d)           

Address 0x7ffffffeceb0 is located in stack of thread T0 at offset 384 in frame                                                                                     #0 0x555555a3eb3c in secplus_v1_callback /home/roger/Tools/rtl_433/src/devices/secplus_v1.c:138                                                            
  This frame has 9 object(s):
    [32, 48) 'cur_tv' (line 212)
    [64, 80) 'res_tv' (line 213)
    [96, 112) 'rolling_str' (line 365)
    [128, 144) 'fixed_str' (line 369)
    [160, 184) 'result_1' (line 139)
    [224, 248) 'result_2' (line 140)
    [288, 312) 'pin_s' (line 319)
    [352, 384) 'buffy' (line 158) <== Memory access at offset 384 overflows this variable                                                                          [416, 448) 'buffi' (line 159)
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork                                                     (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /home/roger/Tools/rtl_433/src/devices/secplus_v1.c:69 in _decode_v1_half                                      Shadow bytes around the buggy address:

0x10007fff5980: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                                                                                                
[... cut ...]
Shadow gap:              cc                                                                                                                                  ==44833==ABORTING                                                                                                                                             
 [Thread 0x7ffff62db7c0 (LWP 44833) exited]                                                                                                                    
 [Inferior 1 (process 44833) exited with code 01]                                                                                                              
 (gdb)
```

zuckschwerdt, 2021-03-16:
Thanks, yes, that looks like a overflow bug in `_decode_v1_half()` https://github.com/merbanan/rtl_433/blob/master/src/devices/secplus_v1.c#L49

zuckschwerdt, 2021-03-16:
should be fixed with 70c54ed

rknobbe, 2021-03-16:
Thanks. Applied that patch and re-ran in gdb, tripped again
````
sdr_handler: mg_mgr_poll 13                                                                                             [22/1800]
sdr_handler: mg_mgr_poll 12                                                                                                     
 sdr_handler: done mg_mgr_poll-ing
sdr_handler: calling sdr_callback

/lib/x86_64-linux-gnu/libasan.so.5(+0x6cd30) [0x7ffff75f5d30]                                                                   
 /home/roger/Tools/rtl_433/build/src/rtl_433(+0x32ec9a) [0x555555882c9a]                                                          
/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7ffff6585210]                                                                       
/home/roger/Tools/rtl_433/build/src/rtl_433(+0x4ed634) [0x555555a41634]                                                          
/home/roger/Tools/rtl_433/build/src/rtl_433(+0x3f40f9) [0x5555559480f9]

/home/roger/Tools/rtl_433/build/src/rtl_433(+0x3f7d57) [0x55555594bd57]

/home/roger/Tools/rtl_433/build/src/rtl_433(+0x41c625) [0x555555970625]

/home/roger/Tools/rtl_433/build/src/rtl_433(+0x320b06) [0x555555874b06]

/home/roger/Tools/rtl_433/build/src/rtl_433(+0x32ff11) [0x555555883f11]

/home/roger/Tools/rtl_433/build/src/rtl_433(+0x431dfc) [0x555555985dfc]

/usr/local/lib/librtlsdr.so.0(+0x6da1) [0x7ffff7211da1]                                                                          
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xab66) [0x7ffff71f9b66]                                                                  
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xf49c) [0x7ffff71fe49c]                                                                
  /lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x10d98) [0x7ffff71ffd98]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xa5a9) [0x7ffff71f95a9]
/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_timeout_completed+0x19c) [0x7ffff71fa65c]                             /usr/local/lib/librtlsdr.so.0(rtlsdr_read_async+0x1a3) [0x7ffff7211ab3]
/home/roger/Tools/rtl_433/build/src/rtl_433(+0x43217f) [0x55555598617f]
/home/roger/Tools/rtl_433/build/src/rtl_433(+0x434d28) [0x555555988d28]
/home/roger/Tools/rtl_433/build/src/rtl_433(+0x34e897) [0x5555558a2897]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7ffff65660b3]
/home/roger/Tools/rtl_433/build/src/rtl_433(_start+0x2e) [0x555555871c6e]
Async read stalled, exiting!

^C
Thread 1 "rtl_433" received signal SIGINT, Interrupt.
secplus_v1_callback (decoder=0x60e000008380, bitbuffer=0x7ffffffecf50) at /home/roger/Tools/rtl_433/src/devices/secplus_v1.c:168
168             if (search_index == -1 || (search_index + 84) > bitbuffer->bits_per_row[0]) {
(gdb) bt
#0  secplus_v1_callback (decoder=0x60e000008380, bitbuffer=0x7ffffffecf50) at /home/roger/Tools/rtl_433/src/devices/secplus_v1.c:168
#1  0x00005555559480f9 in account_event (device=0x60e000008380, bits=0x7ffffffecf50, demod_name=0x555555ad3840 <__func__.7048> "pulse_demod_pcm")
    at /home/roger/Tools/rtl_433/src/pulse_demod.c:28
#2  0x000055555594bd57 in pulse_demod_pcm (pulses=0x7ffff2efa8f8, device=0x60e000008380) at /home/roger/Tools/rtl_433/src/pulse_demod.c:180
#3  0x0000555555970625 in run_ook_demods (r_devs=0x7ffff2efa8e0, pulse_data=0x7ffff2efa8f8) at /home/roger/Tools/rtl_433/src/r_api.c:444
#4  0x0000555555874b06 in sdr_callback (
    iq_buf=0x7ffff321b800 "\216v\200j^vs\222\204\216\215\223\220w\201hkgi~z\222\216\214\217q\203hjlj\177w\226\205\215\227\201\207nrkh\200u\220\213\225\226\204\217k|hjye\214t\217\220\202\225st_q\203t\213\201\216\227\203\204njlsss\216~\225\207\213\221\203\214omvp\205p\203\202\225\235\202\206bric{r\236\216\221\212w\215~{ne{t\216\210\233\231\211\225rxfcvh\223\210\225\226\202\217oupn|r\206\204\215\220\211\207|\204r{rm\203\200\221\223\216\243\205\220eufm\200r\223\212\210\212r\203ovml~y\205\216\222\233\212\217l|igzn\226\211\220\230\205\211n"..., len=262144, ctx=0x555555f17e20 <g_cfg>)
    at /home/roger/Tools/rtl_433/src/rtl_433.c:420

#5  0x0000555555883f11 in sdr_handler (ev=0x7fffffff0050, ctx=0x555555f17e20 <g_cfg>) at /home/roger/Tools/rtl_433/src/rtl_433.c:1266

#6  0x0000555555985dfc in rtlsdr_read_cb ( iq_buf=0x7ffff321b800 "\216v\200j^vs\222\204\216\215\223\220w\201hkgi~z\222\216\214\217q\203hjlj\177w\226\205\215\227\201\207nrkh\200u\220\213\225\226\204\217k|hjye\214t\217\220\202\225st_q\203t\213\201\216\227\203\204njlsss\216~\225\207\213\221\203\214omvp\205p\203\202\225\235\202\206bric{r\236\216\221\212w\215~{ne{t\216\210\233\231\211\225rxfcvh\223\210\225\226\202\217oupn|r\206\204\215\220\211\207|\204r{rm\203\200\221\223\216\243\205\220eufm\200r\223\212\210\212r\203ovml~y\205\216\222\233\212\217l|igzn\226\211\220\230\205\211n"..., len=262144, ctx=0x60b000002fb0) at /home/roger/Tools/rtl_433/src/sdr.c:491  
                        
#7  0x00007ffff7211da1 in _libusb_callback () from /usr/local/lib/librtlsdr.so.0

#8  0x00007ffff71f9b66 in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0

#9  0x00007ffff71fe49c in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0                                                      
 #10 0x00007ffff71ffd98 in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0

#11 0x00007ffff71f95a9 in ?? () from /lib/x86_64-linux-gnu/libusb-1.0.so.0                                                      
 #12 0x00007ffff71fa65c in libusb_handle_events_timeout_completed () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
[... cut ...] 08                                                                                   
#15 0x0000555555988d28 in sdr_start (dev=0x60b000002fb0, cb=0x555555882f85 <sdr_handler>, ctx=0x555555f17e20 <g_cfg>, buf_num=0, buf_len=262144)                                                                                                                      at /home/roger/Tools/rtl_433/src/sdr.c:1486
#16 0x00005555558a2897 in main (argc=9, argv=0x7fffffffe388) at /home/roger/Tools/rtl_433/src/rtl_433.c:1734

zuckschwerdt, 2021-03-16:
Hmm? secplus_v1.c:168 does not look problematic, it's just where the code was when you hit Ctrl-C. But I wonder if it stalls because there is some endless loop in secplus_v1? That could explain the timeout and stall. I might need to add a timer/watchdog around decoder execution...

zuckschwerdt, 2021-03-16:
Well these `return 0` smell like endless `while` loop. https://github.com/merbanan/rtl_433/blob/master/src/devices/secplus_v1.c#L117
Maybe that was thought to be `-1`?

zuckschwerdt, 2021-03-16:
Can you stick a `bitbuffer_printf(bitbuffer, "%s: ", __func__);` at the start of `secplus_v1_callback()` -- then we can capture an input that reproduces this behaviour.

rknobbe, 2021-03-16:
OK, running it now...

zuckschwerdt, 2021-03-16:
Nice puzzle, yes we can exploit this decoder.
- the input must be longer than 11 bytes to keep the loop happy, but less than 130 bits
- we want `_decode_v1_half()` to return `1`, so the loop has no end-condition.
- we get that by having the first bits `0110`
- we need to trick `find_next() into giving us an offset that matches those bits (that took a while to get right)
- e.g. this puts us in an endless loop: `rtl_433 -vv -R 178 -y '18aaaaaaaaaaaaaaaaaaaaaaaaaa'`

rknobbe, 2021-03-16:
what makes it more exciting is I don't know where that secplus_v1 signal is
coming from. Must be one of my  neighbors. I'm in a suburban "target-rich"
environment for SIGINT

zuckschwerdt, 2021-03-16:
Should patched now, but that decoder definitly needs a more thorough look.

That hard-to-track bug didn't really need a genuine secplus signal, just something with a timing of roughly 500µs, enough bits and `0110` somewhere at the beginning.

Thank you all very much for sticking with this and the help on nailing this.

commit 419cb375 (2021-03-16): Fix secplus_v1 endless loop (closes #1662)
closed 2021-03-16 by zuckschwerdt