merbanan/rtl_433 #1664: Telldus weather station supported?

anderser, 2021-03-01:
Anyone know if this Telldus Weather Station is supported? https://www.kjell.com/no/produkter/hjem-fritid/termometer-og-hygrometere/tradlose-termometere/telldus-tradlos-vaerstasjon-p51557 

I have Telldus temperature sensors that work fine, but wonder if all the features of this weather station are supported by RTL433 since it is not on the supported devices list yet.

anderser, 2021-03-03:
For future reference: this weather station worked, but was reported as Cotech-367959

```
model     : Cotech-367959 ID        : 29
Battery   : 1            Temperature: 68.2 F       Humidity  : 42 %          Rain      : 0.9 mm        Wind direction: 23        Wind      : 0.0 m/s
Gust      : 0.3 m/s      Light Intensity: 69627 lux UV Index  : 251          Integrity : CRC
```

closed 2021-03-03 by anderser

zuckschwerdt, 2021-03-03:
What's the product number of the outdoor unit? I'll then add that to the docs of the Cotech decoder.

anderser, 2021-03-03:
The outdoor unit is a Telldus Weather Station. `Item: 14759. Type: FT021T`

anderser, 2021-03-03:
BTW: The UV index is as you can see in output above reported as `251` which does not fit with the Cotech decoder (- P : 8 bit: UV index (1-15))

zuckschwerdt, 2021-03-03:
Preamble, CRC init value, and length are pretty unique here, so it is likely a somewhat modified version, not a completly different thing?
If you discover the correct decoding for this we can add that. Would be best if there was a way to distinguish both types.
The "subtype" (A) and the flags (L) might be hints. https://github.com/merbanan/rtl_433/blob/master/src/devices/cotech_36_7959.c#L88

anderser, 2021-03-03:
I am not sure how to proceed with this as my C and radio knowledge are non existant. But I can try to dump some data if that helps. Here is an output for the weather station using `rtl_433 -A`

```
Detected OOK package	2021-03-03 12:48:09
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-03-03 12:48:09
model     : Cotech-367959 ID        : 29
Battery   : 1            Temperature: 41.4 F       Humidity  : 91 %          Rain      : 15.0 mm       Wind direction: 346       Wind      : 1.2 m/s       Gust      : 1.3 m/s
Light Intensity: 69627 lux                         UV Index  : 251           Integrity : CRC
Analyzing pulses...
Total count:  101,  width: 124.05 ms		(31013 S)
Pulse width distribution:
 [ 0] count:   75,  width:  500 us [476;532]	( 125 S)
 [ 1] count:   26,  width:  992 us [988;1000]	( 248 S)
Gap width distribution:
 [ 0] count:   73,  width:  472 us [448;500]	( 118 S)
 [ 1] count:   27,  width:  960 us [956;972]	( 240 S)
Pulse period distribution:
 [ 0] count:   60,  width:  976 us [976;988]	( 244 S)
 [ 1] count:   27,  width: 1464 us [1444;1492]	( 366 S)
 [ 2] count:   13,  width: 1952 us [1948;1960]	( 488 S)
Pulse timing distribution:
 [ 0] count:  148,  width:  488 us [448;532]	( 122 S)
 [ 1] count:   53,  width:  976 us [956;1000]	( 244 S)
 [ 2] count:    1,  width: 10004 us [10004;10004]	(2501 S)
Level estimates [high, low]:  15933,     20
RSSI: -0.1 dB SNR: 29.0 dB Noise: -29.1 dB
Frequency offsets [F1, F2]:   -4628,      0	(-17.7 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
view at https://triq.org/pdv/#AAB10301E803D02714808080808080808081919180908180808080908081919180808080809081808080808090819191908191808080808080808091809190819180808080908180919080818091908190808080808080808080808080808190808080808081908190808191808255
Attempting demodulation... short_width: 500, long_width: 0, reset_limit: 976, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=500,l=0,r=976'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {127} 00 29 83 a8 18 1a b4 01 2d 06 5c b7 ff f7 f6 e8 
```

anderser, 2021-03-03:
[telldus_samples.zip](https://github.com/merbanan/rtl_433/files/6076089/telldus_samples.zip)

anderser, 2021-03-03:
The rain sensor showed accumulated total since device start, so that works. Lux and UV does not change at all, so they are either not reported or wrong decoding. The panel that comes with the weather station shows some symbols that probably needs UV/lux as input, so I assume they are sent.