merbanan/rtl_433 #1667: FEAT: device class mapping file for use with home assistant discovery script

scyto, 2021-03-07:
Hi @rct i had great fun getting this all working, thanks for you comments in https://github.com/merbanan/rtl_433/issues/1541#issuecomment-766932669 , i may also fork and do a PR and do some documentation help.

One thought, given the topic names for the devices published to MQTT seems pretty deterministic how about some sort of mapping file that could be used to set device class for auto discovery.

I am thinking some sort of key pair file, for my example  
```
DSC-Security-nnnnnn:classs
e.g.
DSC-Security-3775098:motion
```
this could even be extended to create friendly name for publishing via auto-discovery like
```
DSC-Security-3775098:motion:Ground Hallway
```
(i would love to help out on code side, but that's beyond my ability :-( )

I know this isn't very elegant, but as you say, there is no way for anything else to know what the sensors are or what they should be called.

scyto, 2021-03-07:
hmm maybe i am asking for the wrong thing based on wrong expectations?

I expected to see  in MQTT for just one of my sensors the followng:

- main sensor itself (e.g. something that represents on/off - even if it can't be classified)
- battery of the sensor - i see this in discovery topic as sensor/DSC-Security-3775098-B
- tamper state of the sensor - i see this in discovery topic as binary-sensor/DSC-Security-3775098-Tamper

what i don't see is the first one in discovery - should i?
my expectation was yes i should, and that the mapping file i suggested would be way to change it from a generic device class to something specific?

I am running rlt_443 with `sudo rtl_433 -F mqtt:mqtt.server.domain:1883 -C si -M newmodel`
and the discovery script with `python3 rtl_433_mqtt_hass.py -H mqtt.server.domain -D discoverytopicname -r`

rct, 2021-03-07:
@scyto - Sorry there are a bunch of uncommitted changes to this script. While I'd like to pick up the PRs and other things people have submitted but I haven't been able to focus on it.

The quick fix is to add this to your definitions in the script to cover the closed status:

```

    # rct added 2020/12/13,
    # temp for DSC sensors, might not work well as a general attribute
    "closed": {
        "device_type": "binary_sensor",
        "object_suffix": "closed",
        "config": {
            "device_class": "door",
            "force_update": "true",
            "payload_on": "1",
            "payload_off": "0"
        }
    },
```

Note: the way this script currently works there is a single global set of mappings, so this will be the definition for any device that has a field named `closed`.  Since this is a lot less universal than say `tempc` I suspect there might be a conflict at some point.

There are a number of short comings of the current approach. There clearly needs to be some way of customizing names, providing addition per instance info, like device class, selecting which fields to use in naming etc. 

Currently I've mostly worked around the above by letting the script generate the definitions, editing them and then republishing.  I only let this script publish to a throw away topic.  I then publish the definitions I'm happy with to the topic that homeassistant is configured to watch on.

I use the MQTT Explorer desktop GUI for much of my work.

scyto, 2021-03-07:
Thanks for bringing me up to speed.

Looking at the topics in MQTT explorer all my devices report closed (of course some report 1 and some report 0).
I have 5 door, 3 motion, 1 sound and 2 vibration sensors.  

As such I get why reporting them all as door is an issue :-)

I  have yet to match the device numbers in the MQTT topics to the actual devices, hoping the numbers match what I see in dls5 :-) but get why creating one global mapping for 'closed' is an issue, and I get the manual publishing; thanks might give that a go and play.

rct, 2021-03-07:
The ESN used throughout the DSC system are hex values. The ID in rtl_433's output is the decimal version of the ESN.

For convenience, I've left the ESN field as aex string in the DSC output in rtl_433. This should make it an exact match for what you find in DLS.

scyto, 2021-03-07:
thanks, that's great.

I am watching events in MQTT as I trip sensors.  i see door sensor the closed attribute changes.

For the DSC motion sensors I only see an event, no change in the closed / open status, I see this is the same as supplied by rtl_433, is this correct?

scyto, 2021-03-07:
to go one further, two of my motion sensors generate a closed = 0 for a motion event, the other generates a closed = 1 for a motion event!

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-03-07 14:13:08
model     : DSC-Security id        : 2821677
closed    : 1            event     : 0             tamper    : 0             battery_ok: 1             xactivity : 0             xtamper1  : 0             xtamper2  : 0
exception : 0            esn       : 2b0e2d        status    : 195           status_hex: c3            Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-03-07 14:13:12
model     : DSC-Security id        : 3762043
closed    : 0            event     : 1             tamper    : 0             battery_ok: 1             xactivity : 0             xtamper1  : 0             xtamper2  : 0
exception : 0            esn       : 39677b        status    : 129           status_hex: 81            Integrity : CRC
```

rct, 2021-03-08:
The closed == 1 was also event == 0 -- In other words. that is the periodic heartbeat the device transmits so that you know it is alive and can be heard.  The heartbeats are sent event 60-85 minutes depending on the device. 

Also that device might not have been a motion sensor, it's ESN started with 2 not a 3.

gdt, 2023-09-30:
@scyto I think part of what you are asking for is still valid and a lot is improved.   Could you take a look and open fine-grained issues for specific improvements, and close this one?