merbanan/rtl_433 #1668: Auriol AFT 77 B2

Heimetli, 2021-03-07:
I just bought an Auriol AFT 77 B2, hoping that rtl_433 will recognize it.

This is not the case, and from another issue it seems to be quite different from the A1.

The debug view shows me:

Temperature 19.0
[09] {68} a5 c2 b0 20 19 00 01 a7 e0 

Temperature 19.1
[09] {68} a5 c2 b0 20 19 10 01 b6 b0 

Temperature 18.8
[09] {68} a5 c2 b0 20 18 80 01 28 10 

The temperature is in nibbles 9 - 11 and the last two bytes look like a CRC to me.
The lower nibble of the byte with the value 01 is also variable, but I can't find out what it stands for.

Question: how do you determine what CRC is used when you have an unknown protocol?

The prologue sensor in the log has nothing to do with the question, it is just another sensor in my area.

[saturday.zip](https://github.com/merbanan/rtl_433/files/6096686/saturday.zip)
.

zuckschwerdt, 2021-03-07:
`{68}` is 8.5 bytes, drop the indication and unused nibble.
```
a5 c2 b0 20 19 00 01 a7 e
a5 c2 b0 20 19 10 01 b6 b
a5 c2 b0 20 18 80 01 28 1
```
As first step I'd align the data. Look for known 8 bit fields and assume they were aligned, shift to restore that.
Otherwise look for high entropy (randomness) bits and if you find exactly 8 that should be aligned. I.e.
```
a 5c 2b 02 0 190 00 1a 7e
a 5c 2b 02 0 191 00 1b 6b
a 5c 2b 02 0 188 00 12 81
```
Second to last looks like a addition sum, last like some xor sum, maybe crc (parity).
Full data in [BitBench](https://triq.net/bitbench?c=a5c2b020190001a7e&c=a5c2b020191001b6b&c=a5c2b020188001281&c=a5c2b02013100bb73&c=a5c2b02013900c3db&c=a5c2b02014200ccee&c=a5c2b02014800d26c&c=a5c2b02015200dcb9&c=a5c2b02015500dfd2&c=a5c2b02015700e1f8&c=a5c2b02015900e32e&c=a5c2b02016100eb7f&c=a5c2b02016200ec40&c=a5c2b02016300ed55&c=a5c2b02016400ee3e&c=a5c2b02016500ef2b&c=a5c2b02016600f014&c=a5c2b02016800f2c2&c=a5c2b02016900f3d7&c=a5c2b02017000fa3d&c=a5c2b02017100fb28&c=a5c2b02017200fc17&c=a5c2b02017300fd02&c=a5c2b02017400fe69&c=a5c2b02017500ff7c&c=a5c2b020176000043&c=a5c2b020177000156&c=a5c2b020178000295&c=a5c2b020181000b3c&c=a5c2b020182000c03&c=a5c2b020189001394&c=a5c2b020196002000&c=a5c2b020202008d08&c=a5c2b02020800938a&c=a5c2b020212009d5f&c=a5c2b02021500a034&c=a5c2b02021700a21e&c=a5c2b02021800a3dd&c=a5c2b020210009b75&c=a5c2b020205009063&c=a5c2b020203008e1d&c=a5c2b020201008c37&c=a5c2b020214009f21&c=a5c2b02022100ac99&c=a5c2b02022200ada6&c=a5c2b02022800b324&c=a5c2b02022400afd8&c=a5c2b020194001e2a&c=a5c2b020186001057&c=a5c2b02015300ddac&c=a5c2b02015100db86&c=a5c2b02015000da93&c=a5c2b02014700d1af&c=a5c2b02014500cf85&c=a5c2b02014400ce90&c=a5c2b02014300cdfb&c=a5c2b02014100cbd1&c=a5c2b02014000cac4&c=a5c2b02013800c2ce&c=a5c2b02013700c10d&c=a5c2b02013500bf27&c=a5c2b02013400be32&c=a5c2b02013300bd59&c=a5c2b02013200bc4c&c=a5c2b02013000ba66&c=a5c2b02012900b38c&c=a5c2b02012800b299&c=a5c2b02012700b15a&c=a5c2b02012600b04f&c=a5c2b02012500af70&c=a5c2b02012400ae65&c=a5c2b02012300ad0e&c=a5c2b02012200ac1b&f=PRE%3Ah%20ID%3F%3A8h%20%3F%3A8h%20%3F%3A8h%20%3F%3Ah%20TEMP_C%3A12h%20HUM%3F%3A8h%20SUM%3A8h%20CRC%3F%3A8h&cf=ADD&co=4&cl=48&cw=8).

zuckschwerdt, 2021-03-07:
The final byte is a Galois LFSR-digest with bits reflected using gen 0x83, key 0xec (plain xor).
I used our https://github.com/triq-org/revdgst/ for that.

zuckschwerdt, 2021-03-07:
btw, at the same bit-length (8 here) a LFSR digest is weaker than a CRC but "crytographically secure" (whatever that means with a parameter space of ~1M, gen+key+algo). This scheme of checksum+MAC is thought to prevent integrity and tampering, but rather weak -- as shown ;)

merbanan, 2021-03-07:
@zuckschwerdt when do we get a web app for running revdgst and reveng :) ? Would be cool with a emscripten based web site. Maybe a feature for bitbench?

zuckschwerdt, 2021-03-07:
I really need to get around to put that into BitBench. The search space for everything 8-bit is so small we can likely just run plain JS on a web-worker without noticeable delays. :)

Heimetli, 2021-03-08:
Thanks a lot, great work!

You are right, the second to last byte is a sum. When the A at the beginning is discarded all my samples result in the correct sum.

I will try to verify the LFSR next.

Heimetli, 2021-03-08:
> The final byte is a Galois LFSR-digest with bits reflected using gen 0x83, key 0xec (plain xor).
I tried what I could, but had no success :-/

* Which bytes did you use as input for the algorithm?
* Is lfsr_digest8_reflect from util.c the correct funktion for that?

Heimetli, 2021-03-09:
Temperatures below zero are indicated by a sign bit:

Temperature -0.7
[19] {68} a5 c2 b0 28 00 70 01 09 40
Temperature -0.5
[19] {68} a5 c2 b0 28 00 50 00 eb e0
Temperature -0.4
[19] {68} a5 c2 b0 28 00 40 00 da b0

Heimetli, 2021-03-10:
Galois LFSR works now, I wrote my own function

Heimetli, 2021-03-12:
There is one problem left:

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-12 23:22:46
model     : Auriol AFT 77 B2                       id        : 248
Temperature: 17.50 C
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-12 23:22:46
model     : Auriol AFT 77 B2                       id        : 248
Temperature: 17.50 C
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-12 23:22:46
model     : Auriol AFT 77 B2                       id        : 248
Temperature: 17.50 C
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-12 23:22:46
model     : Auriol AFT 77 B2                       id        : 248
Temperature: 17.50 C
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-12 23:22:46
model     : Auriol AFT 77 B2                       id        : 248
Temperature: 17.50 C
*** signal_start = 18910394, signal_end = 19060522, signal_len = 150128, pulses_found = 390
Iteration 1. t: 121    min: 121 (20)    max: 122 (370)    delta 4
Iteration 2. t: 23946    min: 0 (0)    max: 47893 (390)    delta 2282083082
Iteration 3. t: 23946    min: 47893 (390)    max: 0 (0)    delta 292511602
Iteration 4. t: 23946    min: 47893 (390)    max: 0 (0)    delta 0
Distance coding: Pulse length 23946

Short distance: 121, long distance: 243, packet distance: 487

p_limit: 23946
bitbuffer:: Number of rows: 25 
[00] { 0}                            : 
[01] { 0}                            : 
[02] { 0}                            : 
[03] { 0}                            : 
[04] { 0}                            : 
[05] { 0}                            : 
[06] { 0}                            : 
[07] { 0}                            : 
[08] { 0}                            : 
[09] {68} a5 cf 80 20 17 50 0c cb 70 
[10] { 0}                            : 
[11] { 0}                            : 
[12] { 0}                            : 
[13] { 0}                            : 
[14] { 0}                            : 
[15] { 0}                            : 
[16] { 0}                            : 
[17] { 0}                            : 
[18] { 0}                            : 
[19] {68} a5 cf 80 20 17 50 0c cb 70 
[20] { 0}                            : 
[21] { 0}                            : 
[22] { 0}                            : 
[23] { 0}                            : 
[24] {68} a5 cf 80 20 17 50 0c cb 70 
... Maximum number of rows reached. Message is likely truncated.

How to get rid of the rows with 0 bits?

[https://github.com/Heimetli/rtl_433](https://github.com/Heimetli/rtl_433)

merbanan, 2021-03-13:
The signal does not match the row splitter well. But it does not matter as we have a mic to check with.

With the following parameters I only get one message in the output:

'n=name,m=OOK_PPM,s=488,l=976,g=988,r=1964'

merbanan, 2021-03-13:
Set reset at 2500 and it should be fine.

closed 2021-03-13 by Heimetli