merbanan/rtl_433 #1669: rtl_433 on RPi service does not start on boot

bkenobi, 2021-03-09:
**System setup:**
RPi 3 with current Raspbian Buster
Realtek RTL2838 USB adapter (EZCAP EzTV-668 RTL2832U / FC0013 USB DVB-T)
rtl-sdr 11/1/2020 (installed today from git://git.osmocom.org/rtl-sdr.git)
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
udev rule installed
blacklist drivers installed

**Issue**
I've been using rtl_433 for a few months and it works great.  I wanted it to run as a service on my RPi HA system so I could start utilizing NodeRed and Grafana.  When I set up a systemd service I can get it to work every time if rtl_433 is already running before launching.  If I reboot the system does not receive any data.  Since this system is running 24/7 and power failures do happen I need the system to start up unattended.  Currently I have to log in, stop the service, either run rtl_433 1-5 times until I see data is being received or unplug the SDR and plug it back in, then relaunch the service.

This issue sounds similar to others that have been closed but I do not see resolution to the issue (e.g. 1586).  I have run through many tutorials that include sections on installing blacklists for drivers as well as udev rules.  All of these seem to have the same list:

/etc/modprobe.d/blacklist-rtl.conf
```
blacklist dvb_usb_rtl28xxu
blacklist rtl2832
blacklist rtl2830
```

I installed the udev rules for rtl-sdr which is a pretty long list.  The pertinent part for my SDR is idVendor == 0bda:
```
# original RTL2832U vid/pid (hama nano, for example)
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2832", ENV{ID_SOFTWARE_RADIO}="1", MODE="0660", GROUP="plugdev"

# RTL2832U OEM vid/pid, e.g. ezcap EzTV668 (E4000), Newsky TV28T (E4000/R820T) etc.
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2838", ENV{ID_SOFTWARE_RADIO}="1", MODE="0660", GROUP="plugdev"
```

I can provide lsmod and udevadm output or anything else that might help narrow down the issue.  However, I have not located anything that is different between when rtl_433 is correctly receiving data and when it is not.

I do not believe this is directly a service issue since it takes 1-5 times of launching rtl_433 for data to be visible.  I believe this must be a driver or kernel issue but I have not located the issue yet (otherwise this would be a bug report and not a help request).

I have used a simple service file and tried the more robust one in the issue 1651 with the same results.
https://github.com/merbanan/rtl_433/issues/1651

Example fresh session:
```Using username "pi".
Linux Raspi3 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Mar  8 19:02:00 2021 from 192.168.0.182

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@Raspi3:~ $ rtl_433
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
^CSignal caught, exiting!
pi@Raspi3:~ $ rtl_433
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
^CSignal caught, exiting!
pi@Raspi3:~ $ rtl_433
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
^CSignal caught, exiting!
pi@Raspi3:~ $ rtl_433
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
^CSignal caught, exiting!
pi@Raspi3:~ $ rtl_433
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-08 19:17:04
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     temperature: 40.6 F       humidity  : 79 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-08 19:17:04
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     temperature: 40.6 F       humidity  : 79 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-08 19:17:04
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     temperature: 40.6 F       humidity  : 79 %
Integrity : CHECKSUM
^CSignal caught, exiting!
pi@Raspi3:~ $
```

merbanan, 2021-03-09:
Yeah, that might be the case. Try adding -g 20 or something lower.

bkenobi, 2021-03-09:
I see "Fitipower FC0013 tuner" every time whether it's receiving data or not.  I believe I read somewhere that that's the device it's detecting even though I'm using a different SDR.  I tried unplugging and reinstalling the SDR to see if there would be any difference as others have reported.  It appears that does fix it for that session, but I'm worried about restarting after a power failure.  This is what I'm seeing with udevadm info on boot and after removing and installing the SDR:
[image]
I performed a diff to make the change obvious.  I plugged the SDR into the same port even though the DEVNUM changes.

bkenobi, 2021-03-09:
I have been leaving the gain as default which I gather is "0" when left to auto.  I tried using -g 20 and got nothing.  I tried setting the gain to 100 and it appears to limit to 19.7 anyway.  I set the gain to 10 and it used 17.9.  When I ran rtl_433 a 3rd time without any gain, the sensors started picking up.  However, my experience has been that if I run the basic command without any extra switches several times, it starts working on its own.

```
pi@Raspi3:~ $ rtl_433 -g 20
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to 19.700000 dB.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
^CSignal caught, exiting!
pi@Raspi3:~ $ rtl_433 -g 100
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to 19.700000 dB.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
^CSignal caught, exiting!
pi@Raspi3:~ $ rtl_433 -g 10
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to 17.900000 dB.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
^CSignal caught, exiting!
pi@Raspi3:~ $ rtl_433
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 08:28:28
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 135.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 08:28:28
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 135.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 08:28:28
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 135.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
^CSignal caught, exiting!
```

zuckschwerdt, 2021-03-09:
USB V/P 0BDA:2838 should be RTL2832U/FC0012 or RTL2832U/E4000. So I guess that output from librtlsdr is correct.

bkenobi, 2021-03-09:
Is there anything else I can provide that might help diagnose why it doesn't load correctly on boot?  If I unplug and reinstall it's pretty consistent that it works so it seems to me that it must be something detected on boot.  If so, then if I know what's not set correctly in software a software command should be able to potentially get it aligned rather than requiring physically interfacing with the hardware.

zuckschwerdt, 2021-03-09:
Start with `-w /tmp/grab_$(date +%s).cu8 -T 60` to grab 60-second sample files. Do this at different stages (with boot, after boot, manual later) and compare what you get at https://triq.org/iqs/

bkenobi, 2021-03-09:
I'll give it a try.  I believe it will be empty files until it works because I think it's a driver issue.  Doesn't hurt to confirm either way!

bkenobi, 2021-03-09:
I rebooted the system after disabling the service.  In order to give it the best chance for success, i unplugged the SDR and then reinstalled it.  I confirmed that there were no messages in dmesg that would be problematic and then I ran with the above arguments several times.  The first couple runs produced purple snow on the analyzer.  The third data set actually received some data that was processed.

final run with output processed:
```
pi@Raspi3:~/rtl_test $ rtl_433 -w grab_$(date +%s).cu8 -T 60
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:45:53
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 180.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:45:53
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 180.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:45:53
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 180.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:11
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     temperature: 57.4 F       humidity  : 49 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:11
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     temperature: 57.4 F       humidity  : 49 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:11
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     temperature: 57.4 F       humidity  : 49 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:29
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 180.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:29
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 180.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:29
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 180.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:47
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     temperature: 57.6 F       humidity  : 48 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:47
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     temperature: 57.6 F       humidity  : 48 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 12:46:47
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     temperature: 57.6 F       humidity  : 48 %
Integrity : CHECKSUM
Time expired, exiting!
```

dmesg:
[dmesg.txt](https://github.com/merbanan/rtl_433/files/6111347/dmesg.txt)

plots from spectrogram:
[60sec tests.pdf](https://github.com/merbanan/rtl_433/files/6111348/60sec.tests.pdf)

merbanan, 2021-03-09:
The issue could be the tuner not resetting properly. This is most likely a hardware issue, try and see if it works the same with rtl_tcp or some other app.

bkenobi, 2021-03-09:
I tried rtl_tcp and it appears after reboot it works immediately:
```
pi@Raspi3:~ $ rtl_tcp -a 192.168.0.200
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN:

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0013 tuner
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=192.168.0.200:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
client accepted! Galaxy-A10e.routerlogin.net 50662
Allocating 15 zero-copy buffers
set direct sampling 0
Disabled direct sampling mode
set freq correction 0
set gain mode 0
set freq 100000000
set sample rate 1024000
^CSignal caught, exiting!
comm recv bye
Signal caught, exiting!
worker cond timeout
Signal caught, exiting!
all threads dead..
listening...
Use the device argument 'rtl_tcp=192.168.0.200:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
```
When I connect SDR Touch to it and tune to a local radio station it comes in loud and clear.  If I fire up rtl_433 afterward I still get nothing decoded.  I can try running the same capture procedure as before if that would be informative.  I guess I would:

1) reboot
2) run capture
3) run rtl_tcp and confirm function
4) run capture
5) repeat until it works

bkenobi, 2021-03-09:
Based on the steps posted above, this is the results:
1) reboot
2) run capture - no data decoded
3) run rtl_tcp - radio station tuned in loud and clear
4) run capture - no data decoded
5) run capture - data decoded (see below)

```
pi@Raspi3:~ $ rtl_433 -w grab_$(date +%s).cu8 -T 60
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:03:35
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:03:35
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:03:35
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:03:53
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 4.3 km/h     temperature: 58.3 F       humidity  : 44 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:03:53
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 4.3 km/h     temperature: 58.3 F       humidity  : 44 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:03:53
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 4.3 km/h     temperature: 58.3 F       humidity  : 44 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:04:11
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 3.5 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:04:11
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 3.5 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:04:11
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 3.5 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:04:29
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     temperature: 58.1 F       humidity  : 43 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:04:29
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     temperature: 58.1 F       humidity  : 43 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 14:04:29
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     temperature: 58.1 F       humidity  : 43 %
Integrity : CHECKSUM
Time expired, exiting!
```
[60sec tests_1.pdf](https://github.com/merbanan/rtl_433/files/6111743/60sec.tests_1.pdf)

bkenobi, 2021-03-10:
@merbanan Since it appears rtl_tcp works immediately upon boot and rtl_433 doesn't, is there a difference in the way the sdr driver used?  I'm assuming they should be similar which is the reason for suggesting trying the other tool.  Can I help test anything else to determine the difference between these or another similar tool?

Is there any chance that power could be a problem?  This is running on an RPi3 with a 2.5A psu but it does have a usb-microSD adapter, CM15A (X10 controller), and APC UPS plugged into the other USB ports.  The microSD is not mounted and the UPS is only for communicating battery state called once a minute.  I can check the draw on each component with my usb power meter and source a powered hub if this could be a contributing factor.

zuckschwerdt, 2021-03-10:
Just an "empty file" can't happen, as rtl_433 would notice that no I/Q samples are coming in an abort with "Async read stalled, exiting". All files are filled with good data, i.e. there is some noise around a uniform color.

But the purple means it's nearly silence or deaf. Everything digital seems to be working, but the analog part maybe not.
The only difference here is that you start rtl_tcp with 100M, 1024k and rtl_433 at 433.92M, 250k.
Also "picking a radio station" you reset the frequency. Maybe that's bringing your receiver to live?

- Try a run of test with `-s 1024k`
- Also try a run of tests with `-f 433.92M -f 434M -f 433.85M -H 5`

both combined with `-w grab_$(date +%s).cu8 -T 60` to visualize results.

bkenobi, 2021-03-10:
The first command received data correctly.  I forgot to copy the output before rebooting but suffice to say it looks the same as I've posted before with the Acurite 5in1 and the Blueline power meter reader.  The second command looked a bit different:

```
pi@Raspi3:~/rtl_test $ rtl_433 -w grab_$(date +%s).cu8 -T 60 -f 433.92M -f 434M -f 433.85M -H 5
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Tuned to 434.000MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 434000000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 433.850MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 433850000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 433920000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 434.000MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 434000000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 433.850MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 433850000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 433920000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 434.000MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 434000000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 433.850MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 433850000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 433920000                        frequencies: 433920000, 434000000, 433850000
hop_times : 5
Allocating 15 zero-copy buffers
Time expired, exiting!
```
[60sec tests_2.pdf](https://github.com/merbanan/rtl_433/files/6113553/60sec.tests_2.pdf)

It appears that whatever the "-s 1024k" command did may be important here.  I'll try adding that to my service and reboot a few times to see if it fixes things.

zuckschwerdt, 2021-03-10:
Ok, so:
- changing frequency -> does not help.
- higher sample rate -> works.

Open question, do you just need a high sample rate once, to get your receiver going?
E.g. run at boot with
```
rtl_433 -s 1024k -T 5
rtl_433
```
(run for 5 seconds at high sample rate, then run normal.)

bkenobi, 2021-03-10:
I tried with your suggestion and it didn't pick up any sensors for some reason.

```
pi@Raspi3:~/rtl_test $ rtl_433 -w grab_$(date +%s).cu8 -T 5 -s 1024k
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Sample rate set to 1024000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1024000 Hz at cutoff 102400 Hz, 9.8 us
Time expired, exiting!
pi@Raspi3:~/rtl_test $ rtl_433 -w grab_$(date +%s).cu8 -T 60
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Time expired, exiting!
pi@Raspi3:~/rtl_test $
```
As you'd expect, both end up with purple plots.

I took a step back and ran the original grab several times in a row until I got data.  I don't know if the output helps, but since it appears the "-s 1024k" seems to have been a fluke/not repeatable I figured I'd try a previous suggestion again.

```
pi@Raspi3:~/rtl_test $ rtl_433 -w grab_$(date +%s).cu8 -T 60
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Time expired, exiting!
pi@Raspi3:~/rtl_test $ rtl_433 -w grab_$(date +%s).cu8 -T 60
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
[... 81 lines cut ...]
time      : 2021-03-09 22:37:30
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     wind_dir_deg: 202.5       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 22:37:48
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 0.0 km/h     temperature: 33.6 F       humidity  : 99 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 22:37:48
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 0.0 km/h     temperature: 33.6 F       humidity  : 99 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-09 22:37:48
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 0.0 km/h     temperature: 33.6 F       humidity  : 99 %
Integrity : CHECKSUM
Time expired, exiting!
```

zuckschwerdt, 2021-03-10:
Yes, even the "purple" samples look like good data. It's just that gain is too low. No clue what triggers the automatic gain to suddenly work after a few restarts.
Maybe on startup run something like
```
rtl_433 -T 3
rtl_433 -T 3
rtl_433 -T 3
rtl_433 -T 3
rtl_433
```
Or get a R820T(2) receiver…

bkenobi, 2021-03-10:
Is automatic gain control internal to the SDR?  If it has no external control, then it seems as though this could be why some people are seeing issues and others don't.  I'll run these tests shortly and see.  A dirty script to perform the first set could be put in the pre execute part of a service and then launch the actual run of rtl_433.

I'm hoping there's a cleaner way but this might be a good stop gap fix.

bkenobi, 2021-03-10:
I created a script with your suggestion and ran with -T 3 a total of 10 times and then ran without the last time.  After attempt 4 or 5 it started to pick up sensor data.  I can't say when it actually started decoding because it depends on if one of my sensors transmits in the 3 second window.  This isn't the ideal solution, but it seems to work in a single test.

I'll have to add the script (without the last line) to my PreExecute on the service to see if it works after a few reboots.

I see that I can use the -g option to control gain.  Is there a way to see what the gain is set to when working versus not?  If the auto option is not starting up correctly, perhaps I should just specify the appropriate value?

bkenobi, 2021-03-10:
fwiw, here's the results of using your recommendation.  It looks to me like this time the SDR has the gain set correctly starting on the second instance of the command.
[60sec tests_4.pdf](https://github.com/merbanan/rtl_433/files/6118980/60sec.tests_4.pdf)

```
pi@Raspi3:~/rtl_test $ cat rtl_test.sh
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -T 3 -w grab_$(date +%s).cu8
rtl_433 -w grab_$(date +%s).cu8
pi@Raspi3:~/rtl_test $ sh rtl_test.sh
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Time expired, exiting!
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Time expired, exiting!
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Time expired, exiting!
rtl_433 version 20.11-85-g8efd62bb branch master at 202103071801 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Found Fitipower FC0013 tuner
[... 142 lines cut ...]
time      : 2021-03-10 13:28:50
model     : Acurite-5n1  message_type: 56          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 4.3 km/h     temperature: 61.9 F       humidity  : 47 %
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-10 13:29:08
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 0           battery   : 1
wind_speed: 4.3 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-10 13:29:08
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 1           battery   : 1
wind_speed: 4.3 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-03-10 13:29:08
model     : Acurite-5n1  message_type: 49          id        : 2825
channel   : A            sequence_num: 2           battery   : 1
wind_speed: 4.3 km/h     wind_dir_deg: 270.0       Rainfall Accumulation: 32.08 in
Integrity : CHECKSUM
^X^CSignal caught, exiting!
````

Vedeneb, 2021-06-24:
Hi,
is there any update on this? Could you solve the problem? I have exactly the same issue with the FC0013 Tuner

DanielHabenicht, 2021-08-22:
I also got the same problem, RPI4 with a RTL2832U (FC0012) Stick. 
Only after reboot.

zuckschwerdt, 2021-08-22:
Since at least part of the problems seems inproper gain on cold start, you may want to test this:
`rtl_433 -M level -Y autolevel -Y minlevel=-30 -Y magest`
That will report levels, adjust the detection level (default: -12dB), lower the initial level, and use a more precise estimator.

DanielHabenicht, 2021-08-22:
> Since at least part of the problems seems inproper gain on cold start, you may want to test this:
> `rtl_433 -M level -Y autolevel -Y minlevel=-30 -Y magest`
> That will report levels, adjust the detection level (default: -12dB), lower the initial level, and use a more precise estimator.

For me, it's the same behaviour as before. I have to execute the command three times before it reads back values. 
For the use case, I now always execute the command for 60 seconds and then restart.

bkenobi, 2021-12-23:
I've noticed sensors drop recently though I don't know if it's related.  I was using an old version of rtl_433 so I've updated and replaced batteries in one suspect device.  The Accurite 5in1 has solar and batteries for night.  I'll have to see if the data loss starts at night or if it's at all times.  I suppose it's possible that even without the system reporting low batteries the long nights are eating them more than the solar is helping.

loca5790, 2022-04-18:
having what appears to be the same issue on Ubuntu 22.04 with 2 different RTL2832U sticks

zuckschwerdt, 2022-04-18:
libusb error -6 would be LIBUSB_ERROR_BUSY. I guess the descriptor is being kept open for some short time?
Is it possible to delay restart by a few seconds ("cool down")?

gdt, 2023-09-30:
A fix for FC0012 was merged.  There was talk of just applying it for FC0013 also but I don't see that in the current code.  The librtlsdr PR is still pending (I just pinged it).

@bkenobi Can you add code to apply the fix for FC0013 also (see sdr.c) and test?  Or if that's too much, update to latest rtl_sdr git master and test?

Are we ok with just applying the workaround for FC0013 also?  From this PR we know it's buggy; we just aren't 100% sure the workaround is right, is my read.

gdt, 2023-09-30:
@bkenobi See https://github.com/merbanan/rtl_433/commit/42cb90c305a9dd050111b3bfa3cbe9a9ae6795bb

bkenobi, 2023-10-25:
Is that committed to the main trunk or do I need to compile one of the debug ones?  I can give it a try when I know which I need to use.

zuckschwerdt, 2023-10-25:
It's in master since 2023-03-12 #2417

bkenobi, 2023-12-13:
I updated the code and at least on the first attempt the SDR connected immediately without my workaround starts.  I'll continue to monitor and report if it does not fix the problem completely.  Thanks!

gdt, 2024-06-03:
Is this fixed?  I'm having a hard time following. I think
   - fix merged to master
   - no evidence that it's still broken
and thus we should close this.

7 day timeout; speak up if that's wrong.

bkenobi, 2024-06-03:
The fix (#2417) did not work for me. I added back in a manual set of restarts which seem to reliably prime the gain controls and force the SDR to function.

gdt, 2024-06-03:
Given that there was a fix, I'd like to close this one, as it seems more likely than not that what you are still seeing is a different hardware bug.   Could you open a fresh issue with your residual problems?

I wonder if you are able to do something where you capture the data from the priming runs, and see if it is ok vs low-gain, as part of diagnosing this.