merbanan/rtl_433 #1670: Schrader TPMS (Setup as Ford 13-16 F450) reports low readings over ~50 psi

dunarridev1, 2021-03-12:
Gentlemen, im using schrader 35000 sensors, which can be setup to emulate almost any vehicle made using 315 or 433 mhz.  They are the most popular sensor used by tire shops installing new sensors.

Im having an issue.  Over 50 or so psi, the sensors start to give the wrong values.  It seems to read fine up to around 55 or so, then start showing readings of 20 psi and increasing from there as i pressurize
[g002_315M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/6126625/g002_315M_250k.cu8.zip)

 up to 80.  Schrader confirms the sensors are good up to 80 psi.

Sensors setup for 2013-2016 Ford F350 using a Baratec tech 200 Sensor Activation tool.
315 Mhz

I have recorded one of the signals around 60-70 psi....it is attached.

merbanan, 2021-03-14:
Ok, I think I know what might happen.

The message data from the file is the following:
{64} 30 00 92 43 [45] 4d 25 bc

[] is what we believe is the pressure. 0x45 -> 17.55 PSI with current code. If the pressure is one more bit we get 0x145 which then would result in ~-> 81.55 PSI which might be correct in this case.

So the question is then where the last bit is located in the message? My guess is here:
{64} 30 00 92 4[3] 45 4d 25 bc
or here:
{64} 30 00 92 43 45 4d [25] bc

What we need to figure this out is some signals samples that are as close as possible above and below the limit where the error occur. Ideally we need lots of readings and the corresponding reading from an official decoder.

dunarridev1, 2021-03-15:
So would it help if i did a bunch of readings from 60-80 psi and had rtl record them......then post what i got here?

--John

merbanan, 2021-03-15:
Yes, we need to observe what happens with the message when rtl_433 goes from being able to decode the message to not being able to.

dunarridev1, 2021-03-15:
Here you go, i captured a bunch between 60 and 82 psi.  The baseline this starts at seems to be ~58  psi...so hence, if you add the number being output to ~58 psi you get the correct psi of the tire.

[rtl_ford_tpms_error.zip](https://github.com/merbanan/rtl_433/files/6145079/rtl_ford_tpms_error.zip)

dunarridev1, 2021-03-15:
And here is some readings BELOW the error range....
[rtl_ford_tpms_LOW-error.zip](https://github.com/merbanan/rtl_433/files/6145131/rtl_ford_tpms_LOW-error.zip)

merbanan, 2021-03-16:
Fixed in 8038d62

closed 2021-03-16 by merbanan