merbanan/rtl_433 #1672: Does not auto detect frequency when reading file

kroeckx, 2021-03-13:
I've saved the output from my device using `rtl_443 -f 868M -S known`. If I then do `rtl_433 -r g002_868M_1000k.cu8` it's not detecting anything, but if I add -f 868M, it does work.

`rtl_433 -r g002_868M_1000k.cu8 ` gives:
```
rtl_433 version 20.11-87-gad55a535 branch master at 202103101351 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/kurt/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Test mode active. Reading samples from file: g002_868M_1000k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

While `rtl_433 -f 868M -r g002_868M_1000k.cu8` gives:
```
rtl_433 version 20.11-87-gad55a535 branch master at 202103101351 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/kurt/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 150 out of 180 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-180 ]
Test mode active. Reading samples from file: g002_868M_1000k.cu8
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.211152s
model     : Bresser-5in1 id        : 17
Battery   : 1            Temperature: 11.1 C       Humidity  : 71            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 0.0           Rain      : 47.4 mm
Integrity : CHECKSUM
```

The documentation seems to suggest that it would get the parameters from the filename and that I don't need to pass the -f 868M when reading the file.

zuckschwerdt, 2021-03-13:
Someday we need to add that, yes. Note that not the frequency is the problem here but rather the sample rate. You acutally need `-s 1000k` -- which `-f 868M` implicitly sets.

merbanan, 2021-03-16:
-f 868M implies -Y minmax also.

gdt, 2023-09-29:
dup of #1331

closed 2023-09-29 by gdt