merbanan/rtl_433 #1673: Support for Bresser Professional Rain Gauge

kroeckx, 2021-03-13:
Hi,

I have this device: https://www.bresser.de/en/Weather-Time/BRESSER-Professional-Rain-Gauge.html, article number 7002530

And it seems to be detected, but I'm not sure everything is correct. It currently shows me:
```
time      : @0.045663s
model     : Bresser-5in1 id        : 17
Battery   : 1            Temperature: 11.0 C       Humidity  : 71            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 0.0           Rain      : 47.4 mm
Integrity : CHECKSUM
```

The temperature matches what is on the display of the device. The device does not show outdoor humidity, but 71% is possible comparing it to official measurements.

The device does not measure anything related to wind, so I would prefer that it didn't claim 0 for them.

But I'm wondering if the rain value is correct or not. If I press the "since" button on the device, it shows me 368.2 mm.

kroeckx, 2021-03-13:
This seem to be the values I get:
[00] {284} ff ea aa aa aa aa 2d d4 ea ee 46 ff ff ff ef ee fe 8e 8b eb ff 15 11 b9 00 00 00 10 11 01 71 74 14 00 00 00
[00] {284} ff fa aa aa aa aa 2d d4 ea ee 46 ff ff ff ef ee fe 8e 8b eb ff 15 11 b9 00 00 00 10 11 01 71 74 14 00 00 00
[00] {284} ff aa aa aa aa aa 2d d4 ea ee 46 ff ff ff ef ee fe 8e 8b eb ff 15 11 b9 00 00 00 10 11 01 71 74 14 00 00 00
[00] {284} ff fa aa aa aa aa 2d d4 ea ee 46 ff ff ff ef f6 fe 8e 8b eb ff 15 11 b9 00 00 00 10 09 01 71 74 14 00 00 00

Note that following the comment, byte 14 is "11", and I have a "b9" in the in byte 15 where all examples have "00" or "80".

peteakalad, 2021-03-13:
It'll be the same as the bresser 5in1 I have.  The rain amount will be the amount since the rain guage was reset (theres a button on the bottom).  The receiver works out the difference/delta of this reported amount over time.

Check to be sure the rtl433 rain value increments the same amount as the receiver over time.

kroeckx, 2021-03-13:
I was planning to check that, but the receiver didn't report any change yet.

kroeckx, 2021-03-13:
It keeps reporting the humidity at 71, all the samples I have so far say 71. I've put a different device outside and it returned 54. I suspect it's not actually measuring humidity since the device inside doesn't report it, and the value in the message is always 71.

merbanan, 2021-03-13:
@kroeckx 

From the documentation:

Humidity inside
Outdoor temperature (in °C or °F)
Precipitation measurement

So the outdoor unit has only 2 sensors. And the protocol is reused from other products.

Your observation regarding byte 15 looks valid. I'll accept a change which uses the low nibble to select the sensor type. The high nibble top bit is set in a few payload examples. So I think that if the low nibble is 9 then we can output the the rain and temp values. Anything else and it is assumed to 5in1 sensor.

Can you supply a recording of this sensor? rtl_433 -S known

kroeckx, 2021-03-13:
[g001_868M_1000k.cu8.gz](https://github.com/merbanan/rtl_433/files/6135341/g001_868M_1000k.cu8.gz)

kroeckx, 2021-03-13:
I've now also saw one that has 74:
[00] {284} ff ea aa aa aa aa 2d d4 e9 ee 46 ff ff ff ef 99 ff 8b 8b eb ff 16 11 b9 00 00 00 10 66 00 74 74 14 00 00 00 
[g010_868.3M_1000k.cu8.gz](https://github.com/merbanan/rtl_433/files/6135621/g010_868.3M_1000k.cu8.gz)

kroeckx, 2021-03-14:
It's been raining very slowly the last several hours, but it wasn't counting anything, so I went to the device to check it out. I probably flipped the cup used to hold the water twice. The indoor device now shows 2 mm rain fall today, total 370.2. My guess is that the resolution is only 1 mm. This report at least seems to suggest the same: https://github.com/andreafabrizi/BresserWeatherCenter/issues/3

The communication I got now is:
[00] {284} 00 aa aa aa aa aa 2d d4 ed ee 46 ff ff ff ef 9f ff 8b 7d eb ff 12 11 b9 00 00 00 10 60 00 74 82 14 00 00 00 

Which rtl_433 translates as 48.2

kroeckx, 2021-03-14:
After a few more samples, it seems the resolution is 0.25. You need to multiply the value you have now by 2.5.

merbanan, 2021-03-14:
Does https://github.com/merbanan/rtl_433/pull/1676 do what you want ?

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.065339s
model     : Bresser-ProRainGuage                   id        : 17
Battery   : 1            Temperature: 11.1 C       Rain      : 118.5 mm      Integrity : CHECKSUM

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.080108s
model     : Bresser-ProRainGuage                   id        : 17
Battery   : 1            Temperature: 6.6 C        Rain      : 118.5 mm      Integrity : CHECKSUM

```

Output based on the 2 samples you provided.

kroeckx, 2021-03-14:
It looks good to me.

merbanan, 2021-03-14:
Closed by be15f27

closed 2021-03-14 by merbanan