merbanan/rtl_433 #1677: Frequent "gt_tmbbq05_decode: Bad checksum" errors

waded, 2021-03-16:
Running `rtl_433 -C customary -F "<redacted MQTT config>"` version 20.11-87-gad55a535 branch master at 202103101351 inputs file rtl_tcp RTL-SDR I frequently see errors like the following:

gt_tmbbq05_decode: Bad checksum (1e) {32} 3f 01 51 56 : 00111111 00000001 01010001 01010110
gt_tmbbq05_decode: Bad checksum (26) {32} 3f 01 42 d0 : 00111111 00000001 01000010 11010000
gt_tmbbq05_decode: Bad checksum (2a) {32} 3f 01 3c 86 : 00111111 00000001 00111100 10000110
gt_tmbbq05_decode: Bad checksum (2d) {32} 3f 01 3a dc : 00111111 00000001 00111010 11011100

I'm not certain whether there's a Quigg or compatible thermometer in the area, and this is something wrong with the parser, or it's something else. If you'll let me know what additional debug information you'd like to see beyond the above I'll see if I can pick it up again.

zuckschwerdt, 2021-03-16:
Thanks! That's expected in decoders, there is always some stray signal getting bounced at the sanity checks. You should only see those messages when debugging though. Fixed now.

closed 2021-03-16 by zuckschwerdt

zuckschwerdt, 2021-03-16:
this is what other decoders might think about that signal:
```
Badly formatted OS v2.1 message
Unrecognized Msg in OS v3
wt450_callback: wrong size of bit per row 32
Fineoffset_WH0290: short package. Row length: 32. Header index: 56
philips_aj7010_decode: wrong number of bits (32)
norgo_decode: wrong size of bit per row 32
fineoffset_wh1080: short package. Header index: 56
secplus_v2_callback: DECODE_FAIL_SANITY
lacrosse_breezepro_decode: Wrong packet length: 32
lacrosse_wr1_decode: Packet too short: 32 bits
lacrosse_th_decode: Packet too short: 32 bits
lacrosse_r1_decode: Packet too short: 32 bits
secplus_v1_callback:return  DECODE_ABORT_LENGTH
```

waded, 2021-03-16:
Thanks. I'm running a build off c32bc39 now. With the change you made I expect I'll just not see the bad checksum error anymore, which is fine with me - the weather station I'm looking for is unaffected.

I do wonder if it's possible that is some sort of wireless barbeque thermometer - one of my neighbors runs a smoker all the time. Next I talk with him I'll ask him if he's got one, and what model.

zuckschwerdt, 2021-03-16:
Yes, the codes look like a temperature gauge of sorts. Try running with `-A` to capture more (timing) info if you like. That's a better indication of what we are looking at.

commit c32bc396 (2021-03-16): minor: fix gt_tmbbq05 verbose output (closes #1677)