merbanan/rtl_433 #1679: Kinda Fixed in 8038d62 - PSI is off

dunarridev1, 2021-03-16:
Now works over 55 psi, but reports readings a bit too high.  I took some samples, and took PSI readings, and named the files appropriately, so you know what PSI it should be reporting. (Should have done this last time for you, sorry).

Fixed in 8038d62

_Originally posted by @merbanan in https://github.com/merbanan/rtl_433/issues/1670#issuecomment-800569812_

merbanan, 2021-03-16:
Well the code has this comment:

P = Pressure, maybe PSI scale 0.25?

rtl_433 -r ../signals/schrader/rtl_error/80_psi.cu8 ->

Pressure  : 85.80 PSI    Temperature: 74.0 F       code      : 564a25        Integrity : CHECKSUM

So we take the the first byte in the code field 56 and add the 1 because the 2 is set.
0x156 should give 80
0x14e should give 77.7
0x127 should give 71.7

Plotting this didn't give a linear relation.
 
https://docs.google.com/spreadsheets/d/1Sosmw9eOzyW6wMEcmAeNYpemcboC7NYWf5Pu3SXJSc0/edit?usp=sharing

So to understand how this is done we need more values. If you get lets say 20-40 readings in the 0-80 range we should be able to get a better understanding on how to decode the pressure value.

dunarridev1, 2021-03-16:
[rtl_error.zip](https://github.com/merbanan/rtl_433/files/6152717/rtl_error.zip)

Here's 21 readings.  

To do each reading im running rtl_433, and as soon as i see a signal releasing the trigger, quiting RTL, and renaming the files to the PSI reading.  Then repeating.  Due to time it takes me to release the air trigger, the readings will not be perfect, but very close.  As you said, the more readings the better.  I hope 21 is good for you.  If not ill get you more.

merbanan, 2021-03-17:
I matched the PSI calculation to the curve. Closed with 68f255f

closed 2021-03-17 by merbanan