merbanan/rtl_433 #1680: rtl_433 and WGR800 wind sensor

domotic-29, 2021-03-17:
I want to get WGR800X data from rtl_433, but I don't see anything.
However, apparently you worked to implement it (
[https://github.com/merbanan/rtl_433/issues/1026](https://github.com/merbanan/rtl_433/issues/1026) ).

I currently receive those data using rfxcom device and heyu software.
The "raw" data shown by heyu are "ID 0xED Data 0x1a8904edf0c01000014e",
from the source (line 415) I see it uses 0x1a89 to recognize it is a
WGR800.

Do you thing I can add it to have:
#define ID_WGR800   0x1984
#define ID_WGR800_2 0x1a89 // Wind2
#define ID_WGR800a  0x1994 // unconfirmed version

and:
if (sensor_id == ID_WGR800 || sensor_id == ID_WGR800a || sensor_id ==
ID_WGR800_2)

and it should work, or are those different things.

Also, heyu is open source software, maybe you can look at its code in
order to add devices ?
[https://github.com/HeyuX10Automation/heyu/blob/master/oregon.c](https://github.com/HeyuX10Automation/heyu/blob/master/oregon.c)

Can you help me to get this working?

Thanks,

domotic-29, 2021-03-17:
I moved the sensor and now I get the data also via rtl_433.
Sorry for disturbing.

closed 2021-03-17 by domotic-29

zuckschwerdt, 2021-03-17:
Glad to hear! It might have been an issue with the center frequency. There is drift over time and maybe interference. Try varying a little, e.g. `-f 433.87M` or  `-f 433.97M`