merbanan/rtl_433 #1682: Problem with remote sensor Buro H999G / Fujian Jimei H999G

mVlad77, 2021-03-19:
I'm collected packets from an external sensor in the format  DATA: CRC - my CRC  
format:   https://hsto.org/webt/dq/mj/iu/dqmjiuixxbenuwzxubzgfftypow.png
orig: https://habr.com/ru/post/432704/
eng:  https://sudonull.com/post/7136-Using-an-external-wireless-thermometer-Buro-H999-in-conjunction-with-homemade-devices

it is not possible to choose an algorithm for calculating the CRC. The last column is the most suitable algorithm that I could find - please HELP :)

05166632: 00 - 00
051B6642: 00 - 00
05187632: 00 - 00
05939602: 00 - 00
05127642: 01 - 18
0516A612: 01 - 18
0554A642: 01 - 18
05949602: 01 - 18
0514A622: 02 - 30
0553A612: 02 - 30
0554A612: 03 - 28
0592A6F1: 03 - 28
05118612: 04 - 60
0552A612: 04 - 60
0516A602: 04 - 60
0594A6F1: 04 - 60
0593A6F1: 05 - 78
0555A612: 05 - 78
05526632: 05 - 78
05147642: 06 - 50
051A6642: 06 - 50
05197632: 06 - 50
05119612: 06 - 50
0592A6E1: 06 - 50
05176632: 06 - 50
0516A6F1: 07 - 48
0555A642: 07 - 48
0597A612: 08 - C0
05166642: 08 - C0
0514A602: 08 - C0
05188622: 08 - C0
0591A6F1: 09 - D8
05506632: 09 - D8
0513A602: 09 - D8
05507662: 09 - D8
051B7632: 0A - F0
0555A622: 0A - F0
05139612: 0A - F0
05167642: 0A - F0
0513A6F1: 0A - F0
05186642: 0A - F0
0596A602: 0B - E8
0595A622: 0B - E8
0515A612: 0B - E8
05117642: 0B - E8
0514A6F1: 0B - E8
051A7632: 0C - A0
0513A612: 0C - A0
0512A6F1: 0C - A0
05129612: 0C - A0
05178622: 0C - A0
0598A612: 0C - A0
0554A622: 0C - A0
05146632: 0C - A0
0515A6F1: 0D - B8
0597A602: 0D - B8
05136632: 0D - B8
0514A612: 0D - B8
0556A642: 0D - B8
0553A622: 0D - B8
05176642: 0E - 90
0596A612: 0E - 90
05198622: 0E - 90
0551A612: 0E - 90
0595A632: 0E - 90
0515A602: 0E - 90
0556A612: 0F - 88
0555A632: 0F - 88
051F7632: 10 - 18
05588662: 10 - 18
05119652: 10 - 18
05199642: 10 - 18
05109612: 11 - 00
05189602: 11 - 00
051996E1: 11 - 00
05158622: 11 - 00
051E8612: 11 - 00
05179642: 12 - 28
051F8602: 12 - 28
059C9642: 12 - 28
051B8622: 13 - 30
05578662: 14 - 78
05109602: 14 - 78
051D8622: 14 - 78
059D9642: 14 - 78
051E96F1: 15 - 60
05128632: 15 - 60
05189642: 16 - 48
05109652: 16 - 48
051E9602: 16 - 48
051E8652: 16 - 48
05199602: 17 - 50
05119612: 17 - 50
05148622: 17 - 50
051F8622: 18 - D8
05558662: 18 - D8
05129602: 18 - D8
0593A632: 18 - D8
05108632: 19 - C0
051C96F1: 19 - C0
05188622: 19 - C0
051C9602: 1A - E8
051A8612: 1A - E8
051A9642: 1A - E8
05149612: 1A - E8
05129652: 1A - E8
051C8652: 1A - E8
0595A622: 1A - E8
05159652: 1B - F0
051D8612: 1B - F0
051B9602: 1B - F0
05139652: 1C - B8
0551B665: 1C - B8
051B8612: 1C - B8
0591A642: 1C - B8
051D9602: 1C - B8
051D8652: 1C - B8
0594A622: 1C - B8
051A8652: 1D - A0
059A9632: 1D - A0
051A9602: 1D - A0
051B96E1: 1D - A0
0593A622: 1D - A0
05548662: 1E - 88
0592A632: 1E - 88
05118632: 1F - 90
0558E645: 1F - 90
0595A632: 1F - 90
05198622: 1F - 90
05538662: 1F - 90
051D96F1: 1F - 90
0595A6F1: 20 - 30
0555A695: 20 - 30
0590A682: 20 - 30
055C9685: 20 - 30
05999692: 20 - 30
0594A612: 20 - 30
0591A623: 21 - 28
059F9662: 21 - 28
055AA685: 21 - 28
05539695: 21 - 28
0593A612: 21 - 28
05979655: 22 - 00
059296E1: 22 - 00
055A9695: 22 - 00
0595A602: 23 - 18
0557A6A5: 23 - 18
059F9632: 23 - 18
059F9672: 24 - 50
0591A6D2: 24 - 50
0594A675: 24 - 50
0551C675: 24 - 50
0591A662: 25 - 48
0590A692: 25 - 48
059E9632: 25 - 48
05979615: 25 - 48
0592A642: 25 - 48
0590A672: 26 - 60
05559695: 26 - 60
0594A6F1: 26 - 60
059F9622: 26 - 60
0591A682: 26 - 60
05979663: 27 - 78
05158602: 28 - F0
05979694: 28 - F0
0592A604: 28 - F0
059E9674: 28 - F0
059E9652: 28 - F0
05569685: 29 - E8
0554A6A5: 29 - E8
059A9672: 29 - E8
0559B685: 29 - E8
0590A642: 29 - E8
059F96F3: 29 - E8
0591A652: 2A - C0
059B9662: 2A - C0
0593A645: 2A - C0
0550B685: 2A - C0
0594A632: 2A - C0
055A9695: 2A - 00
0593A632: 2B - D8
0592A623: 2B - D8
0591A603: 2B - D8
051586F1: 2B - D8
059F96A3: 2B - D8
0557B685: 2B - D8
0591A6A2: 2C - 90
059896B2: 2C - 90
059796F3: 2C - 90
059E9642: 2D - 88
059D9662: 2D - 88
05979643: 2D - 88
0592A632: 2D - 88
0593A605: 2D - 88
05979665: 2D - 88
0593A622: 2E - A0
05989603: 2E - A0
059896E2: 2E - A0
055FB685: 2E - A0
059A9682: 2F - B8
0550B695: 2F - B8
059B9672: 2F - B8
0591A642: 2F - B8
0592A683: 2F - B8

peteakalad, 2021-03-19:
Is the above is from an oscillosope, are the sensors decoding with rtl_433?  

The devices themselves look similar to digoo / baldr 433Mhz sensors I have which decode with the Prologue and Bresser 3ch respectively.  So maybe look at crc decoding of those, it might be the same? 
https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_3ch.c
https://github.com/merbanan/rtl_433/blob/master/src/devices/prologue.c

zuckschwerdt, 2021-03-19:
Are those the raw bits? Nothing in the last byte beyond `2f`? That's just 6-bit, but should go to `3f`.

mVlad77, 2021-03-19:
No other data comes in. I think it's CRC there are only 4 last bits.
XBZZСССС, 
X - send with KEY pressed, 
B - battery OK ???????
ZZ - 00, 01,10  
СССС - CRC
And so I picked it up with collisions (1 bit difference)
is a RAW - it comes as it is written

merbanan, 2021-03-20:
@mVlad77 look into revdgst or reveng. You should record som signals with rtl_433 -S unknown and post to this issue. Then we should be able to aid in the process.

zuckschwerdt, 2021-03-20:
If these codes are correct than it's 4 bits checksum. E.g.
```
051986220 E
051986221 F
```
Also the entropy does not suffice for a CRC, it's a simple addition. E.g. above `E+1=F` is a hint.

gdt, 2023-09-30:
@mVlad77  This issue has been quiet for 2.5 years.   What's the status and next steps?