merbanan/rtl_433 #1684: Ecowitt WH51(B): doesn't work with default 1 Msps (915 MHz version)

roger-, 2021-04-01:
For some reason it doesn't decode anything when run with `rtl_433 -f 915000000` (with default 1 Msps) but it does pick up some data (see [here](https://github.com/merbanan/rtl_433/files/6245948/wh.zip)).

From #1404 I saw someone had better luck with 250 ksps but I had to adjust the center frequency to get things working properly to `rtl_433 -f 914915000 -s 250k`.

The device was very near to the antenna, so I don't think it's a matter of 4x more noise getting in.

ecowitness, 2021-04-08:
This works for me. Thanks. frequency 914915000 sample_rate 250000

You saved my WH51B from the garbage can. I was giving it one last try before moving on when I found your post.

merbanan, 2021-04-10:
@roger- Hi you are probably hitting the base frequency perfectly. Try adjusting it a little bit. And you have to submit some signal samples for us to be able to help better. rtl_433 -f 915000000 -S unknown should give good files.

closed 2021-04-10 by merbanan

roger-, 2021-04-10:
Thought I attached a file — did it work?