merbanan/rtl_433 #1689: Support for Hosmart Driveway PIR and JNStar Driveway Beam

whiteduck22, 2021-04-11:
PR raised on rtl_433_test #391 - I wonder if someone is able to help analyse the capture logs for these two devices.  Many thanks

whiteduck22, 2021-04-14:
Here is the output from rtl_433 -A for the JNSTAR:
```
rtl_433 -A -r g009_433.92M_250k.cu8 
rtl_433 version 20.11 branch  at 202011122224 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/ian/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g009_433.92M_250k.cu8
Detected OOK package	@0.108784s
Analyzing pulses...
Total count: 1200,  width: 833.41 ms		(208353 S)
Pulse width distribution:
 [ 0] count:    1,  width: 2092 us [2092;2092]	( 523 S)
 [ 1] count:  575,  width:  184 us [164;216]	(  46 S)
 [ 2] count:  624,  width:  404 us [380;436]	( 101 S)
Gap width distribution:
 [ 0] count:   48,  width: 3444 us [3416;3496]	( 861 S)
 [ 1] count:  527,  width:  356 us [324;380]	(  89 S)
 [ 2] count:  624,  width:  184 us [156;212]	(  46 S)
Pulse period distribution:
 [ 0] count:    1,  width: 5588 us [5588;5588]	(1397 S)
 [ 1] count: 1151,  width:  568 us [528;612]	( 142 S)
 [ 2] count:   47,  width: 3628 us [3616;3644]	( 907 S)
Pulse timing distribution:
 [ 0] count:    1,  width: 2092 us [2092;2092]	( 523 S)
 [ 1] count: 1199,  width:  184 us [156;216]	(  46 S)
 [ 2] count: 1152,  width:  380 us [324;436]	(  95 S)
 [ 3] count:   48,  width: 3444 us [3416;3496]	( 861 S)
Level estimates [high, low]:  15924,     18
RSSI: -0.1 dB SNR: 29.5 dB Noise: -29.6 dB
Frequency offsets [F1, F2]:    8702,      0	(+33.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB00B0401082C00B8017C0D748355+AAB023042F082C00B8017C0D749292A192A1A1A1A192A1A192A1A19292A1A1A1A1929292929355+AAB0220401082C00B8017C0D749292A192A1A1A1A192A1A192A1A19292A1A1A1A19292929255
Attempting demodulation... short_width: 184, long_width: 404, reset_limit: 3500, sync_width: 2092
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=184,l=404,r=3500,g=0,t=0,y=2092'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {1199} d0 93 0f e8 49 87 f4 24 c3 fa 12 61 fd 09 30 fe 84 98 7f 42 4c 3f a1 26 1f d0 93 0f e8 49 87 f4 24 c3 fa 12 61 fd 09 30 fe 84 98 7f 42 4c 3f a1 26 1f d0 93 0f e8 49 87 f4 24 c3 fa 12 61 fd 09 30 fe 84 98 7f 42 4c 3f a1 26 1f d0 93 0f e8 49 87 f4 24 c3 fa 12 61 fd 09 30 fe 84 98 7f 42 4c 3f a1 26 1f d0 93 0f e8 49 87 f4 24 c3 fa 12 61 fd 09 30 fe 84 98 7f 42 4c 3f a1 26 1f d0 93 0f e8 49 87 f4 24 c3 fa 12 61 fd 09 30 fe 84 98 7f 42 4c 3f a1 26 1e 

Detected OOK package	@0.942568s
Analyzing pulses...
Total count:  801,  width: 554.76 ms		(138689 S)
Pulse width distribution:
 [ 0] count:  385,  width:  168 us [156;184]	(  42 S)
 [ 1] count:  416,  width:  392 us [380;404]	(  98 S)
Gap width distribution:
 [ 0] count:   32,  width: 3456 us [3452;3468]	( 864 S)
 [ 1] count:  352,  width:  368 us [356;388]	(  92 S)
 [ 2] count:  416,  width:  200 us [188;212]	(  50 S)
Pulse period distribution:
 [ 0] count:   32,  width: 3628 us [3624;3644]	( 907 S)
 [ 1] count:  768,  width:  568 us [524;608]	( 142 S)
Pulse timing distribution:
 [ 0] count:  801,  width:  184 us [156;212]	(  46 S)
 [ 1] count:  768,  width:  380 us [356;404]	(  95 S)
 [ 2] count:   32,  width: 3456 us [3452;3468]	( 864 S)
 [ 3] count:    1,  width: 10004 us [10004;10004]	(2501 S)
Level estimates [high, low]:  15915,     18
RSSI: -0.1 dB SNR: 29.5 dB Noise: -29.6 dB
Frequency offsets [F1, F2]:    8725,      0	(+33.3 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB00B040100B8017C0D8027148255+AAB023041F00B8017C0D8027148181908190909090819090819090818190909090818181818255+AAB023040100B8017C0D8027148181908190909090819090819090818190909090818181818355
Attempting demodulation... short_width: 168, long_width: 392, reset_limit: 3472, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=168,l=392,r=3472,g=392,t=90,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 25 
[00] { 1} 80          : 1
[01] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[02] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[03] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[04] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[05] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[06] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[07] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[08] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[09] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[10] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[11] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[12] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[13] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[14] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[15] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[16] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[17] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[18] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[19] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[20] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[21] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[22] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[23] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
[24] {25} d0 93 0f 80 : 11010000 10010011 00001111 1
... Maximum number of rows reached. Message is likely truncated.
```

whiteduck22, 2021-04-14:
Here is the output from rtl_433 -A for the HOSMART:

`rtl_433 -A -r g011_433.92M_250k.cu8 
rtl_433 version 20.11 branch  at 202011122224 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/ian/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g011_433.92M_250k.cu8
Detected FSK package	@0.270628s
Analyzing pulses...
Total count:  152,  width: 302.40 ms		(75599 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   69,  width: 1236 us [1232;1240]	( 309 S)
 [ 2] count:   81,  width:  408 us [408;416]	( 102 S)
 [ 3] count:    1,  width:   12 us [12;12]	(   3 S)
Gap width distribution:
 [ 0] count:    1,  width: 14172 us [14172;14172]	(3543 S)
 [ 1] count:   69,  width:  412 us [408;416]	( 103 S)
 [ 2] count:   75,  width: 1236 us [1232;1240]	( 309 S)
 [ 3] count:    5,  width: 9272 us [9268;9280]	(2318 S)
 [ 4] count:    1,  width: 1920 us [1920;1920]	( 480 S)
Pulse period distribution:
 [ 0] count:    1,  width: 14172 us [14172;14172]	(3543 S)
 [ 1] count:  144,  width: 1648 us [1644;1652]	( 412 S)
 [ 2] count:    5,  width: 9684 us [9680;9692]	(2421 S)
 [ 3] count:    1,  width: 2332 us [2332;2332]	( 583 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:  144,  width: 1236 us [1232;1240]	( 309 S)
 [ 2] count:  150,  width:  412 us [408;416]	( 103 S)
 [ 3] count:    1,  width:   12 us [12;12]	(   3 S)
 [ 4] count:    1,  width: 14172 us [14172;14172]	(3543 S)
 [ 5] count:    5,  width: 9272 us [9268;9280]	(2318 S)
 [ 6] count:    1,  width: 1920 us [1920;1920]	( 480 S)
 [ 7] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  15983,     65
RSSI: -0.1 dB SNR: 23.9 dB Noise: -24.0 dB
Frequency offsets [F1, F2]:     630,  -6013	(+2.4 kHz, -22.9 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB0130801000004D4019C000C375C2438078000008455+AAB02B0803000004D4019C000C375C2438078000009292A1A1A192A19292A1929292A1A192A1A192A1A1A1A192A555+AAB02B0802000004D4019C000C375C24380780000092A1929292A192A192A192A1A192A192A192A192A1A1A192A555+AAB02C0801000004D4019C000C375C24380780000092A1929292A192A192A192A1A192A192A192A192A1A1A192A6B055
Attempting demodulation... short_width: 408, long_width: 1236, reset_limit: 14176, sync_width: 12
Use a flex decoder with -X 'n=name,m=FSK_PWM,s=408,l=1236,r=14176,g=0,t=0,y=12'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] {150} 3a 46 de 9d 23 6f 4e 91 b7 a8 ab 55 d4 55 aa ea 2a d5 74 
[01] { 0}                                                          : 

Detected FSK package	@0.680236s
Analyzing pulses...
Total count:  152,  width: 302.80 ms		(75701 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   69,  width: 1236 us [1232;1244]	( 309 S)
 [ 2] count:   81,  width:  408 us [408;416]	( 102 S)
 [ 3] count:    1,  width:   20 us [20;20]	(   5 S)
Gap width distribution:
 [ 0] count:    1,  width: 14172 us [14172;14172]	(3543 S)
 [ 1] count:   69,  width:  412 us [412;416]	( 103 S)
 [ 2] count:   75,  width: 1236 us [1236;1244]	( 309 S)
 [ 3] count:    5,  width: 9284 us [9284;9292]	(2321 S)
 [ 4] count:    1,  width: 1960 us [1960;1960]	( 490 S)
Pulse period distribution:
 [ 0] count:    1,  width: 14172 us [14172;14172]	(3543 S)
 [ 1] count:  144,  width: 1648 us [1644;1656]	( 412 S)
 [ 2] count:    5,  width: 9700 us [9696;9704]	(2425 S)
 [ 3] count:    1,  width: 2372 us [2372;2372]	( 593 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:  144,  width: 1236 us [1232;1244]	( 309 S)
 [ 2] count:  150,  width:  412 us [408;416]	( 103 S)
 [ 3] count:    1,  width:   20 us [20;20]	(   5 S)
 [ 4] count:    1,  width: 14172 us [14172;14172]	(3543 S)
 [ 5] count:    5,  width: 9284 us [9284;9292]	(2321 S)
 [ 6] count:    1,  width: 1960 us [1960;1960]	( 490 S)
 [ 7] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  15959,      8
RSSI: -0.1 dB SNR: 33.0 dB Noise: -33.1 dB
Frequency offsets [F1, F2]:     560,  -6084	(+2.1 kHz, -23.2 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB0130801000004D4019C0014375C244407A800008455+AAB02B0803000004D4019C0014375C244407A800009292A1A1A192A19292A1929292A1A192A1A192A1A1A1A192A555+AAB02B0802000004D4019C0014375C244407A8000092A1929292A192A192A192A1A192A192A192A192A1A1A192A555+AAB02C0801000004D4019C0014375C244407A8000092A1929292A192A192A192A1A192A192A192A192A1A1A192A6B055
Attempting demodulation... short_width: 408, long_width: 1236, reset_limit: 14176, sync_width: 20
Use a flex decoder with -X 'n=name,m=FSK_PWM,s=408,l=1236,r=14176,g=0,t=0,y=20'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] {150} 3a 46 de 9d 23 6f 4e 91 b7 a8 ab 55 d4 55 aa ea 2a d5 74 
[01] { 0} `

whiteduck22, 2021-04-17:
Just wondered if anyone knows how to decode this and get support added into the next release? Sorry I don't know where to start. Thanks a lot.

zuckschwerdt, 2021-04-17:
Start with the pdv data links and the flex suggestion. JNSTAR. The data is simple PWM, the flex decoder with -X 'n=jnstar,m=OOK_PWM,s=180,l=390,r=4000,g=500' should work. See the conf folder on how to write flex conf file.

The Hosmart looks difficult, there are overlong pulses in there that maybe sync but are not detected properly.

whiteduck22, 2021-04-17:
Thank you very much. The Hosmart device seems to be a well liked Driveway Alarm and I've found it pretty reliable so would be great to solve it. Will go take a look at your suggestion for the jnstar. Thanks again

whiteduck22, 2021-05-22:
If anyone can help with the Hosmart that would be great. Otherwise perhaps a photocell taped onto the receiver is the simpler option. Thanks

whiteduck22, 2022-02-14:
For info I saw slightly down this thread a suggestion. I will try it https://community.home-assistant.io/t/driveway-alarm-hack/50264

whiteduck22, 2022-02-14:
I am a bit lost. I've seen on the post above in the link to the emacros or hosmart suggestion. I tried the flex decoder but this triggers when my InkBird ITH-20R sends a signal so perhaps similar patterns. Should I give up as I haven't got my head around how to decode on trig.org. Thanks a lot.

zuckschwerdt, 2022-02-14:
The Hosmart signal is hard to correctly read for rtl_433. Not sure if anything can be done about that.
You can maybe try to read with `-s 1024k` and perhaps `-Y minmax`

whiteduck22, 2022-02-14:
Thank you. Earlier I had some joy in detecting motion which I have posted to the Home Assistant forum in the link above. 'Will test it for reliability and feedback here. For me I just want to trigger a snapshot so not worried about decoding specific messages. All the best, Ian

sbalabanov, 2022-12-11:
I think I reversed engineered eMacros/Hosmart to `-X "n=Hosmart-HS002,m=FSK_PWM,s=416,l=1248,r=14240,g=1252,t=332,y=0,bits=25,invert,repeats>=3,unique,get=id:@0:{12},get=model_id:@12:{8},get=channel:@20:{4},get=battery_ok:@24:{1}:[0:1 1:0]"`
More details on Home Assistant thread.

gdt, 2023-09-30:
@whiteduck22 @sbalabanov What's the status of this issue and next steps?   (I would like to see the progress move from issue comments into the sources.)

klohner, 2024-06-26:
@whiteduck22 Can you zip up some .cu8 captures of this signal and post here?