merbanan/rtl_433 #1690: Windows Building rtl_433 Instruction Problem

MaxFun, 2021-04-11:
In the Windows Building rtl_433 Instruction ; 
"go into the cmake/modules folder and open FindLibUSB.cmake with a text editor"
I just downloaded librtlsdr from https://github.com/steve-m/librtlsdr and there is no FindLibUSB.cmake anywhere in the download.
If this file is still needed, where do I get it?

zuckschwerdt, 2021-04-11:
Either follow https://github.com/merbanan/rtl_433/blob/master/do_sysroot.sh for a MinGW-w64 build.
Or for MSVC
- Install PothosSDR from https://downloads.myriadrf.org/builds/PothosSDR/
- Install OpenSSL (if you want TLS) from e.g. Chocolatey `choco install openssl`
- Build with CMake and `Visual Studio 15 2017` or `Visual Studio 16 2019` generator.
  (use `-DENABLE_RTLSDR=OFF/ON -DENABLE_SOAPYSDR=OFF/ON -DENABLE_OPENSSL=OFF/ON` to your liking)

There are also some hints in the new Release build action at https://github.com/merbanan/rtl_433/blob/feat-relact/.github/workflows/release.yml which I'll merge for the next release.

MaxFun, 2021-04-11:
I not sure I understand your instructions.

 

I have installed MinGW-w64, so I assume I would follow https://github.com/merbanan/rtl_433/blob/master/do_sysroot.sh

When I open the link, it indicates it a “executable file”. Do I down load the file and change the extension to “.exe” and run it?

 

Or do I open a MinGW-w64 terminal and enter the each line one at time?

Also there is “# or git clone https://github.com/osmocom/rtl-sdr.git” part way down the code. Does that mean only run the code down to there and stop?

 

If MSVC is Microsoft Visual C++, I do not have it or never used it.

zuckschwerdt, 2021-04-11:
> Also there is “# or git clone ... Does that mean only run the code down to there and stop?

It's just a comment that we can either clone or download (curl) the source.

> do_sysroot.sh When I open the link, it indicates it a “executable file”.

You need the MinGW terminal, then either enter the wanted commands or run the whole file (use `sh file.sh`)
The script is written for Cross-compilation (the CMAKE_TOOLCHAIN_FILE stuff), no idea if it works natively.

Also the script builds 4 variants of the project (static/dynamic, 32/64-bit) -- you likely only want one.
You need at least libusb and rtl-sdr. Either as pre-compiled download or compiled yourself.
If you just install pothos you'll get all that "for free" and lots of other nice SDR software.
It should then just be something like line `98` and you're done.

TL;DR all builds (native/cross, 32/64, static/dynamic, gcc/msvc) work, but preparing the needed libs and path settings is complicated to get right -- sorry :/

You can still get our official builds from https://bintray.com/beta/#/chzu/dist/rtl_433
Starting next month we'll release builds here instead.

MaxFun, 2021-04-11:
Thanks responding so quickly
I think I will try installing PothosSDR if I can.

MaxFun, 2021-04-11:
Installed PothosSDR-2021.02.28-vc16-x64.exe and used to tutorial https://github.com/pothosware/PothosSDR/wiki/Tutorial to get started. The tutorial(2017) has the SDR Source, but the PothosSDR app only has Soapy Source. Tried the Soapy Source but it failed with a RTL-SDR device, error below.
It appears the PothosSDR app is no long compatible with RTL-SDR device

19:15:18.270000] ::1: Found Fitipower FC0012 tuner 
[19:15:20.110000] SoapySDR: Using format CF32.
[19:15:25.649000] SoapySDR: rtlsdr_get_device_usb_strings(0) failed
[19:24:30.983000] SoapySDR: rtlsdr_get_device_usb_strings(0) failed

zuckschwerdt, 2021-04-11:
Should be the same librtlsdr (for https://github.com/pothosware/SoapyRTLSDR/ ) that we use. There should also be `rtl_sdr` which works with the raw librtlsdr (instead of Soapy). Not sure why rtlsdr_get_device_usb_strings would fail, permission problems? Did you properly uninstall the DVB-T driver with Zadig?

MaxFun, 2021-04-11:
I tried the SDR-RTL with an old(2019) working copy of rtl_433 and it works fine, see below

C:\RTL\RtlWinLenA12_27_19\librtlsdr\build\src>rtl_433_12_27_19 -M level
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Trying conf file at "C:\RTL\RtlWinLenA12_27_19\librtlsdr\build\src\rtl_433.conf"...
Trying conf file at "C:\Users\MySelf\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...

        Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
        A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 104 out of 134 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-134 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-04-11 13:04:40
model     : Acurite 606TX Sensor                   id        : 37
Battery   : OK           Temperature: 22.0 C       Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.929 MHz
RSSI      : -0.3 dB      SNR       : 17.5 dB       Noise     : -17.8 dB

Mindavi, 2021-05-18:
Scripting the build (while downloading the sources for deps and building them?) might be nice to get this working. I'm not working very often on Windows anymore so I'm not sure if I will find the time to look at that.

I think I (re)wrote the Windows instructions some time ago when I wanted to do a build. It appears to have gone out of date since.

MaxFun, 2021-05-21:
Thanks for the update. Hoping that you or someone else finds the time to get this working again. Have a new sensor and hoping it might be in the new build.

zuckschwerdt, 2021-05-22:
An up-to-date build is here https://github.com/merbanan/rtl_433/releases -- let us know what's works and what's broken!

If you want to build yourself with all featues and don't want to fiddle with setting up deps:
- Needs Visual Studio 15 2017 or Visual Studio 16 2019
- Needs CMake
- Install recent PothosSDR (2021.05.01-vc16-x64 tested)
- Install OpenSSL (`choco install openssl`)
- cmake with  -G"Visual Studio 15 2017" -DENABLE_RTLSDR=ON -DENABLE_SOAPYSDR=ON -DENABLE_OPENSSL=ON --config Release
- `cmake --build .`

Completly untested (let me know how to improve) but roughly those steps are used to build the Release binaries.

MaxFun, 2021-05-27:
I download the up-to-date build and tried to run rtl_433-rtlsdr.exe in mingw-w64 terminal. It appeared to run and find the SDR but did not output any detected sensors.

C:\RTL\rtl_433-win-msvc-x64-21.05>rtl_433-rtlsdr
rtl_433 version 21.05 branch  at 202105091238 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\RTL\rtl_433-win-msvc-x64-21.05\rtl_433.conf"...
Trying conf file at "C:\Users\MySelf\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 (non-zero-copy) user-space buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us

After waiting a while, I pressed clt + c which produced the following

pressing ctl+c caused the following output
“Signal caught, exiting!
LIBUSB_ERROR_NOT_FOUND: Entity not found!
Check your RTL-SDR dongle, USB cables, and power supply.
WARNING: async read failed (-5).”
C:\RTL\rtl_433-win-msvc-x64-21.05>

To verify the SDR could be read, I ran SDRSharp.exe which opened and showed my sensor popping up.
Next I tried the SDR-RTL with an old(2019) working copy of rtl_433 which worked on 4/11/21 (see above)
but now fails

C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0>echo off
Microsoft Windows [Version 10.0.19041.985]
(c) Microsoft Corporation. All rights reserved.

C:\>cd C:\RTL\RtlWinLenA12_27_19\librtlsdr\build\src

C:\RTL\RtlWinLenA12_27_19\librtlsdr\build\src>rtl_433_12_27_19 -M level
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Trying conf file at "C:\RTL\RtlWinLenA12_27_19\librtlsdr\build\src\rtl_433.conf"...
Trying conf file at "C:\Users\MySelf\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...

        Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
        A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 104 out of 134 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-134 ]

C:\RTL\RtlWinLenA12_27_19\librtlsdr\build\src>

Download the latest Zadig and install driver, same results for old and new rtl_433.

My windows 10 Pro has had 3 updates since  4/11/21

Next I will try building per instructions with Visual studio 2019

zuckschwerdt, 2021-05-27:
Not sure about the LIBUSB_ERROR_NOT_FOUND, something went wrong with the SDR I guess.

The fitipower tuner needs a gain setting for proper operation. Try -g 20.

MaxFun, 2021-05-27:
No luck, same thing

C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0>echo off
Microsoft Windows [Version 10.0.19041.985]
(c) Microsoft Corporation. All rights reserved.

C:\>cd C:\RTL\rtl_433-win-msvc-x64-21.05

C:\RTL\rtl_433-win-msvc-x64-21.05>rtl_433-rtlsdr -g 20
rtl_433 version 21.05 branch  at 202105091238 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\RTL\rtl_433-win-msvc-x64-21.05\rtl_433.conf"...
Trying conf file at "C:\Users\MySelf\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to 19.200000 dB.
Tuned to 433.920MHz.
Allocating 15 (non-zero-copy) user-space buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us

Mindavi, 2021-05-27:
I guess running it with -vvv (`rtl_433.exe -vvv`) will give some insight in if any data is received. It might be that there's a regression causing that particular sensor to misbehave. However, we do have samples for that particular sensor, so our regression suite should catch any regressions.

merbanan, 2021-05-27:
The FC0012 tuner is not the best tuner for rt-sdr.

MaxFun, 2021-05-27:
Problem solved by switch to Rafael Micro R820T/2 SDR
Thanks

C:\RTL\rtl_433-win-msvc-x64-21.05>rtl_433-rtlsdr
rtl_433 version 21.05 branch  at 202105091238 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\RTL\rtl_433-win-msvc-x64-21.05\rtl_433.conf"...
Trying conf file at "C:\Users\MySelf\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 (non-zero-copy) user-space buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-05-27 14:53:12
model     : Acurite-Tower id        : 4908
channel   : B            battery_ok: 1             Temperature: 24.0 C       Humidity  : 16 %          Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-05-27 14:53:12

MaxFun, 2021-05-30:
After I started to get data from running rtl_433-rtlsdr.exe from rtl_433-win-msvc-x64-21.05, it appears that the option  “–F csv” is not what I thought it should be. Attached are two files, one without option –F (Test_5_29_21_915mh .txt) and one with option “–F csv” (Test_5_30_21_915mhCsv.txt). Am I missing something?

Did find what I think is one of the three sensor of my new weather station which I was hoping for with the update. Is there a location with more information on the sensors rtl_433 can decode such as, FCC ID or venders model number or what is thought the sensors parameters mean. 

I see “model     : LaCrosse-TH2 Sensor ID : 356995” which maybe my “LTV-TH2 Sensor”. The other two sensor are “LTV-W1 Sensor” & “LTV-R2 Sensor”

[Test_5_30_21_915mhCsv .txt](https://github.com/merbanan/rtl_433/files/6566317/Test_5_30_21_915mhCsv.txt)
[Test_5_29_21_915mh .txt](https://github.com/merbanan/rtl_433/files/6566318/Test_5_29_21_915mh.txt)

zuckschwerdt, 2021-05-30:
"without -F" defaults to `-F kv`. The CSV look right. Perhaps import into a spreadsheet to see the columns better.
Anyway, recommended is `-F json` :)

There is https://triq.org/rtl_433/ with the docs, and https://triq.org/explorer/ has "Decoders" with some info.

MaxFun, 2021-05-30:
Thanks,
OK, just didn't expect 339 commas for 39 variables in a line, that what I counted in the text file for csv.

zuckschwerdt, 2021-05-30:
If you just enable the decoders you need then the number of fields is reduced.

gdt, 2023-09-30:
Calling this a help request which has concluded, plus there's no real reason to expect the build system to be the same, so closing.  Feel free to open a discussion or post on the list for help, or an issue with specific wr0ng things, or better yet a PR with fixes.

closed 2023-09-30 by gdt