merbanan/rtl_433 #1691: Viking 02812 thermometer misidentified

Jaif, 2021-04-11:
I have some Viking 02812 thermometer/hygrometer that are identified as Fineoffset-WH5.
While the humidity value is correct, the temperature is 40 degrees C too low. Looking at the code it seems like it should probably be handled the same way as Fineoffset-WH2 instead.
[g001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/6292277/g001_433.92M_250k.cu8.zip)

merbanan, 2021-04-11:
src/rtl_433 -R 18:no-wh5 -r ../signals/viking/g001_433.92M_250k.cu8

```
time      : @0.089140s
model     : Rosenborg-66796                        ID        : 55
Temperature: 3.8 C       Humidity  : 54 %          Integrity : CRC
```
I did not look too much into the signal properties (pre/post-amble) but my conclusion before was that it was not possible to differentiate between the 2 different types. I suspect that is still the case and I will not spend any time to try to figure out if that is the case with this sensor when there is a working solution out of the box.

closed 2021-04-11 by merbanan

petterreinholdtsen, 2022-11-03:
I've seen a similar issue with a thermometer a bit far away from the receiver.  Most of the time it is classified as Fineoffset-WH2, but once in a while it is classified as Fineoffset-WH5 with a different ID number.  I suspect the signal is too weak once in a while, and that some bits were not detected.

zuckschwerdt, 2022-11-03:
The WH5 is just a single bit shorter than the WH2 in the preamble byte. But the temperature is encoded different. Nothing much we can do there :/
https://github.com/merbanan/rtl_433/blob/eee869e78a9ddb70a2284cecdb4e5c139f1658d4/src/devices/fineoffset.c#L83-L96