merbanan/rtl_433 #1692: Sensor EMOS 6018

a-r-m-e-n, 2021-04-14:
Hi,

I would like to ask you for support. I try to receive data from sensor EMOS 6018 and I caught following waves:
[wave_links.txt](https://github.com/merbanan/rtl_433/files/6312181/wave_links.txt)

In the file there are links to waves.

Now, I try to write function for that device that would demodulate it ... and later decode it. Based on the advice from rtl_433 I selected following parameters (as an average/guess) in the function:
r_device emos_e6018  = {
        .name        = "EMOS E6018 Temperature Sensor",
        .modulation  = OOK_PULSE_PWM,
        .short_width = 300,
        .long_width  = 823,
        .reset_limit = 823,
        .sync_width  = 1873,
        .tolerance   = 50,
        .decode_fn   = &emos_e6018_decode,
        .disabled    = 0,
        .fields      = output_fields,
};

However, these parameters are probably not good/robust because it sometimes work / sometimes not ... :(

Could you please advice/help me how can I fine tune these parameters ? What to do to better guess ?

Thank you very much.

zuckschwerdt, 2021-04-14:
Looks good. Are you getting no bitbuffer at all, or unexpected values? If you can grab a problematic .cu8 post a zip here.

a-r-m-e-n, 2021-04-14:
Ok, thank you very much for answer. Yes, sometimes I am not getting bitbuffer at all/function for decode is not called while for other sensors it is ... Could it be that there might be some distortions from other signals/sensors ? I will try to gather more sample data, derive r_device parameters from them using -A switch and then make an average over them ? Is this the best approach ? What do you think about tolerance ? How to decide for best fit ?

zuckschwerdt, 2021-04-14:
With a brief look at the pdv links, all your r_device parameters look good. Perhaps use `-S unknown` to catch 20-50 samples. Then try different `-X` or r_device on those .cu8 and see what works best. And find the samples that do not work and investigate why :)

a-r-m-e-n, 2021-04-14:
Ok, thank you. I provided only links where signals are nice :), clean :) and decoding works :) Ok, I will investigate :) Once I feel it seems to be usable I'll put it on a branch ...

kuenkin, 2023-08-02:
Hello,
With the latest version this sensor is detected as EMOS-6016 (protocol 214) giving zero values for the wind speed and wind direction:
```
[rtl_433_2] time      : 2023-08-02T22:03:22+0200
[rtl_433_2] model     : EMOS-E6016   House Code: 177
[rtl_433_2] Channel   : 1            Battery_OK: 0             Temperature_C: 26.9       Humidity  : 49            WindSpeed m_s: 0.0        Wind direction: 0.0       Radio Clock: 2015-01-01T06:26:24                    Integrity : CHECKSUM
[rtl_433_2] Modulation: ASK          Freq      : 433.9 MHz
[rtl_433_2] RSSI      : -1.2 dB      SNR       : 15.5 dB       Noise     : -16.7 dB
```
```
[rtl_433_2] time      : 2023-08-02T22:10:14+0200
[rtl_433_2] model     : EMOS-E6016   House Code: 177
[rtl_433_2] Channel   : 1            Battery_OK: 0             Temperature_C: 29.1       Humidity  : 37            WindSpeed m_s: 0.0        Wind direction: 0.0       Radio Clock: 2015-01-01T06:33:17                    Integrity : CHECKSUM
[rtl_433_2] Modulation: ASK          Freq      : 433.9 MHz
[rtl_433_2] RSSI      : -1.0 dB      SNR       : 15.2 dB       Noise     : -16.1 dB
```
```
[rtl_433_2] time      : 2023-08-02T22:11:11+0200
[rtl_433_2] model     : EMOS-E6016   House Code: 177
[rtl_433_2] Channel   : 1            Battery_OK: 0             Temperature_C: 29.3       Humidity  : 37            WindSpeed m_s: 0.0        Wind direction: 0.0       Radio Clock: 2015-01-01T06:34:15                    Integrity : CHECKSUM
[rtl_433_2] Modulation: ASK          Freq      : 433.9 MHz
[rtl_433_2] RSSI      : -1.1 dB      SNR       : 15.6 dB       Noise     : -16.7 dB
```
The temperature value and humidity value reported are sensible and seems correct .

zuckschwerdt, 2023-08-02:
Oh, how nice and unexpected (EMOS is a rebrander and similar named devices might be of different manufacturers).

Can you post some codes captured with `rtl_433 -R 214:v` -- we might be able to spot a "feature bit" or something to tell how the difference between both.

kuenkin, 2023-08-02:
Sure,
```
time      : 2023-08-02 22:52:59
model     : EMOS-E6016   House Code: 177
Channel   : 1            Battery_OK: 1
Temperature_C: 30.4      Humidity  : 30
WindSpeed m_s: 0.0       Wind direction: 0.0
Radio Clock: 2015-01-01T07:16:03                   Integrity :
CHECKSUM
[pulse_slicer_pwm] EMOS E6016 weatherstation with DCF77
codes     : {120}aaa583b10f109d00c1301e00045200, {120}aaa583b10
f109d00c1301e00045201, {120}aaa583b10f109d00c1301e00045202, {12
0}aaa583b10f109d00c1301e00045203, {28}aaa583b
```
```
time      : 2023-08-02 22:53:56
model     : EMOS-E6016   House Code: 177
Channel   : 1            Battery_OK: 1
Temperature_C: 30.4      Humidity  : 30
WindSpeed m_s: 0.0       Wind direction: 0.0
Radio Clock: 2015-01-01T07:17:01                   Integrity :
CHECKSUM
[pulse_slicer_pwm] EMOS E6016 weatherstation with DCF77
codes     : {79}aaa583b10f109d1d983a, {33}203e1b0b8, {3}2, {15}
dd52, {1}0, {18}49210, {9}978, {11}b12, {3}2, {28}24724a4, {19}
5db82, {42}4b6019d2054, {1}0, {2}8, {7}ca, {1}0, {43}c3da2978eb
4, {66}44104c07800138808, {120}aaa583b10f109d1041301e0004e203,
{120}aaa583b10f109d1041301e0004e204, {120}aaa583b10f109d1041301
e0004e205, {0}
```
```
time      : 2023-08-02 22:54:14
model     : EMOS-E6016   House Code: 177
Channel   : 1            Battery_OK: 1
Temperature_C: 30.4      Humidity  : 30
WindSpeed m_s: 0.0       Wind direction: 0.0
Radio Clock: 2015-01-01T07:17:16                   Integrity :
CHECKSUM
[pulse_slicer_pwm] EMOS E6016 weatherstation with DCF77
codes     : {120}aaa583b10f109d1401301e0004a600, {120}aaa583b10
f109d1401301e0004a601, {120}aaa583b10f109d1401301e0004a602, {12
0}aaa583b10f109d1401301e0004a603, {120}aaa583b10f109d1401301e00
04a604, {120}aaa583b10f109d1401301e0004a605, {0}
```

zuckschwerdt, 2023-08-02:
Here is [a BitBench comparing the data](https://triq.net/bitbench#c=%5Be6018%5D&c=aaa583b10f109d00c1301e00045201&c=aaa583b10f109d1041301e0004e201&c=aaa583b10f109d1401301e0004a601&c=aaa583b10f109ed7412f1d0004a801&c=aaa5837597815d6c01121800429501&c=aaa583b10f109ee6c12f1d00043701&c=aaa5837597815d7b81111800422301&c=%5Be6016%5D&c=aaa583ff955a9f18c0c92500a2c700&c=aaa583ff955a9f18c0c92500a2c701&c=aaa583ff955a9f18c0c92500a2c702&c=aaa583ff955a9f18c0c92500a2c703&c=aaa583ff955a9f18c0c92500a2c704&c=aaa583ff955a9f18c0c92500a2c705&f=MODEL%3F%3A8h8h8h%20ID%3F%3A8d%20%3F2b%20DT%3A6d-4d-5dT5d%3A6d%3A6d%20CH%3A2d%20TEMP%3A12d%20HUM%3F8d%20WSPEED%3A8d%20WINDIR%3A4d%20BAT%3A4b%20CHK%3A8h%20REPEAT%3A8h&cw=4). Could be that just the first flag bit indicates if wind data is available? Or maybe that just flags if the DCF77 time is valid?

Interesting that the simpler e6018 sensor is counting some kind of time.

gdt, 2023-09-29:
@a-r-m-e-n @kuenkin What's the status and path forward?

saaj, 2026-10-27:
For future readers. A couple days ago I bought EMOS E06018 (`asin=B0886VXTFG` on amazon.nl). It wasn't working with Home Assistant's "rtl_433 (next)" built from master (specifically, `rtl_433 version 25.12-400-g406f9a33 branch master at 202610201107`). After digging into a rabbit hole of thinking the manufacturer has changed something and not getting any signal to analyse, I realised it was in fact poorly tuned device.

```
# rtl_433 -s 1000000 -M level       
rtl_433 version 25.12-400-g406f9a33 branch master at 202610201107 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2026-10-27 19:06:36
model     : EMOS-E6016   House Code: 99
Channel   : 3            Battery_OK: 0             Temperature_C: 23.9 C     Humidity  : 53
WindSpeed m_s: 0.0 m/s   Wind direction: 0.0       Radio Clock: 2026-10-27T20:06:36
Integrity : CHECKSUM
Modulation: ASK          Freq      : 434.1 MHz
RSSI      : 1.1 dB       SNR       : 4.1 dB        Noise     : -4.1 dB
```

`Freq      : 434.1 MHz` is rather far from "433.92MHz" that's on the label of EMOS E06018. After adding `sample_rate 1000k` into `rtl_433`'s config, the sensor appeared in my MQTT. The default 250KHz bandwidth apparently wasn't enough for it.