merbanan/rtl_433 #1693: Unterstützung für TFA Weatherhub 30.3301.02

Vogtlaender, 2021-04-14:
Hello, I would like to ask you to integrate the WEATHERHUB 30.3301.
https://www.tfa-dostmann.de/produkt/temperatursender-mit-wasserfestem-kabelfuehler-weatherhub-30-3301/

merbanan, 2021-04-14:
Hi, grab some signals and their matching values and we'll try to add the needed support.

https://github.com/merbanan/rtl_433/blob/master/docs/CONTRIBUTING.md#supporting-additional-devices-and-test-data

Vogtlaender, 2021-04-17:
hello, unfortunately I don't have the knowledge to do that.
I tried the following command: rtl_433 -f 868.25M -S all -T 60
Is the command correct?
The neighbour's weather station is also recorded

merbanan, 2021-04-17:
rtl_433 -f 868.25M -S unknown and then record the station values when signals are recorded. On the rtl_sdr device remove the antenna and place the TFA Weatherhub 30.3301.02 nearby so that only the TFA can be heard.

Vogtlaender, 2021-04-18:
Hello,
thanks for your help. I created the files. Where should I upload it? I only recorded a temperature of 23.5.
Unfortunately, the device does not have its own display. So it can be that the temperature deviates.

Vogtlaender, 2021-04-27:
#merbanan 
Hello you can
please tell me where I can upload the files?

merbanan, 2021-04-27:
Zip and upload in this issue.

Vogtlaender, 2021-04-27:
thanks
[werte_21_6.zip](https://github.com/merbanan/rtl_433/files/6383298/werte_21_6.zip)

Vogtlaender, 2021-04-29:
#merbanan
hallo sind die dateinen ok?

merbanan, 2021-04-29:
Kein signal. Remove the antenna from the device and then place the transmitter just next to the receiver and record again.

You can look at the signal via https://triq.org/pdv/.

wittj, 2022-11-26:
Hi, following up the previous posts here I would also like to ask for support in integration of this device. For this I have recorded a couple of signals from a device. The first three signales have been recorded after inserting batteries. After that the device sends a signal every 7 minutes. I'm sorry, but I cannot tell the temperature readings. It might be around 18-19 °C here in the room.

[rtl_433.zip](https://github.com/merbanan/rtl_433/files/10096708/rtl_433.zip)

My device has a QR code attached. The code reads as 021E53B0C695 13.01.2021 MA10100. The third information possibly points to to a relation with another issue where support for MobileAlerts sensors has been requested: https://github.com/merbanan/rtl_433/issues/2079 . At least some information linked in the other issue about MA10100 that information is transmitted every 7 minutes matches TFA device here.

If I can support the integration by providing more information please let me know.

gdt, 2023-09-30:
Is anyone actively working on this issue?

emilic14, 2023-11-09:
Dont think so... although it would be great! ^^

gdt, 2024-06-23:
Can someone retest with git master?  A lot has changed, and there isn't really enough info in this PR to keep it open.