merbanan/rtl_433 #1694: experiences running under VMware?

davepacheco, 2021-04-16:
I'm having trouble decoding any signals on Linux under VMware on MacOS, but I have no trouble running everything directly on MacOS -- same SDR device, same antenna, same location.  Under VMware I seem to be seeing checksum errors and no messages are decoded.  I'm wondering if folks have reported success (or problems) running under VMware?

My setup:

- I'm reading temperature and humidity data from some Acurite devices -- 06044 and 06002RM / 592TXR.
- Using SDRplay RSP 1a
- Using SoapySDR.  I've built SoapySDR, SoapySDRPlay, and rtl_433 from source.

On a 2014 Macbook Pro using SDRPlay API 3, this all worked pretty easily.  I ran `rtl_433 -d '' -R 40`.  I immediately see data points from the Acurite devices.  (I don't need the `-R 40` to see the data, but that's what I ran.)

I tried this on a 2020 Macbook Pro running VMware Fusion 11.5.7 that's running Ubuntu 20.04.02.  I'm using the SDRplay API 3 there too, and also built SoapySDR, SoapySDRPlay, and rtl_433 from source.  When I run `rtl_433 -R 40 -d ''`, I see nothing, even after 30 minutes.  I even tried this: saw data on the older Mac under MacOS, then disconnected the device and connected it to the newer Mac and tried it on VMware and saw nothing, then reconnected it back to the older Mac and saw all the data again.  The device and antenna haven't moved, so I think that rules out a reception problem.

When I run with `-A` under VMware, I see a lot of signals, but it can't decode them.  Here's an example that even has 7 bytes in the first row, like I expect from the correct messages:

```
Detected OOK package    2021-04-04 22:27:39
Analyzing pulses...
Total count:   85,  width: 53.51 ms        (13377 S)
Pulse width distribution:
 [ 0] count:    4,  width:  640 us [608;736]    ( 160 S)
 [ 1] count:   45,  width:  400 us [380;404]    ( 100 S)
 [ 2] count:   36,  width:  212 us [212;220]    (  53 S)
Gap width distribution:
 [ 0] count:    4,  width:  584 us [520;608]    ( 146 S)
 [ 1] count:   46,  width:  208 us [176;224]    (  52 S)
 [ 2] count:   32,  width:  392 us [376;400]    (  98 S)
 [ 3] count:    1,  width:  148 us [148;148]    (  37 S)
 [ 4] count:    1,  width:  296 us [296;296]    (  74 S)
Pulse period distribution:
 [ 0] count:    3,  width: 1216 us [1212;1220]    ( 304 S)
 [ 1] count:    2,  width:  936 us [924;948]    ( 234 S)
 [ 2] count:   77,  width:  608 us [512;624]    ( 152 S)
 [ 3] count:    2,  width:  380 us [368;392]    (  95 S)
Pulse timing distribution:
 [ 0] count:    8,  width:  612 us [520;736]    ( 153 S)
 [ 1] count:   77,  width:  396 us [376;404]    (  99 S)
 [ 2] count:   82,  width:  212 us [176;224]    (  53 S)
 [ 3] count:    1,  width:  148 us [148;148]    (  37 S)
 [ 4] count:    1,  width:  296 us [296;296]    (  74 S)
 [ 5] count:    1,  width: 10004 us [10004;10004]    (2501 S)
Level estimates [high, low]:   1073,      9
RSSI: -23.7 dB SNR: 41.5 dB Noise: -65.2 dB
Frequency offsets [F1, F2]:     995,      0    (+3.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB00F06010264018C00D40094012827148055+AAB01106010264018C00D40094012827148292A155+AAB01006030264018C00D400940128271492A155+AAB00F06020264018C00D4009401282714A155+AAB01006010264018C00D400940128271492A155+AAB00F06010264018C00D4009401282714A155+AAB01006010264018C00D4009401282714A3A155+AAB01406010264018C00D40094012827149292929292A155+AAB01206010264018C00D4009401282714929292A155+AAB01006010264018C00D400940128271492A155+AAB01106010264018C00D4009401282714A492A155+AAB01606010264018C00D400940128271492929292929292A155+AAB01006020264018C00D400940128271492A155+AAB00F06020264018C00D4009401282714A155+AAB01006010264018C00D400940128271492A155+AAB00F06030264018C00D4009401282714A155+AAB01406010264018C00D40094012827149292929292A155+AAB00F06010264018C00D40094012827149055+AAB00F06020264018C00D40094012827148055+AAB01106010264018C00D40094012827149292A155+AAB01006010264018C00D400940128271492A155+AAB01806010264018C00D4009401282714929292929292929292A155+AAB01006010264018C00D400940128271492A155+AAB00F06020264018C00D4009401282714A155+AAB01006010264018C00D400940128271492A155+AAB00F06020264018C00D4009401282714A155+AAB01006010264018C00D4009401282714A2A555
Attempting demodulation... short_width: 212, long_width: 400, reset_limit: 612, sync_width: 640
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=212,l=400,r=612,g=0,t=0,y=640'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {56} 55 de 08 b4 05 77 82
[01] {25} 28 02 ef 80          : 00101000 00000010 11101111 1
```

The ones I've poked at have a bad checksum:

```
Detected OOK package    2021-04-04 22:32:52
acurite_txr_decode: bad checksum: {58} 75 44 fd a5 7f 4c 43 80
Analyzing pulses...
Total count:   86,  width: 53.00 ms        (13251 S)
Pulse width distribution:
 [ 0] count:   34,  width:  208 us [180;244]    (  52 S)
 [ 1] count:   47,  width:  396 us [392;404]    (  99 S)
 [ 2] count:    1,  width:   76 us [76;76]    (  19 S)
 [ 3] count:    1,  width:  116 us [116;116]    (  29 S)
 [ 4] count:    3,  width:  600 us [600;604]    ( 150 S)
Gap width distribution:
 [ 0] count:   48,  width:  212 us [208;220]    (  53 S)
 [ 1] count:   31,  width:  400 us [396;424]    ( 100 S)
 [ 2] count:    1,  width:  108 us [108;108]    (  27 S)
 [ 3] count:    1,  width:  284 us [284;284]    (  71 S)
 [ 4] count:    3,  width:  612 us [612;612]    ( 153 S)
 [ 5] count:    1,  width:  156 us [156;156]    (  39 S)
Pulse period distribution:
 [ 0] count:   80,  width:  604 us [452;616]    ( 151 S)
 [ 1] count:    1,  width:  288 us [288;288]    (  72 S)
 [ 2] count:    3,  width: 1212 us [1212;1216]    ( 303 S)
 [ 3] count:    1,  width:  368 us [368;368]    (  92 S)
Pulse timing distribution:
 [ 0] count:   82,  width:  212 us [180;244]    (  53 S)
 [ 1] count:   78,  width:  396 us [392;424]    (  99 S)
 [ 2] count:    1,  width:   76 us [76;76]    (  19 S)
 [ 3] count:    2,  width:  112 us [108;116]    (  28 S)
 [ 4] count:    6,  width:  604 us [600;612]    ( 151 S)
 [ 5] count:    1,  width:  284 us [284;284]    (  71 S)
 [ 6] count:    1,  width:  156 us [156;156]    (  39 S)
 [ 7] count:    1,  width: 10004 us [10004;10004]    (2501 S)
Level estimates [high, low]:   1000,      8
RSSI: -24.3 dB SNR: 41.9 dB Noise: -66.2 dB
Frequency offsets [F1, F2]:     815,      0    (+3.1 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB017080100D4018C004C0070025C011C009C2714809090908155+AAB014080300D4018C004C0070025C011C009C2714908155+AAB013080200D4018C004C0070025C011C009C27148155+AAB014080100D4018C004C0070025C011C009C2714908155+AAB013080100D4018C004C0070025C011C009C27148155+AAB019080100D4018C004C0070025C011C009C27149090909090908155+AAB015080100D4018C004C0070025C011C009C271490908155+AAB014080100D4018C004C0070025C011C009C2714908155+AAB015080100D4018C004C0070025C011C009C271483908155+AAB014080100D4018C004C0070025C011C009C2714908155+AAB01A080100D4018C004C0070025C011C009C2714909090909090908155+AAB014080100D4018C004C0070025C011C009C2714908555+AAB013080100D4018C004C0070025C011C009C2714A155+AAB015080100D4018C004C0070025C011C009C271490908155+AAB013080200D4018C004C0070025C011C009C27148155+AAB014080100D4018C004C0070025C011C009C2714908155+AAB013080300D4018C004C0070025C011C009C27148155+AAB016080100D4018C004C0070025C011C009C2714909090B155+AAB013080300D4018C004C0070025C011C009C2714C455+AAB015080100D4018C004C0070025C011C009C271490908155+AAB014080100D4018C004C0070025C011C009C2714908155+AAB016080100D4018C004C0070025C011C009C27149090868155+AAB014080100D4018C004C0070025C011C009C2714908155+AAB017080100D4018C004C0070025C011C009C2714909090908155+AAB015080100D4018C004C0070025C011C009C271490908155+AAB013080200D4018C004C0070025C011C009C27148155+AAB016080100D4018C004C0070025C011C009C27149090908755
```

Any idea why this wouldn't work?

merbanan, 2021-04-20:
My guess is that something in the usb transfer breaks down and you are not getting a continuous signal. Supporting this type of issue is beyond the scope of this project. Maybe test with VirtualBox or QEmu to see if it works better.

closed 2021-04-20 by merbanan

rct, 2021-04-20:
@davepacheco - The USB transfers are very latency sensitive, you are likely dropping bytes.  It is possible there is some tuning you could do in terms of which driver and emulated USB hardware is used, but no guarantees.  

My recollection is at least on Windows, VMware had better USB performance versus VirtualBox, Hyper-V, etc.   But remember, things are probably tuned for highest throughput of bulk devices like HDDs, not small high volume, latency sensitive transfers like the SDR.

This problem isn't limited to just emulation for virtualization. I had an ATOM server motherboard (8 core C2750) that was pretty well configured, but couldn't run an RTL-SDR because of latency.  It had great throughput for bulk devices.

I see you aren't using an RTL-SDR.  FWIW, there is a utility, IIRC called `rtl_test` that you could run to find out if there were any problems doing continuous transfers from the RTL-SDR.  If it was quiet, there were no lost bytes. Maybe there is something similar for the SDRplay.

----

If you really want to run in that configuration, change the topology so that you run rtl-tcp on the host OS.  Have the VM guest connect to rtl-tcp on the host.

----

If you are doing 433mhz devices like the Acurites, with no FSK demod, you don't need a lot of CPU horsepower. I'm running an RTL-SDR on an RPI 1B (the first one with a single core) since about 2015.  It is decoding a couple of dozen 433 mhz devices. IIRC uses about 30% of the RPI's CPU continuously.

davepacheco, 2021-04-20:
> @davepacheco - The USB transfers are very latency sensitive, you are likely dropping bytes.
...
> I see you aren't using an RTL-SDR. FWIW, there is a utility, IIRC called `rtl_test` that you could run to find out if there were any problems doing continuous transfers from the RTL-SDR. If it was quiet, there were no lost bytes. Maybe there is something similar for the SDRplay.

This is exactly something I've been wondering about.  I figured that if this were a latency problem, then some component must be explicitly deciding to drop bytes, and it could expose a counter or some other way to observe that this has happened.  I will take a look at how `rtl_test` works and see if the SDRplay device, driver, and/or libraries expose any such counters.  Thanks!

zuckschwerdt, 2021-04-21:
#1698 might be related

zuckschwerdt, 2021-04-21:
Not sure if that's the issue but the `20.02` version from Debian/Ubuntu does not feature the Acurite 3N1 and Atlas decoder. At least `20.11` is needed.

davepacheco, 2021-04-21:
Thanks for the idea.  I built everything from source.  IIRC, the Soapy in Ubuntu did not include the sdrplay module, or else it wasn't built for SDRplay API 3 (I don't remember which -- I ran into both problems at times).